Proxmark/proxmark3 #627: OSX USB comm issues ?

TomHarkness, 2018-07-11:
Hi guys,

Latest os x high sierra with the latest client / firmware build using macports to install all libraries (plot window is working).

If I let the client sit for a while (time varies), somewhere between 20 - 50 secs and try and issue HW tune again the pm3 completely hangs as shown below (no lights flash physically for at least 20secs) and no data is shown:

Measuring antenna characteristics, please wait...........
**No response from Proxmark. Aborting...**
proxmark3>
proxmark3>
proxmark3> hw tune

Measuring antenna characteristics, please wait...
# LF antenna: 29.56 V @   125.00 kHz
# LF antenna: 22.41 V @   134.00 kHz
# LF optimal: 29.56 V @   125.00 kHz
# HF antenna: 21.30 V @    13.56 MHz
Displaying LF tuning graph. Divisor 89 is 134khz, 95 is 125khz.

If I then re issue the command as shown above the info pops up right away as was previously expected but this time the hardware tune obviously doesn't actually do anything as no lights flash and the data comes back instantly. Bare with me here because it gets stranger - IF I were to issue a search command first either lf or hf and have a card over the antenna, search works fine and then HW tune works as expected without the above issue. Sometimes the first instance of "lf search" or "hf search" commands lag as above with HW tune but eventualy complete - sometimes showing the "waiting for response....." message. 

I've tested this both on the pm3 easy and the RDV3 - I have both and yes they are genuine.

HW VER output pm3 easy:

proxmark3> hw ver
[[[ Cached information ]]]

Prox/RFID mark3 RFID instrument
bootrom: master/v3.0.1-386-g708cfa2-suspect 2018-07-11 05:11:22
os: master/v3.0.1-386-g708cfa2-suspect 2018-07-11 05:11:24
LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
HF FPGA image built for 2s30vq100 on 2017/10/27 at 08:30:59

uC: AT91SAM7S256 Rev B
Embedded Processor: ARM7TDMI
Nonvolatile Program Memory Size: 256K bytes. Used: 196984 bytes (75). Free: 65160 bytes (25).
Second Nonvolatile Program Memory Size: None
Internal SRAM Size: 64K bytes
Architecture Identifier: AT91SAM7Sxx Series
Nonvolatile Program Memory Type: Embedded Flash Memory

HW VER RDV3:

proxmark3> hw ver
[[[ Cached information ]]]

Prox/RFID mark3 RFID instrument
bootrom: master/v3.0.1-386-g708cfa2-suspect 2018-07-11 05:11:22
os: master/v3.0.1-386-g708cfa2-suspect 2018-07-11 05:11:24
LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
HF FPGA image built for 2s30vq100 on 2017/10/27 at 08:30:59

uC: AT91SAM7S512 Rev B
Embedded Processor: ARM7TDMI
Nonvolatile Program Memory Size: 512K bytes. Used: 196984 bytes (38). Free: 327304 bytes (62).
Second Nonvolatile Program Memory Size: None
Internal SRAM Size: 64K bytes
Architecture Identifier: AT91SAM7Sxx Series
Nonvolatile Program Memory Type: Embedded Flash Memory

TomHarkness, 2018-07-11:
Update:

This seems related to XQuartz and using the plot window. As soon as I remove XQuartz this issue went away. Commands are now instant.

pwpiwi, 2018-07-12:
Duplicate of issue #283.

TomHarkness, 2018-07-12:
Look to be the same issue. Can definitely confirm that it was XQuartz causing the issue. Client is snappy and responsive even after long periods of being left alone now.

drandreas, 2018-10-04:
I think this issue is related to app-nap. A feature where background applications are put to sleep.
In our case the timeout of `uart_receive` changes from 30ms to 10s.

Apple provides a Obj-C API to disable this feature: (beginActivityWithOptions)[https://developer.apple.com/documentation/foundation/nsprocessinfo]. This also fixes the issue described above. I'm happy to provide a fix, if foreign languages are accepted 😄 .

iceman1001, 2018-10-05:
Bring on the fix!

pwpiwi, 2018-10-05:
Indeed, this would be a long long awaited fix.

> In our case the `select` timeout in `uart_receive` changes from 30ms to 10s.

Is this an observation or is this documented somewhere?

> An alternative solution is to separate rx and tx into two threads. 

Hmm, are we sure that the underlying rx and tx functions are thread save on all OS?

> Then an extended `uart_receive` timeout would delay tx. This would also allow for a relaxed polling e.g. 100ms.

I am not sure if I understand this. Delaying tx (sending a command) would still result in a delayed rx (receiving the response) and still running into a timeout?

drandreas, 2018-10-05:
> Indeed, this would be a long long awaited fix.
> 
> > In our case the `select` timeout in `uart_receive` changes from 30ms to 10s.
> 
> Is this an observation or is this documented somewhere?

I did trace the code execution with _Instruments_

> 
> > An alternative solution is to separate rx and tx into two threads.
> 
> Hmm, are we sure that the underlying rx and tx functions are thread save on all OS?
> 

I never encountered an OS where it was an issue. But you are right its a grey area. Thats why I opted for disabling AppNap for now: #687.

Can any one test if it compiles on Windows?

drandreas, 2018-10-05:
Apple WWDC several workshops related to AppNap in 2013. In this one they explain how macOS batch processes timer timeouts and extends timer periods like we encountered:
https://developer.apple.com/videos/play/wwdc2013/209/
The talk also covers how an energy efficient app should be designed.

pwpiwi, 2018-10-06:
Compiles on Windows. Travis tests passed for different OSX versions and Linux.

Any Apple user out there to test the fix? @TomHarkness, @Kaicastledine, @digitalentropy ?

digitalentropy, 2018-10-06:
Don’t have current access to macOS system, but will test as soon as I can.  I imagine others will be able to test sooner.

TomHarkness, 2018-10-08:
Did some brief testing and the client seems properly responsive with the gui now! Well done!

PR #687 (2018-10-08): osx: disable app-nap during serial comm
Apple introduced app-nap with OS X 10.10. This feature saves power by throttling apps running in background. However, it also less accurate timers in systemcalls. In our case a 30ms select timeout would take up to 10s.

This patch uses an API also added with 10.10 to disable app-nap as long as the serial port is polled and fixed #283, fixed #627.
closed 2018-10-08 by pwpiwi