Proxmark/proxmark3 #628: mfkeys.lua broken

TomHarkness, 2018-07-12:
When running the mfkeys I get the following output. Tried with both pm3 easy and RDV3.

proxmark3> script run mfkeys
--- Executing: mfkeys.lua, args ''
This script implements check keys.
It utilises a large list of default keys (currently 92 keys).
If you want to add more, just put them inside mf_default_keys.lua.
Found a NXP MIFARE CLASSIC 1k | Plus 2k tag
Testing block 3, keytype 0, with 85 keys
.../client/scripts/mfkeys.lua:60: invalid value (nil) at index 93 in table for 'concat'

pwpiwi, 2018-07-12:
In line 72 of mfkeys.lua, try replacing
```
start = start+n+1
```
with
```
start = start+n
```

TomHarkness, 2018-07-14:
This seems to fix the issue.

pwpiwi, 2018-07-17:
Are you going to raise a Pull Request with the fix?

TomHarkness, 2018-07-18:
Apologies for delay on that - been a bit flat out with work.

closed 2018-07-18 by TomHarkness

pwpiwi, 2018-07-19:
No problem. I pushed the fix - in the end it wasn't too much work.