Proxmark/proxmark3 #633: Make targets OS X

TomHarkness, 2018-07-29:
Firstly - still getting my head around the codebase, don't hate me if this is an obvious thing that I am missing :P.

OSX High Sierra.

I can compile fine via "make clean; make" on the command line. However:

I get an error "No target exists for 'all clean" when executing the make file from within Atom editor with the popular build-make plugin to execute makefile targets.

If I change the entries in the makefile to be seperate like:

all: %: client/% bootrom/% armsrc/% recovery/% mfkey/% nonce2key/%
clean: %: client/% bootrom/% armsrc/% recovery/% mfkey/% nonce2key/%

I can separately run clean and make all to compile.

pwpiwi, 2018-07-31:
I don't know "Atom editor" and the "popular build-make plugin" (OSX is misrepresented among the developers - mostly Linux and Windows here) and therefore cannot provide a solution for your issue. But it seems not to be related to the Proxmark code. I suggest to post your question in a relevant (Atom editor?) forum.

TomHarkness, 2018-08-02:
Will follow up on Atom the Atom forums. Thanks pwpiwi! 

My main environment is OS X so had hoped to be able to set up a reasonable development environment for pm3 stuff. I can otherwise compile everything, debug and run fine. Just this one "nice feature" that I can't get to work in the IDE. No issues though I will press on!

pwpiwi, 2018-08-03:
If you can solve your issue, please come back here and write a short note. I will keep this open for a few weeks.

TomHarkness, 2018-09-26:
Not yet sorry. Been focusing on some custom antennas that will be released soon. I’ll get back to the software side of things soon :) I do intend to fix this eventually lol.

pwpiwi, 2018-11-30:
Any news? Close?

pwpiwi, 2019-04-25:
@TomHarkness: still around? What happened to your issue with the Atom Editor?

closed 2019-08-12 by pwpiwi