Proxmark/proxmark3 #634: Got 0 keys from proxmark.

w1109790800, 2018-07-30:
when i use the Nested 
i got an error "Got 0 keys from proxmark. "

TomHarkness, 2018-08-03:
More info please. What is the output of "hw ver" and "hw tune"? This is a common issue with mismatched client / firmware versions.

w1109790800, 2018-08-05:
proxmark3> hw version
uC: AT91SAM7S256 Rev D          
Embedded Processor: ARM7TDMI          
Nonvolatile Program Memory Size: 256K bytes          
Second Nonvolatile Program Memory Size: None          
Internal SRAM Size: 64K bytes          
Architecture Identifier: AT91SAM7Sxx Series          
Nonvolatile Program Memory Type: Embedded Flash Memory  

proxmark3> hw tune

Measuring antenna characteristics, please wait...          
# LF antenna:  0.00 V @   125.00 kHz          
# LF antenna:  0.00 V @   134.00 kHz          
# LF optimal:  0.00 V @ 12000.00 kHz          
# HF antenna:  0.00 V @    13.56 MHz          
# Your LF antenna is unusable.          
# Your HF antenna is unusable.

pwpiwi, 2018-08-05:
hw version shows version information on bootrom and os since 2009. Your output doesn't. Is it possible that your firmware is *that* old?

pwpiwi, 2018-08-12:
And the usual advise may apply here as well: don't mix client and firmware versions. If you want to use a newer client, you need to update (flash) the firmware as well.

closed 2018-09-26 by pwpiwi