Proxmark/proxmark3 #636: hf mf broken after flash from old version

ralik, 2018-08-03:
Hi all,
I've always been working with a proxmark loaded with a version from July 2014.
Today, I thought it would be a good idea to check newer commands. I flashed with latest GitHub repo and have lost the ability to (on cards with which I never had any problem):
- use darkside attack.
- complete a nested attack.
- simply auth and read a block easily.
- simply writing a block.
- even hf 14a reader is having a hard time finding anything.

I just keep getting a massive amount of error messages, "Can't select card" or "Multiple tags detected. Collision after Bit xx" with whatever card<->antenna positions until the PM3 crashes.

I tried to go a few commits back and re-flash but same problem.
I was thinking mehhh just forget about it and go back to my cave ¯\_(ツ)_/¯
But flashing with my good old 2014 Bootrom+fullimage, did not help at all... The hf mf commands are not back! Looks like flashing a newer version has made some irreversible changes to my PM3.

Any idea what happened here? And what can I try to fix that issue.
Any help would be highly appreciated...

pwpiwi, 2018-08-03:
Sensitivity has increased since the very old versions. Results in clipping at low distances. Leave a distance of 2 cm between antenna and card and it should work again.

marshmellow42, 2018-08-03:
pwpiwi probably nailed it, but make sure you flash the bootrom and the fullimage.  And use the client that matches the firmware..  
and nothing in the code could cause something irreversible, are you sure the device is taking the flashing?

ralik, 2018-08-03:
I did try every possible angle, distance (as I used to before anyway), but this time, it just fails after like 5 s then there is no angle/position that will help.
Device is being flashed (at least according to ./flasher). Also tried on linux with old github repo. hw version shows the difference.

I went from a r839 to a V3.x.x, went back to r839 and lost all "normal" functioning of my old PM3.

marshmellow42, 2018-08-03:
I assume you checked the obvious like antenna connections, hw tune?

ralik, 2018-08-03:
Yes, all good here. 
I have a PCB printed antenna and a couple of handmade ones.

pwpiwi, 2018-08-03:
Please let us see some output:
* when flashing to current release
* client start
* hw ver
* hw tune

ralik, 2018-08-04:
Here is an example of a simple block read

```
proxmark3> hf mf rdbl 0 A FFFFFFFFFFFF
--block no:0, key type:A, key:ff ff ff ff ff ff            
#db# ISO14443A Timeout set to 1060 (10ms)          
#db# ISO14443A Timeout set to 10 (0ms)          
#db# ISO14443A Timeout set to 1060 (10ms)          
#db# rand tag nonce len: 4          
#db# auth uid: 8f237915 nt: 75a0c162          
#db# Cmd CRC response error.          
#db# Read block error          
#db# READ BLOCK FINISHED          
isOk:00          
proxmark3> hf mf rdbl 0 A FFFFFFFFFFFF
--block no:0, key type:A, key:ff ff ff ff ff ff            
#db# ISO14443A Timeout set to 1060 (10ms)          
#db# ISO14443A Timeout set to 10 (0ms)          
#db# ISO14443A Timeout set to 1060 (10ms)          
#db# Multiple tags detected. Collision after Bit 12          
#db# Can't select card          
#db# READ BLOCK FINISHED          
isOk:00          
proxmark3> hf mf rdbl 0 A FFFFFFFFFFFF
--block no:0, key type:A, key:ff ff ff ff ff ff            
#db# ISO14443A Timeout set to 1060 (10ms)          
#db# ISO14443A Timeout set to 10 (0ms)          
#db# ISO14443A Timeout set to 1060 (10ms)          
#db# rand tag nonce len: 4          
#db# auth uid: 8f237915 nt: caa01618          
#db# Authentication failed. Card timeout.          
#db# Auth error          
#db# READ BLOCK FINISHED          
isOk:00          
proxmark3> hf mf rdbl 0 A FFFFFFFFFFFF
--block no:0, key type:A, key:ff ff ff ff ff ff            
#db# ISO14443A Timeout set to 1060 (10ms)          
#db# ISO14443A Timeout set to 10 (0ms)          
#db# ISO14443A Timeout set to 1060 (10ms)          
#db# Multiple tags detected. Collision after Bit 27          
#db# Can't select card          
#db# READ BLOCK FINISHED          
isOk:00          
proxmark3> hf mf rdbl 0 A FFFFFFFFFFFF
--block no:0, key type:A, key:ff ff ff ff ff ff            
#db# ISO14443A Timeout set to 1060 (10ms)          
#db# ISO14443A Timeout set to 10 (0ms)          
#db# ISO14443A Timeout set to 1060 (10ms)          
#db# rand tag nonce len: 4          
#db# auth uid: 8f237915 nt: 3074dace          
#db# Authentication failed. Card timeout.          
#db# Auth error          
#db# READ BLOCK FINISHED          
isOk:00          
proxmark3> hf mf rdbl 0 A FFFFFFFFFFFF
--block no:0, key type:A, key:ff ff ff ff ff ff            
#db# ISO14443A Timeout set to 1060 (10ms)          
#db# ISO14443A Timeout set to 10 (0ms)          
#db# ISO14443A Timeout set to 1060 (10ms)          
#db# Can't select card          
#db# READ BLOCK FINISHED          
isOk:00          
proxmark3> hf mf rdbl 0 A FFFFFFFFFFFF
--block no:0, key type:A, key:ff ff ff ff ff ff            
#db# ISO14443A Timeout set to 1060 (10ms)          
#db# ISO14443A Timeout set to 10 (0ms)          
#db# ISO14443A Timeout set to 1060 (10ms)          
#db# Multiple tags detected. Collision after Bit 32          
#db# Can't select card          
#db# READ BLOCK FINISHED          
isOk:00          
proxmark3> hf mf rdbl 0 A FFFFFFFFFFFF
--block no:0, key type:A, key:ff ff ff ff ff ff            
#db# ISO14443A Timeout set to 1060 (10ms)          
#db# ISO14443A Timeout set to 10 (0ms)          
#db# ISO14443A Timeout set to 1060 (10ms)          
#db# Can't select card          
#db# READ BLOCK FINISHED          
isOk:00          
proxmark3> hf mf rdbl 0 A FFFFFFFFFFFF
--block no:0, key type:A, key:ff ff ff ff ff ff            
#db# ISO14443A Timeout set to 1060 (10ms)          
#db# ISO14443A Timeout set to 10 (0ms)          
#db# ISO14443A Timeout set to 1060 (10ms)          
#db# Multiple tags detected. Collision after Bit 4          
#db# Can't select card          
#db# READ BLOCK FINISHED          
isOk:00          
proxmark3> hf mf rdbl 0 A FFFFFFFFFFFF
--block no:0, key type:A, key:ff ff ff ff ff ff            
#db# ISO14443A Timeout set to 1060 (10ms)          
#db# ISO14443A Timeout set to 10 (0ms)          
#db# ISO14443A Timeout set to 1060 (10ms)          
#db# rand tag nonce len: 4          
#db# auth uid: 8f237915 nt: c02ad681          
#db# READ BLOCK FINISHED          
isOk:01 data:8f 23 79 15 c0 28 04 00 00 00 00 00 21 27 20 00           
proxmark3
```

ralik, 2018-08-04:
Sorry, my bad.

Here it is:

```
pm3 --> hf mf dbg 0
#db# Debug level: 0          
pm3 --> hf mf rdbl 0 A 9c06d18d39e4
--block no:0, key type:A, key:9C 06 D1 8D 39 E4            
isOk:00          
pm3 --> hf mf rdbl 0 A 9c06d18d39e4
--block no:0, key type:A, key:9C 06 D1 8D 39 E4            
isOk:00          
pm3 --> hf mf rdbl 0 A 9c06d18d39e4
--block no:0, key type:A, key:9C 06 D1 8D 39 E4            
isOk:00          
pm3 --> hf mf rdbl 0 A 9c06d18d39e4
--block no:0, key type:A, key:9C 06 D1 8D 39 E4            
isOk:00          
pm3 --> hf mf rdbl 0 A 9c06d18d39e4
--block no:0, key type:A, key:9C 06 D1 8D 39 E4            
#db# Multiple tags detected. Collision after Bit 28          
isOk:00          
pm3 --> hf mf rdbl 0 A 9c06d18d39e4
--block no:0, key type:A, key:9C 06 D1 8D 39 E4            
#db# Multiple tags detected. Collision after Bit 22          
isOk:00          
pm3 --> hf mf rdbl 0 A 9c06d18d39e4
--block no:0, key type:A, key:9C 06 D1 8D 39 E4            
isOk:00          
pm3 --> hf mf rdbl 0 A 9c06d18d39e4
--block no:0, key type:A, key:9C 06 D1 8D 39 E4            
isOk:00          
pm3 --> hf mf rdbl 0 A 9c06d18d39e4
--block no:0, key type:A, key:9C 06 D1 8D 39 E4            
isOk:00          
pm3 --> hf mf rdbl 0 A 9c06d18d39e4
--block no:0, key type:A, key:9C 06 D1 8D 39 E4            
isOk:00          
pm3 --> hf mf rdbl 0 A 9c06d18d39e4
--block no:0, key type:A, key:9C 06 D1 8D 39 E4            
isOk:01 data:C2 B8 C0 19 A3 08 04 00 62 63 64 65 66 67 68 69
```

ralik, 2018-08-06:
Yes, I have tried a few distro to see if there were any differences. But I get pretty much the same poor results. 
Also I went back to an old rev I have on my PC (ProxSpace from 2012), that is the first one I used when I purchased my PM3, it has a few traces in the client's folder.
At this time I don't believe the wording "Multiple tags detected" or "Collision after Bit" existed in the code. 
I have a 6.6MB proxmark3.log in that folder which only contains this wording from when I used it yesterday. Everything else is from back in 2012 and said that sentence.
Maybe I should JTAG-flash to fully revert back to the original version where I had no issue with HF MF

pwpiwi, 2018-08-08:
Don't mix client and firmware version. Iceman client must go with Iceman firmware. Official client must go with official firmware. And both client and firmware must be of the same version.

Flashing the firmware is reversible. No need to use JTAG-flash when flasher still worls.

BarikMana, 2018-08-09:
I did flash firmware with iceman's firmware before using his client.
And I did flash fullimage+bootroom+fgpaimage+osimage before using the old 2012 proxspace client. 
Yet I am still seeing "Multiple tags detected" or "Collision after Bit" which appears to have been introduced after 2012.

ralik, 2018-08-09:
In the last output, I flashed with iceman's firmware and was using iceman's firmware from the same commit.
When I tested with an old ProxSpace from 2012, I flashed bootroom, fpgaimage, fullimage, osimage from the same 2012 version and I was using the client from the same 2012 version.
Yet, I am still seeing "Multiple tags detected" or "Collision after Bit" even though this does not appear to be in the source code from 2012. Meaning that it appears to be some leftover from newer 2018 revisions that was not erased after flashing to a 2012 older version.

marshmellow42, 2018-08-09:
I cannot speak for icemans repo, as it says, use at your own risk.

iceman1001, 2018-08-09:
I see you are using linux..
try
**./update.sh**

pwpiwi, 2018-08-14:
Issue fixed using ./update.sh?

iceman1001, 2018-08-14:
...this issue is a bit inconsistent.

**For testing purposes,** 
- Which pm3 device do you have?  [old green pcb,  rdv10 (orange pcb),  rdv20 (blue pcb) ,  rdv30 easy or rdv40? ]
- test one repo at the same time.  ie  compile / flash  from same repo.
- output from     hw version,   hw tune,  hw status.
- 14a testing,    use different distances between tag and antenna.  Normally 1-2cm.

if hw tune reports bad values from antenna,  then check your connection with antenna.  The client should report back values in order for the pm3 to function normal.

Barmallini, 2018-08-14:
Same issue. PM3 easy (black)
Prox/RFID mark3 RFID instrument
bootrom: master/v3.0.1-382-gab20cc3-suspect 2018-07-31 11:46:17
os: master/v3.0.1-382-gab20cc3-suspect 2018-07-31 11:46:30
LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
HF FPGA image built for 2s30vq100 on 2017/10/27 at 08:30:59

uC: AT91SAM7S256 Rev D
Embedded Processor: ARM7TDMI
Nonvolatile Program Memory Size: 256K bytes. Used: 194936 bytes (74%). Free: 672
08 bytes (26%).
Second Nonvolatile Program Memory Size: None
Internal SRAM Size: 64K bytes
Architecture Identifier: AT91SAM7Sxx Series
Nonvolatile Program Memory Type: Embedded Flash Memory

proxmark3> hw tune

Measuring antenna characteristics, please wait.........
# LF antenna: 23.92 V @   125.00 kHz
# LF antenna: 18.84 V @   134.00 kHz
# LF optimal: 23.65 V @   122.45 kHz
# HF antenna: 14.57 V @    13.56 MHz
Displaying LF tuning graph. Divisor 89 is 134khz, 95 is 125khz.

proxmark3> hf mf rdbl 0 A FFFFFFFFFFFF
--block no:0, key type:A, key:ff ff ff ff ff ff
#db# Can't select card
#db# READ BLOCK FINISHED
isOk:00

iceman1001, 2018-08-14:
ok, so different distance between tag and antenna...
and pm3 easy have a lot of cheap clones out on aliexpress etc,  they (the clones) have been known to be problematic

Barmallini, 2018-08-15:
Yes, i have aliexpress clone.
[image]

It snoop something, but not as shown at https://code.google.com/archive/p/proxmark3/wikis/RunningPM3.wiki section "Get the UID of a Mifare card using ’snooping’ capabilities"

ralik, 2018-08-16:
> Issue fixed using ./update.sh?
Yes
Old board (green one from 2011-2012)

Distance/angle/position doesn't have much effect (except from what you would expect) after that. Even hf mf hard runs smoothly.

closed 2018-08-16 by pwpiwi

hackcrypto, 2022-01-26:
> Sensitivity has increased since the very old versions. Results in clipping at low distances. Leave a distance of 2 cm between antenna and card and it should work again.

Thanks bro it WORK !