Proxmark/proxmark3 #640: Build on OSX segfaults clang

evanrinehart, 2018-08-05:
After cloning and simply typing `make' it goes smoothly for a while until this

```
gcc -MT obj/hardnested/hardnested_bf_core_AVX512.o -MMD -MP -MF obj/hardnested/hardnested_bf_core.Td  -std=c99 -D_ISOC99_SOURCE -I. -I../include -I../common -I../common/polarssl -I../zlib -I../uart -I/opt/local/include -I../liblua -Wall -g -O3  -DWITH_ISO14443a_StandAlone -DWITH_ISO15693 -DWITH_ISO14443a -DWITH_ISO14443b -DWITH_ICLASS -DWITH_LEGICRF -DWITH_HITAG -DWITH_CRC -DWITH_HFSNOOP -DWITH_GUI -mmmx -msse2 -mavx -mavx2 -mavx512f -c -o obj/hardnested/hardnested_bf_core_AVX512.o hardnested/hardnested_bf_core.c
clang: error: unable to execute command: Segmentation fault: 11
clang: error: clang frontend command failed due to signal (use -v to see invocation)
Apple LLVM version 7.0.2 (clang-700.1.81)
Target: x86_64-apple-darwin15.2.0
Thread model: posix
clang: note: diagnostic msg: PLEASE submit a bug report to http://developer.apple.com/bugreporter/ and include the crash backtrace, preprocessed source, and associated run script.
clang: note: diagnostic msg:
********************

PLEASE ATTACH THE FOLLOWING FILES TO THE BUG REPORT:
Preprocessed source(s) and associated run script(s) are located at:
clang: note: diagnostic msg: /var/folders/p2/6nrm_0tx61j60mr2yfzgls340000gn/T/hardnested_bf_core-4b7009.c
clang: note: diagnostic msg: /var/folders/p2/6nrm_0tx61j60mr2yfzgls340000gn/T/hardnested_bf_core-4b7009.sh
clang: note: diagnostic msg:

********************
make[1]: *** [obj/hardnested/hardnested_bf_core_AVX512.o] Error 254
make: *** [client/all] Error 2
```

pwpiwi, 2018-08-09:
What is the result of
```
echo | gcc -E -mavx512f -
```

evanrinehart, 2018-08-09:
```
λ> echo | gcc -E -mavx512f -
# 1 "<stdin>"
# 1 "<built-in>" 1
# 1 "<built-in>" 3
# 331 "<built-in>" 3
# 1 "<command line>" 1
# 1 "<built-in>" 2
# 1 "<stdin>" 2
```

evanrinehart, 2018-08-12:
Ok thanks.

closed 2018-08-12 by evanrinehart