Proxmark/proxmark3 #641: Lots of function not working after #635

Fl0-0, 2018-08-06:
Lots of errors since https://github.com/Proxmark/proxmark3/commit/5125e4263c36063f49d3b410b78da293b705b7d2

- unknown command for lf commands
- Unknown card format for hf legic

```
Prox/RFID mark3 RFID instrument          
bootrom: Missing/Invalid version information
os: master/v2.2.0-892-g472345d-suspect 2018-08-06 07:59:29
fpga_lf.bit built for 2s30vq100 on 2015/03/06 at 07:38:04
fpga_hf.bit built for 2s30vq100 on 2017/10/27 at 08:30:59          
uC: AT91SAM7S512 Rev B          
Embedded Processor: ARM7TDMI          
Nonvolatile Program Memory Size: 512K bytes. Used: 174249 bytes (33%). Free: 350039 bytes (67%).          
Second Nonvolatile Program Memory Size: None          
Internal SRAM Size: 64K bytes          
Architecture Identifier: AT91SAM7Sxx Series          
Nonvolatile Program Memory Type: Embedded Flash Memory          
proxmark3> lf search
#db# unknown command:: 0x0205          
Waiting for a response from the proxmark...          
You can cancel this operation by pressing the pm3 button          
command execution time out          
NOTE: some demods output possible binary
  if it finds something that looks like a tag          
False Positives ARE possible
          

Checking for known tags:
          
#db# unknown command:: 0x0218          
Waiting for a response from the proxmark...          
You can cancel this operation by pressing the pm3 button          
Command timed out          
#db# Uknown frame length: 2          
#db# unknown command:: 0x0225          
Waiting for a response from the proxmark...          
You can cancel this operation by pressing the pm3 button          
command execution time out          

No Data Found! - maybe not an LF tag?
          
proxmark3> 
proxmark3> hf legic reader
#db# setting up legic card          
#db# Unknown card format: 26  
proxmark3> lf em 410xread
#db# unknown command:: 0x021c 
proxmark3> lf hid read
#db# unknown command:: 0x020b proxmark3> lf t55xx detect
#db# unknown command:: 0x0214          
Waiting for a response from the proxmark...          
You can cancel this operation by pressing the pm3 button          
command execution time out          
proxmark3> 
```

marshmellow42, 2018-08-06:
Can you show the output of make clean && make all?

marshmellow42, 2018-08-06:
You os version output indicates a very old version, did you flash the device?

Fl0-0, 2018-08-06:
I always flash after code updates.
Maybe the reported 'old version' is related to issue #642 .

```
flo@kali:~/tools/NFC/proxmark3$ git pull -v
From https://github.com/Proxmark/proxmark3
 = [up to date]      master     -> origin/master
 = [up to date]      gh-pages   -> origin/gh-pages
 = [up to date]      unstable   -> origin/unstable
Already up to date.
flo@kali:~/tools/NFC/proxmark3$ make clean && make
make -C client  clean
make[1]: Entering directory '/home/flo/tools/NFC/proxmark3/client'
rm -f proxmark3 flasher fpga_compress  proxmark3.exe  flasher.exe  fpga_compress.exe obj/uart_posix.o obj/uart_win32.o obj/util.o obj/util_posix.o obj/ui.o obj/comms.o obj/crapto1/crapto1.o obj/crapto1/crypto1.o obj/polarssl/des.o obj/polarssl/aes.o obj/polarssl/bignum.o obj/polarssl/rsa.o obj/polarssl/sha1.o obj/mfkey.o obj/loclass/cipher.o obj/loclass/cipherutils.o obj/loclass/ikeys.o obj/loclass/elite_crack.o obj/loclass/fileutils.o obj/whereami.o obj/mifarehost.o obj/parity.o obj/crc.o obj/crc16.o obj/crc64.o obj/iso14443crc.o obj/iso15693tools.o obj/graph.o obj/cmddata.o obj/lfdemod.o obj/emv/crypto_polarssl.o obj/emv/crypto.o obj/emv/emv_pk.o obj/emv/emv_pki.o obj/emv/emv_pki_priv.o obj/emv/test/cryptotest.o obj/emv/apduinfo.o obj/emv/dump.o obj/emv/tlv.o obj/emv/emv_tags.o obj/emv/dol.o obj/emv/emvcore.o obj/emv/test/crypto_test.o obj/emv/test/sda_test.o obj/emv/test/dda_test.o obj/emv/test/cda_test.o obj/emv/cmdemv.o obj/cmdhf.o obj/cmdhflist.o obj/cmdhf14a.o obj/cmdhf14b.o obj/cmdhf15.o obj/cmdhfepa.o obj/cmdhflegic.o obj/cmdhficlass.o obj/cmdhfmf.o obj/cmdhfmfu.o obj/cmdhfmfhard.o obj/hardnested/hardnested_bruteforce.o obj/cmdhftopaz.o obj/cmdhw.o obj/cmdlf.o obj/cmdlfawid.o obj/cmdlfcotag.o obj/cmdlfem4x.o obj/cmdlffdx.o obj/cmdlfgproxii.o obj/cmdlfhid.o obj/cmdlfhitag.o obj/cmdlfio.o obj/cmdlfindala.o obj/cmdlfjablotron.o obj/cmdlfnexwatch.o obj/cmdlfnoralsy.o obj/cmdlfparadox.o obj/cmdlfpcf7931.o obj/cmdlfpresco.o obj/cmdlfpyramid.o obj/cmdlfsecurakey.o obj/cmdlft55xx.o obj/cmdlfti.o obj/cmdlfviking.o obj/cmdlfvisa2000.o obj/cmdlfpac.o obj/cmdparser.o obj/cmdmain.o obj/scripting.o obj/cmdscript.o obj/pm3_binlib.o obj/pm3_bitlib.o obj/protocols.o obj/deflate.o obj/adler32.o obj/trees.o obj/zutil.o obj/inflate.o obj/inffast.o obj/inftrees.o obj/proxgui.o obj/proxguiqt.o obj/proxguiqt.moc.o obj/hardnested/hardnested_bf_core_NOSIMD.o obj/hardnested/hardnested_bitarray_core_NOSIMD.o obj/hardnested/hardnested_bf_core_MMX.o obj/hardnested/hardnested_bitarray_core_MMX.o obj/hardnested/hardnested_bf_core_SSE2.o obj/hardnested/hardnested_bitarray_core_SSE2.o obj/hardnested/hardnested_bf_core_AVX.o obj/hardnested/hardnested_bitarray_core_AVX.o obj/hardnested/hardnested_bf_core_AVX2.o obj/hardnested/hardnested_bitarray_core_AVX2.o obj/hardnested/hardnested_bf_core_AVX512.o obj/hardnested/hardnested_bitarray_core_AVX512.o obj/*.o *.moc.cpp ui/ui_overlays.h
cd ../liblua && make clean
make[2]: Entering directory '/home/flo/tools/NFC/proxmark3/liblua'
rm -f liblua.a lua luac lapi.o lcode.o lctype.o ldebug.o ldo.o ldump.o lfunc.o lgc.o llex.o lmem.o lobject.o lopcodes.o lparser.o lstate.o lstring.o ltable.o ltm.o lundump.o lvm.o lzio.o lauxlib.o lbaselib.o lbitlib.o lcorolib.o ldblib.o liolib.o lmathlib.o loslib.o lstrlib.o ltablib.o loadlib.o linit.o  lua.o luac.o
make[2]: Leaving directory '/home/flo/tools/NFC/proxmark3/liblua'
make[1]: Leaving directory '/home/flo/tools/NFC/proxmark3/client'
make -C bootrom  clean
make[1]: Entering directory '/home/flo/tools/NFC/proxmark3/bootrom'
rm -rf obj/*.o
rm -rf obj/*.elf
rm -rf obj/*.s19
rm -rf obj/*.map
rm -rf obj/*.d
rm -rf version.c
make[1]: Leaving directory '/home/flo/tools/NFC/proxmark3/bootrom'
make -C armsrc  clean
make[1]: Entering directory '/home/flo/tools/NFC/proxmark3/armsrc'
rm -rf obj/*.o
rm -rf obj/*.elf
rm -rf obj/*.s19
rm -rf obj/*.map
rm -rf obj/*.d
rm -rf obj/*.z
rm -rf obj/*.bin
rm -rf version.c
make[1]: Leaving directory '/home/flo/tools/NFC/proxmark3/armsrc'
make -C recovery  clean
make[1]: Entering directory '/home/flo/tools/NFC/proxmark3/recovery'
rm -f bootrom.bin fullimage.bin proxmark3_recovery.bin
make[1]: Leaving directory '/home/flo/tools/NFC/proxmark3/recovery'
make -C tools/mfkey  clean
make[1]: Entering directory '/home/flo/tools/NFC/proxmark3/tools/mfkey'
rm -f crypto1.o crapto1.o parity.o util_posix.o mfkey.o mfkey32 mfkey64  mfkey32.exe  mfkey64.exe
make[1]: Leaving directory '/home/flo/tools/NFC/proxmark3/tools/mfkey'
make -C client  all
make[1]: Entering directory '/home/flo/tools/NFC/proxmark3/client'
Compiling liblua, using platform linux
cd ../liblua && make linux
make[2]: Entering directory '/home/flo/tools/NFC/proxmark3/liblua'
make all SYSCFLAGS="-DLUA_USE_LINUX" SYSLIBS="-Wl,-E -ldl -lreadline -ltermcap -lncurses"
make[3]: Entering directory '/home/flo/tools/NFC/proxmark3/liblua'
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o lapi.o lapi.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o lcode.o lcode.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o lctype.o lctype.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o ldebug.o ldebug.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o ldo.o ldo.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o ldump.o ldump.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o lfunc.o lfunc.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o lgc.o lgc.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o llex.o llex.c
[... cut ...]
gcc -std=c99 -D_ISOC99_SOURCE -I../../include -I../../common -I../../client -Wall -O3 -c -o mfkey.o ../../client/mfkey.c
gcc -std=c99 -D_ISOC99_SOURCE -I../../include -I../../common -I../../client -Wall -O3  -o mfkey32 crypto1.o crapto1.o parity.o util_posix.o mfkey.o mfkey32.c
gcc -std=c99 -D_ISOC99_SOURCE -I../../include -I../../common -I../../client -Wall -O3  -o mfkey64 crypto1.o crapto1.o parity.o util_posix.o mfkey.o mfkey64.c
make[1]: Leaving directory '/home/flo/tools/NFC/proxmark3/tools/mfkey'
```

marshmellow42, 2018-08-06:
pushed the fix.  please try again.

sorry about that i missed one WITH_LF ifdef...

Fl0-0, 2018-08-06:
OK thanks, 

LF functions are OK !
I will retry hf legic commands this afternoon.

Fl0-0, 2018-08-06:
hf legic works. Certainly an antenna error i think. 

Closing.

closed 2018-08-06 by Fl0-0