Proxmark/proxmark3 #642: Bootloader version not reported correctly since #631

Fl0-0, 2018-08-06:
**bootrom: Missing/Invalid version information**

```
$ ./flasher /dev/ttyACM0 -b ../bootrom/obj/bootrom.elf ../armsrc/obj/fullimage.elf
Loading ELF file '../bootrom/obj/bootrom.elf'...
Loading usable ELF segments:
0: V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94
1: V 0x00200000 P 0x00100200 (0x00000c8c->0x00000c8c) [R X] @0x298

Loading ELF file '../armsrc/obj/fullimage.elf'...
Loading usable ELF segments:
0: V 0x00102000 P 0x00102000 (0x00026f78->0x00026f78) [R X] @0x94
1: V 0x00200000 P 0x00128f78 (0x00001934->0x00001934) [RW ] @0x2700c
Note: Extending previous segment from 0x26f78 to 0x288ac bytes

Waiting for Proxmark to appear on /dev/ttyACM0 .
 Found.
Entering bootloader...
(Press and release the button only to abort)
Waiting for Proxmark to appear on /dev/ttyACM0 .......
 Found.

Flashing...
Writing segments for file: ../bootrom/obj/bootrom.elf
 0x00100000..0x001001ff [0x200 / 1 blocks]. OK
 0x00100200..0x00100e8b [0xc8c / 7 blocks]....... OK

Writing segments for file: ../armsrc/obj/fullimage.elf
 0x00102000..0x0012a8ab [0x288ac / 325 blocks]..................................................................................................................................................................................................................................................................................................................................... OK

Resetting hardware...
All done.

Have a nice day!
$ ./proxmark3 /dev/ttyACM0
Prox/RFID mark3 RFID instrument          
bootrom: Missing/Invalid version information
os: master/v2.2.0-892-g472345d-suspect 2018-08-06 08:27:53
fpga_lf.bit built for 2s30vq100 on 2015/03/06 at 07:38:04
fpga_hf.bit built for 2s30vq100 on 2017/10/27 at 08:30:59          
uC: AT91SAM7S512 Rev B          
Embedded Processor: ARM7TDMI          
Nonvolatile Program Memory Size: 512K bytes. Used: 174249 bytes (33%). Free: 350039 bytes (67%).          
Second Nonvolatile Program Memory Size: None          
Internal SRAM Size: 64K bytes          
Architecture Identifier: AT91SAM7Sxx Series          
Nonvolatile Program Memory Type: Embedded Flash Memory          
proxmark3>
```

Fl0-0, 2018-08-06:
OK found the problem:

In commit https://github.com/Proxmark/proxmark3/commit/472345daee388f8d6786ead2f04b4ae0ff0462d4 the version.c generation has been moved to armsrc Makefile and removed from Makefile.common.

The bootrom which only include Makefile.common cannot embed the version information.

Fl0-0, 2018-08-06:
Two solutions: move the version.c in common directory and generate it in the common Makefile or add the generation of version.c in the bootrom Makefile:

```
diff --git a/bootrom/Makefile b/bootrom/Makefile
index 59c22aa..c0d8543 100644
--- a/bootrom/Makefile
+++ b/bootrom/Makefile
@@ -10,7 +10,7 @@
 ARMSRC = 
 THUMBSRC = cmd.c usb_cdc.c bootrom.c
 ASMSRC = ram-reset.s flash-reset.s
-VERSIONSRC =
+VERSIONSRC = version.c
 
 ## There is a strange bug with the linker: Sometimes it will not emit the glue to call
 ## BootROM from ARM mode. The symbol is emitted, but the section will be filled with
@@ -22,6 +22,11 @@ VERSIONSRC =
 # stdint.h provided locally until GCC 4.5 becomes C99 compliant
 APP_CFLAGS = -I.
 
+# version.c should be remade on every compilation
+.PHONY: version.c
+version.c: default_version.c
+       perl ../tools/mkversion.pl .. > $@ || $(COPY) $^ $@ 
+
 # Do not move this inclusion before the definition of {THUMB,ASM,ARM}SRC
 include ../common/Makefile.common
```
Issue fixed:

```
rox/RFID mark3 RFID instrument          
bootrom: master/v2.2.0-894-gdfdca20-dirty-suspect 2018-08-06 12:35:26
os: master/v2.2.0-894-gdfdca20-dirty-suspect 2018-08-06 12:35:27
fpga_lf.bit built for 2s30vq100 on 2015/03/06 at 07:38:04
fpga_hf.bit built for 2s30vq100 on 2017/10/27 at 08:30:59
```

iceman1001, 2018-08-06:
In iceman fork I added the mkversion.pl call to the bootrom makefile.

Fl0-0, 2018-08-06:
OK Icemann !

See PR #643

marshmellow42, 2018-08-06:
btw @pwpiwi, probably should add fpga_version_info.c to gitignore?

closed 2018-08-07 by Fl0-0