Proxmark/proxmark3 #644: rename udev rule file

blshkv, 2018-08-06:
https://github.com/Proxmark/proxmark3/blob/master/driver/77-mm-usb-device-blacklist.rules

The udev rule file was probably copied from modemmanager (mm).
This file cases a collision with mm if it is getting installed. A user can overwrite an important rule accidentally.

marshmellow42, 2018-08-08:
This file was intended to be a modem manager blacklist rule.  If you read the contents it tells you how to implement it.  Specifically stating to add contents instead of replacing if file already exists.

However, it is my understanding that this process is not necessary anymore anyway, due to serial port identifier changes...

closed 2018-10-02 by pwpiwi

blshkv, 2018-10-02:
so what is the conclusion? Have you deleted the rule file?

pwpiwi, 2018-10-02:
No, because some users might still need it (at least some forum users still report issue with modem manager). Wasn't @marshmellow42 clear enough? It is not a good idea to simply copy it.
```
# copy this file to /etc/udev/rules.d (or add the entry to the end of an existing file)
# and restart udev hotplug:
#
#  'sudo udevadm control --reload-rules'
#
```

blshkv, 2018-10-03:
Ol, noted. Thanks!