Proxmark/proxmark3 #647: stuck in reboot-loop

magic-heinz, 2018-08-09:
My Proxmark3 RDV got stuck in an infinite reboot-loop while flashing the newest firmware.
It is still looping if i only connect it to power (so it can´t be a udev thing)

When i try to flash the bootrom it says something like this:

Waiting for Proxmark to appear on /dev/ttyACM0
Found
Sending bytes to proxmark failed
Waiting for a response from the proxmark...
You can cancel this operation by pressing the pm3 button

Then it doesn´t do anything until a few seconds later it clicks and reboots.
Also the LEDs A and C are allways on.

What can i do?
Thanks for helping!

marshmellow42, 2018-08-09:
Are you holding the pm3 button?

magic-heinz, 2018-08-09:
Yes I am

Update: i can flash a bootrom from 2 years ago without errors but it doesn´t fix the constant restarting. Also with every restart it vanishes from lsusb and only reappears when i physically disconnect it and reconnect it whilst holding the pm3 button.

magic-heinz, 2018-08-09:
When trying to flash the firmware it starts flashing but gets stuck halfway.... and restarts with a click

magic-heinz, 2018-08-09:
also just to add to the picture: I´m working on native kali

iceman1001, 2018-08-09:
dunno anything about Kali.

magic-heinz, 2018-08-09:
I think something messed up the firmware install at a very bad moment and now i can´t re-flash.
Is there any way to fix the firmware without additional equipment or do i have to get a jtag device?

iceman1001, 2018-08-09:
if it loops when you connect the usb,  sounds like a udev thingy,  but for Kali?

magic-heinz, 2018-08-09:
it allways loops, even if i just connect it to a micro usb charger

iceman1001, 2018-08-09:
did you try  unplug / press btn / keep btn pressed / plug in / flash /  release btn?

magic-heinz, 2018-08-09:
yes, and i was able to flash an old bootrom but the fullimage gets stuck every time

magic-heinz, 2018-08-10:
I finally got it working again!

I used the Easy Flasher from exploitagency for windows and it fixed it!

Thanks for the support!

closed 2018-08-10 by magic-heinz