Proxmark/proxmark3 #650: pm3 Rdv 4.0 does not appear as serial device on macOS

samyk, 2018-08-12:
Hi!

pm3 Rdv 4.0 doesn't appear as a serial device on macOS (10.13.6). I suspect another kernel extension is taking the device over. If I run Linux through Parallels and attach the USB device to the Linux VM, then the serial device does appear and is usable on Linux. 

I've attempted unloading various USB/HID related kext's in the past to resolve this without success. Will investigate more when time permits.

```
$ ls /dev/tty.*
/dev/tty.Bluetooth-Incoming-Port   /dev/tty.BoseQuietComfort35-SPPD   /dev/tty.BoseQuietComfort35-SPPD-1 /dev/tty.LightBlue-Bean

$ system_profiler SPUSBDataType # connected usb devices
...
        PM3 Device:

          Product ID: 0x4b8f
          Vendor ID: 0x9ac4
          Version: 1.00
          Serial Number: 888
          Speed: Up to 12 Mb/sec
          Manufacturer: proxmark.org
          Location ID: 0x14200000 / 8
          Current Available (mA): 500
          Current Required (mA): 500
          Extra Operating Current (mA): 0
...

$ sudo kextstat
Index Refs Address            Size       Wired      Name (Version) UUID <Linked Against>
    1  136 0xffffff7f80afc000 0x9e30     0x9e30     com.apple.kpi.bsd (17.7.0) 8CF59A67-A723-4FB8-A127-09A4B32E1874
    2   10 0xffffff7f80fd2000 0x3990     0x3990     com.apple.kpi.dsep (17.7.0) E085363B-71AD-4ED8-A63E-3EEC4F643ED9
    3  169 0xffffff7f80a11000 0x21150    0x21150    com.apple.kpi.iokit (17.7.0) 695D2CC5-AB90-4CD4-9862-2C50A3A989B8
    4  177 0xffffff7f80a33000 0xd430     0xd430     com.apple.kpi.libkern (17.7.0) 26F1FC40-6578-4254-AF12-3D397814B5CB
    5  154 0xffffff7f80a00000 0x3f60     0x3f60     com.apple.kpi.mach (17.7.0) 05B0DB8B-BF52-4A99-B18A-BE623583B448
    6   85 0xffffff7f80a04000 0xce70     0xce70     com.apple.kpi.private (17.7.0) 409868DB-B2D5-4249-A98A-3C2BC6E7A754
    7   92 0xffffff7f80ba5000 0x5ea0     0x5ea0     com.apple.kpi.unsupported (17.7.0) 37241BAB-747F-4D47-BBD5-C01F3577D845
    8    1 0xffffff7f80a41000 0x10000    0x10000    com.apple.kec.Libm (1) BC3F7DA4-03EA-30F7-B44A-62C249D51C10 <4>
    9    8 0xffffff7f814b2000 0xb0000    0xb0000    com.apple.kec.corecrypto (1.0) B081B8C1-1DFF-342F-8DF2-C3AA925ECA3A <7 6 5 4 3 1>
   10    0 0xffffff7f82f67000 0xf000     0xf000     com.apple.kec.pthread (1) E64F7A49-CBF0-3251-9F02-3655E3B3DD31 <7 6 5 4 3 1>
   11   31 0xffffff7f80e39000 0x9000     0x9000     com.apple.iokit.IOACPIFamily (1.4) 95DA39BB-7C39-3742-A2E5-86C555E21D67 <7 6 4 3>
   12   46 0xffffff7f80bab000 0x35000    0x35000    com.apple.iokit.IOPCIFamily (2.9) AEA4C7F5-CCB0-338B-B471-CF28A9792522 <7 6 5 4 3>
   13    9 0xffffff7f80e42000 0x1d000    0x1d000    com.apple.driver.AppleSMC (3.1.9) 2D1E4A3F-A19E-3F12-878D-807CB79CD55F <12 11 7 6 5 4 3 1>
   14    2 0xffffff7f83027000 0x9c000    0x9c000    com.apple.driver.AppleACPIPlatform (6.1) B0282791-F78C-3A3D-A829-629AD847C7EA <13 12 11 7 6 5 4 3 1>
   15   15 0xffffff7f8172f000 0x30000    0x30000    com.apple.iokit.IONetworkingFamily (3.4) 5A99D802-B477-3D83-9F76-304605479B9F <7 6 5 4 3 1>
   16    2 0xffffff7f8176d000 0x2d000    0x2d000    com.apple.iokit.IOTimeSyncFamily (680.15) 6DD2AAD0-FC4B-32EC-B0FD-AC55A727A789 <15 5 4 3 1>
   17    7 0xffffff7f8193b000 0x7000     0x7000     com.apple.iokit.IOReportFamily (31) D2F2FBDF-4EE4-38BA-99F5-B699F886F413 <5 4 3>
   18    4 0xffffff7f83290000 0x7000     0x7000     com.apple.driver.IOSlaveProcessor (1) 582BDE04-9EB3-3A87-8842-4FF06D1F72C6 <4 3>
   19    3 0xffffff7f836e8000 0x18000    0x18000    com.apple.driver.AppleSEPManager (1.0.1) 52D654B2-BC1C-3DB9-895A-8EAE7993B17B <18 12 9 7 6 5 4 3 1>
   20    1 0xffffff7f80e61000 0x8000     0x8000     com.apple.driver.AppleBusPowerController (1.0) 79B78C02-CBE8-3C9B-A19D-D8AB5875A429 <13 11 6 5 4 3>
   21    8 0xffffff7f80e69000 0x8000     0x8000     com.apple.driver.usb.AppleUSBCommon (1.0) E7111D99-B893-3F96-9741-71C88BC6F66B <5 4 3 1>
   22   27 0xffffff7f80e71000 0x79000    0x79000    com.apple.iokit.IOUSBHostFamily (1.2) CE61D704-9BFF-308E-BC1C-56FF071DBB74 <21 20 7 6 5 4 3 1>
   23    3 0xffffff7f82b6a000 0x1d000    0x1d000    com.apple.driver.KernelRelayHost (1) 49218104-0F45-3612-AF32-76BE251F5B29 <22 5 4 3>
   24    1 0xffffff7f83715000 0x3c000    0x3c000    com.apple.driver.AppleCredentialManager (1.0) 56F9E63D-81D4-35B8-83E9-E4BD2202AAB2 <23 19 18 9 7 6 5 4 3 1>
   25   14 0xffffff7f80c26000 0x28000    0x28000    com.apple.iokit.IOStorageFamily (2.1) F27A8A2A-6662-3608-83BD-415037509E01 <7 6 5 4 3 1>
   26    7 0xffffff7f80b62000 0x2b000    0x2b000    com.apple.iokit.IOSCSIArchitectureModelFamily (404.30.2) 4734E6C9-B881-32C8-BE1F-4D9848A52895 <6 5 4 3 1>
   27    3 0xffffff7f82f92000 0x19000    0x19000    com.apple.iokit.IOSCSIBlockCommandsDevice (404.30.2) 34091931-8856-37D2-B4BB-82F28AC66162 <26 25 6 5 4 3 1>
   28    3 0xffffff7f82faf000 0x2e000    0x2e000    com.apple.iokit.IOUSBMassStorageDriver (140.70.2) 91925DA5-1D2A-3444-A53B-1432B6363BA7 <26 25 22 6 5 4 3 1>
   29    5 0xffffff7f81562000 0x15000    0x15000    com.apple.driver.AppleMobileFileIntegrity (1.0.5) 4FAE8181-B0B4-3A30-B0F3-77F48D2486E5 <9 7 6 5 4 3 2 1>
   30    1 0xffffff7f82fe4000 0x12000    0x12000    com.apple.driver.AppleUSBTDM (439.70.3) 1CE56483-ADE1-3C2B-ABF6-2F7BA1E3B14E <29 28 27 26 25 22 7 6 5 4 3 1>
   31    0 0xffffff7f83751000 0x64000    0x64000    com.apple.driver.AppleKeyStore (2) D3957B25-2814-309B-B9AF-780544562883 <30 29 24 23 19 18 9 7 6 5 4 3 1>
   32    7 0xffffff7f80ef2000 0x9b000    0x9b000    com.apple.iokit.IOUSBFamily (900.4.1) 2837DD62-F2B0-3357-A164-D44FEF6810E0 <22 21 12 7 5 4 3 1>
   33    0 0xffffff7f837fa000 0x16000    0x16000    com.virtualhere.vhhcd (1.0.97) 58ECD881-BFD6-3477-B1F2-9C91F0A3D640 <32 5 4 3>
   34    2 0xffffff7f81e33000 0xc000     0xc000     com.apple.driver.AppleEffaceableStorage (1.0) CA2AB641-FF39-316A-BA4F-5AF561C66AC7 <7 5 4 3 1>
   35    0 0xffffff7f83837000 0xc000     0xc000     com.apple.driver.AppleFDEKeyStore (28.30) 691E35C5-E9F8-33D8-8492-9731EC9D2A6D <34 9 7 6 5 4 3 1>
   36    0 0xffffff7f83c52000 0x19000    0x19000    com.apple.driver.DiskImages (480.60.1) 7D6AED48-938B-315C-A581-356240778C5F <25 7 6 5 4 3 1>
   37    2 0xffffff7f81ab3000 0x5000     0x5000     com.apple.kext.AppleMatch (1.0.0d1) 2D2996CA-A15E-39E6-9D6B-2BE6DF131972 <4 1>
   38    1 0xffffff7f81ab8000 0x21000    0x21000    com.apple.security.sandbox (300.0) BDFF700A-6746-3643-A0A1-852628695B04 <37 29 25 7 6 5 4 3 2 1>
   39    0 0xffffff7f81ed2000 0x9000     0x9000     com.apple.security.quarantine (3) 80C4548A-999E-3F96-85B0-6C6F95A542A7 <38 37 7 6 5 4 2 1>
   40    0 0xffffff7f83c4f000 0x2000     0x2000     com.apple.security.TMSafetyNet (8) 5B78C369-AA3C-326F-8AE3-C200955DD9A7 <7 6 5 4 2 1>
   41    0 0xffffff7f83d23000 0x8000     0x8000     com.apple.nke.applicationfirewall (183) CBC1954F-5BBE-3FD5-B1BD-24E999226FEF <7 6 5 4 3 1>
   42    0 0xffffff7f82f29000 0x19000    0x19000    com.apple.driver.AppleUSBLegacyHub (900.4.1) F34036CF-9F86-3025-8C0B-18273A15F4A3 <32 5 4 3 1>
[... 118 lines cut ...]
  161    0 0xffffff7f81e54000 0x12000    0x12000    com.apple.driver.AppleGraphicsDevicePolicy (3.20.13) 88C6EFDE-DBA9-3FD2-892F-AF527A14EAA9 <110 109 102 12 11 7 6 5 4 3 1>
  162    0 0xffffff7f83bef000 0x5000     0x5000     com.apple.driver.AudioAUUC (1.70) 71FC1A81-3747-3E5A-857E-8C657B2D2BD9 <122 102 12 11 7 5 4 3 1>
  163    2 0xffffff7f837b5000 0xe000     0xe000     com.apple.iokit.IOSerialFamily (11) 6FE74518-ECD9-32D6-88BE-8F5BCAC911D4 <7 6 5 4 3 1>
  164    0 0xffffff7f839f7000 0xa000     0xa000     com.apple.iokit.IOBluetoothSerialManager (6.0.7f10) CE7E273D-4839-3223-8608-FCBB4C708435 <163 7 5 4 3 1>
  165    0 0xffffff7f81edc000 0x11000    0x11000    com.apple.driver.usb.cdc.ncm (5.0.0) 00AE6600-E9E4-3FF9-99ED-E5CD8D8F2C3F <84 83 82 22 17 15 6 5 4 3 1>
  166    0 0xffffff7f83d64000 0xd000     0xd000     com.apple.driver.usb.cdc.ecm (5.0.0) DD7CC27B-1AFC-3180-8B78-36E7F7821A59 <84 83 54 22 15 5 4 3 1>
  167    0 0xffffff7f837c6000 0xf000     0xf000     com.apple.driver.usb.cdc.acm (5.0.0) 1CC8F6CF-6B2A-38D3-BE90-A767515F2BDE <163 84 83 82 22 5 4 3 1>
  168    0 0xffffff7f83895000 0x5000     0x5000     com.apple.driver.AppleHWSensor (1.9.5d0) D1A8CE3E-2398-34FD-BA1A-F37AEF1A334A <5 4 3>
  169    0 0xffffff7f80fcf000 0x2000     0x2000     com.nvidia.CUDA (1.1.0) E13478CB-B251-3C0A-86E9-A6B56F528FE8 <4 1>
  170    1 0xffffff7f83158000 0x5000     0x5000     com.apple.kext.triggers (1.0) 00226670-5F9E-31BC-B59C-A0B2FE6D7797 <7 6 5 4 3 1>
[... cut ...]
  181    0 0xffffff7f81e96000 0x26000    0x26000    com.apple.driver.AppleFWOHCI (5.5.9) B774B8EA-3452-3A10-A98E-62C53B1BD4D1 <180 12 7 6 5 4 3 1>
  182    0 0xffffff7f83462000 0x44000    0x44000    com.apple.iokit.AppleBCM5701Ethernet (10.3.1) 2741164A-412A-3562-AA38-CAA0290AD2E8 <147 60 15 12 11 6 5 4 3 1>
  183    0 0xffffff7f837ea000 0xe000     0xe000     com.apple.iokit.IOFireWireIP (2.2.9) AD37305B-E664-3DFB-961B-FB9260E508E9 <180 15 5 4 3 1>
  184    0 0xffffff7f8297c000 0x61000    0x61000    com.apple.driver.AppleUSBAudio (312.6) 5365B95C-F273-3472-985D-0125550E9392 <122 54 22 12 5 4 3 1>
```

iceman1001, 2018-08-13:
Don't know much of OSX but the wiki section for MacOS has some info about kext etc.
https://github.com/Proxmark/proxmark3/wiki/MacOS

@TomHarkness  also mentioned something about very cheap USB-C --> USB adaptor could spook things.

TomHarkness, 2018-08-13:
Try:

ls /dev/cu.u* 

You'll likely find the device there. I'm running multiple versions of os X with minimal hassle.

$ ls /dev/cu.*
/dev/cu.Bluetooth-Incoming-Port  /dev/cu.raspberrypi-SerialPort   **/dev/cu.usbmodem881**
$

after some initial issues with kernel extensions I don't even bother anymore and have had no issue at all. If you installed the kext, find and remove it. Usually: sudo rm -rf /System/Library/Extensions/Proxmark3.kext

Later versions of OS X seem to have much better serial handling.

samyk, 2018-08-13:
There should always be an associated tty.* for a cu.* - I validated that there is no cu.* as well. I'll try the kext tips.

samyk, 2018-08-13:
Same issue after following the kext suggestions on the pm3 docs. I think I'll need to individually remove kexts and test.

pwpiwi, 2018-08-14:
> Serial Number: 888

You are using Iceman's firmware. Do you have the same issue with the official firmware (this repository)?

iceman1001, 2018-08-16:
@samyk   would you mind testing the install.sh  found here:
https://github.com/RfidResearchGroup/proxmark3/blob/master/install.sh

it got updated for MacOS..

TomHarkness, 2018-08-17:
Same install.sh will work for this repo - haven't had a chance to do a PR yet.

samyk, 2018-08-18:
Install was successful though pm3 still doesn't show up at a serial device.

TomHarkness, 2018-08-18:
Can you post the output of "ls /dev/*"   ?

TomHarkness, 2018-08-19:
Looking through your kext - you may have some cdc conflict. 

 84    3 0xffffff7f81b1b000 0x9000     0x9000     com.apple.driver.usb.cdc (5.0.0) 9DAB9F8C-B608-3FC8-B09C-5F0BF5F1BD1C <83 82 22 5 4 3 1>

165    0 0xffffff7f81edc000 0x11000    0x11000    com.apple.driver.usb.cdc.ncm (5.0.0) 00AE6600-E9E4-3FF9-99ED-E5CD8D8F2C3F <84 83 82 22 17 15 6 5 4 3 1>
  166    0 0xffffff7f83d64000 0xd000     0xd000     com.apple.driver.usb.cdc.ecm (5.0.0) DD7CC27B-1AFC-3180-8B78-36E7F7821A59 <84 83 54 22 15 5 4 3 1>
  167    0 0xffffff7f837c6000 0xf000     0xf000     com.apple.driver.usb.cdc.acm (5.0.0) 1CC8F6CF-6B2A-38D3-BE90-A767515F2BDE <163 84 83 82 22 5 4 3 1>

samyk, 2018-08-19:
I have another PM3 that *does* show up as a serial device (`/dev/tty.usbmodem1421`), though it uses a different PID/VID (`0x504d:0x2d2d`). I also have a 3rd proxmark that did show up until I upgraded bootloader/firmware, so I'm sure one of the kext's is just taking over the new PID/VID.

```
tigerblood:/Users/samy/Code/proxmark/test/proxmark3$ ls /System/Library/Extensions/
360Controller.kext                        AppleThunderboltIP.kext
ALF.kext                                  AppleThunderboltNHI.kext
AMD10000Controller.kext                   AppleThunderboltPCIAdapters.kext
AMD2400Controller.kext                    AppleThunderboltUTDM.kext
AMD2600Controller.kext                    AppleTopCase.kext
AMD3800Controller.kext                    AppleTyMCEDriver.kext
AMD4600Controller.kext                    AppleUSBACM.kext
AMD4800Controller.kext                    AppleUSBAudio.kext
AMD5000Controller.kext                    AppleUSBCDC.kext
AMD6000Controller.kext                    AppleUSBCommon.kext
AMD7000Controller.kext                    AppleUSBDMM.kext
AMD8000Controller.kext                    AppleUSBDisplays.kext
AMD9000Controller.kext                    AppleUSBECM.kext
AMD9500Controller.kext                    AppleUSBEEM.kext
AMDFramebuffer.kext                       AppleUSBEthernet.kext
AMDLegacyFramebuffer.kext                 AppleUSBEthernetHost.kext
AMDLegacySupport.kext                     AppleUSBFTDI.kext
AMDMTLBronzeDriver.bundle                 AppleUSBMultitouch.kext
AMDRadeonVADriver.bundle                  AppleUSBNCM.kext
AMDRadeonVADriver2.bundle                 AppleUSBNetworking.kext
AMDRadeonX3000.kext                       AppleUSBSerial.kext
AMDRadeonX3000GLDriver.bundle             AppleUSBTopCase.kext
AMDRadeonX4000.kext                       AppleUSBWCM.kext
AMDRadeonX4000GLDriver.bundle             AppleUSBiBridge.kext
AMDRadeonX4000HWServices.kext             AppleUpstreamUserClient.kext
AMDRadeonX5000.kext                       AppleVADriver.bundle
AMDRadeonX5000GLDriver.bundle             AppleWWANAutoEject.kext
AMDRadeonX5000HWServices.kext             AppleXsanScheme.kext
AMDRadeonX5000MTLDriver.bundle            Apple_iSight.kext
AMDRadeonX5000Shared.bundle               AudioAUUC.kext
AMDShared.bundle                          BCMWLANFirmware4364_Datastore.kext
AMDSupport.kext                           BCMWLANFirmware4364_Hashstore.kext
ATIRadeonX2000.kext                       BootCache.kext
ATIRadeonX2000GA.plugin                   BridgeAudioCommunication.kext
ATIRadeonX2000GLDriver.bundle             BridgeAudioController.kext
ATIRadeonX2000VADriver.bundle             CellPhoneHelper.kext
AirPortBrcmNIC-MFG.kext                   CoreStorage.kext
Apple16X50Serial.kext                     DSACL.ppp
AppleACPIPlatform.kext                    DSAuth.ppp
AppleAHCIPort.kext                        DUMMY.kext
AppleAPIC.kext                            DVFamily.bundle
AppleAVEBridge.kext                       Dont Steal Mac OS X.kext
AppleActuatorDriver.kext                  EAP-KRB.ppp
AppleBCMWLANBusInterfacePCIe.kext         EAP-RSA.ppp
AppleBCMWLANCore.kext                     EAP-TLS.ppp
AppleBCMWLANCoreV3_Datastore.kext         EnergiaFTDIDrivers2.2.18.pkg
AppleBacklight.kext                       FTDI Installer.pmdoc
AppleBacklightExpert.kext                 FTDIUSBSerialDriver.kext
AppleBluetoothMultitouch.kext             GeForce.kext
AppleBluetoothRemote.kext                 GeForceAIRPlugin.bundle
AppleBusPowerController.kext              GeForceGA.bundle
AppleCameraInterface.kext                 GeForceGLDriver.bundle
AppleCredentialManager.kext               GeForceMTLDriver.bundle
AppleEFIRuntime.kext                      GeForceTesla.kext
AppleEffaceableNOR.kext                   GeForceTeslaGLDriver.bundle
AppleEffaceableStorage.kext               GeForceTeslaVADriver.bundle
AppleEmbeddedOSSupportHost.kext           GeForceVADriver.bundle
AppleFDEKeyStore.kext                     HFS.kext
AppleFIVRDriver.kext                      HFSEncodings.kext
AppleFSCompressionTypeDataless.kext       HoRNDIS.kext
AppleFSCompressionTypeZlib.kext           IO80211Family.kext
AppleFWAudio.kext                         IO80211FamilyV2.kext
AppleFileSystemDriver.kext                IOACPIFamily.kext
AppleGFXHDA.kext                          IOAHCIFamily.kext
AppleGraphicsControl.kext                 IOATAFamily.kext
AppleGraphicsPowerManagement.kext         IOAVBFamily.kext
[... 316 lines cut ...]
bpf47                         ptyra                         ttyvb
bpf48                         ptyrb                         ttyvc
bpf49                         ptyrc                         ttyvd
bpf5                          ptyrd                         ttyve
bpf50                         ptyre                         ttyvf
bpf51                         ptyrf                         ttyw0
bpf52                         ptys0                         ttyw1
bpf53                         ptys1                         ttyw2
bpf54                         ptys2                         ttyw3
bpf55                         ptys3                         ttyw4
bpf56                         ptys4                         ttyw5
bpf57                         ptys5                         ttyw6
bpf58                         ptys6                         ttyw7
bpf59                         ptys7                         ttyw8
[... cut ...]
bpf61                         ptysa                         ttywb
bpf62                         ptysb                         ttywc
bpf63                         ptysc                         ttywd
bpf64                         ptysd                         ttywe
bpf65                         ptyse                         ttywf
bpf66                         ptysf                         urandom
bpf67                         ptyt0                         xcpm
bpf68                         ptyt1                         zero
```

TomHarkness, 2018-08-20:
This is strange. Does the PM# show up under /dev if you hold the button down while plugging in - and keep it held while checking ?

samyk, 2018-08-26:
I removed some old kext's -- just unloading them didn't allow the PM3 to show up, but once I rebooted, PM3 became available! I also upgraded Parallels during this time to v14 -- possibly related but unsure.

Here's my current kextstat in case it helps someone else:
```
$ sudo kextstat
Index Refs Address            Size       Wired      Name (Version) UUID <Linked Against>
    1  131 0xffffff7f80afc000 0x9e30     0x9e30     com.apple.kpi.bsd (17.7.0) 8CF59A67-A723-4FB8-A127-09A4B32E1874
    2   10 0xffffff7f80fdd000 0x3990     0x3990     com.apple.kpi.dsep (17.7.0) E085363B-71AD-4ED8-A63E-3EEC4F643ED9
    3  153 0xffffff7f80a11000 0x21150    0x21150    com.apple.kpi.iokit (17.7.0) 695D2CC5-AB90-4CD4-9862-2C50A3A989B8
    4  161 0xffffff7f80a33000 0xd430     0xd430     com.apple.kpi.libkern (17.7.0) 26F1FC40-6578-4254-AF12-3D397814B5CB
    5  143 0xffffff7f80a00000 0x3f60     0x3f60     com.apple.kpi.mach (17.7.0) 05B0DB8B-BF52-4A99-B18A-BE623583B448
    6   83 0xffffff7f80a04000 0xce70     0xce70     com.apple.kpi.private (17.7.0) 409868DB-B2D5-4249-A98A-3C2BC6E7A754
    7   90 0xffffff7f80ba5000 0x5ea0     0x5ea0     com.apple.kpi.unsupported (17.7.0) 37241BAB-747F-4D47-BBD5-C01F3577D845
    8    1 0xffffff7f80a41000 0x10000    0x10000    com.apple.kec.Libm (1) BC3F7DA4-03EA-30F7-B44A-62C249D51C10 <4>
    9    9 0xffffff7f814bd000 0xb0000    0xb0000    com.apple.kec.corecrypto (1.0) B081B8C1-1DFF-342F-8DF2-C3AA925ECA3A <7 6 5 4 3 1>
   10    0 0xffffff7f82f7f000 0xf000     0xf000     com.apple.kec.pthread (1) E64F7A49-CBF0-3251-9F02-3655E3B3DD31 <7 6 5 4 3 1>
   11   28 0xffffff7f80e39000 0x9000     0x9000     com.apple.iokit.IOACPIFamily (1.4) 95DA39BB-7C39-3742-A2E5-86C555E21D67 <7 6 4 3>
   12   41 0xffffff7f80bab000 0x35000    0x35000    com.apple.iokit.IOPCIFamily (2.9) AEA4C7F5-CCB0-338B-B471-CF28A9792522 <7 6 5 4 3>
   13    8 0xffffff7f80e42000 0x1d000    0x1d000    com.apple.driver.AppleSMC (3.1.9) 2D1E4A3F-A19E-3F12-878D-807CB79CD55F <12 11 7 6 5 4 3 1>
   14    2 0xffffff7f8303f000 0x9c000    0x9c000    com.apple.driver.AppleACPIPlatform (6.1) B0282791-F78C-3A3D-A829-629AD847C7EA <13 12 11 7 6 5 4 3 1>
   15   13 0xffffff7f8173a000 0x30000    0x30000    com.apple.iokit.IONetworkingFamily (3.4) 5A99D802-B477-3D83-9F76-304605479B9F <7 6 5 4 3 1>
   16    2 0xffffff7f81778000 0x2d000    0x2d000    com.apple.iokit.IOTimeSyncFamily (680.15) 6DD2AAD0-FC4B-32EC-B0FD-AC55A727A789 <15 5 4 3 1>
   17    6 0xffffff7f81946000 0x7000     0x7000     com.apple.iokit.IOReportFamily (31) D2F2FBDF-4EE4-38BA-99F5-B699F886F413 <5 4 3>
   18    4 0xffffff7f832a8000 0x7000     0x7000     com.apple.driver.IOSlaveProcessor (1) 582BDE04-9EB3-3A87-8842-4FF06D1F72C6 <4 3>
   19    3 0xffffff7f83700000 0x18000    0x18000    com.apple.driver.AppleSEPManager (1.0.1) 52D654B2-BC1C-3DB9-895A-8EAE7993B17B <18 12 9 7 6 5 4 3 1>
   20    1 0xffffff7f80e61000 0x8000     0x8000     com.apple.driver.AppleBusPowerController (1.0) 79B78C02-CBE8-3C9B-A19D-D8AB5875A429 <13 11 6 5 4 3>
   21    8 0xffffff7f80e69000 0x8000     0x8000     com.apple.driver.usb.AppleUSBCommon (1.0) E7111D99-B893-3F96-9741-71C88BC6F66B <5 4 3 1>
   22   24 0xffffff7f80e71000 0x79000    0x79000    com.apple.iokit.IOUSBHostFamily (1.2) CE61D704-9BFF-308E-BC1C-56FF071DBB74 <21 20 7 6 5 4 3 1>
   23    3 0xffffff7f82b82000 0x1d000    0x1d000    com.apple.driver.KernelRelayHost (1) 49218104-0F45-3612-AF32-76BE251F5B29 <22 5 4 3>
   24    1 0xffffff7f8372d000 0x3c000    0x3c000    com.apple.driver.AppleCredentialManager (1.0) 56F9E63D-81D4-35B8-83E9-E4BD2202AAB2 <23 19 18 9 7 6 5 4 3 1>
   25   10 0xffffff7f80c26000 0x28000    0x28000    com.apple.iokit.IOStorageFamily (2.1) F27A8A2A-6662-3608-83BD-415037509E01 <7 6 5 4 3 1>
   26    4 0xffffff7f80b62000 0x2b000    0x2b000    com.apple.iokit.IOSCSIArchitectureModelFamily (404.30.2) 4734E6C9-B881-32C8-BE1F-4D9848A52895 <6 5 4 3 1>
   27    2 0xffffff7f82faa000 0x19000    0x19000    com.apple.iokit.IOSCSIBlockCommandsDevice (404.30.2) 34091931-8856-37D2-B4BB-82F28AC66162 <26 25 6 5 4 3 1>
   28    2 0xffffff7f82fc7000 0x2e000    0x2e000    com.apple.iokit.IOUSBMassStorageDriver (140.70.2) 91925DA5-1D2A-3444-A53B-1432B6363BA7 <26 25 22 6 5 4 3 1>
   29    5 0xffffff7f8156d000 0x15000    0x15000    com.apple.driver.AppleMobileFileIntegrity (1.0.5) 4FAE8181-B0B4-3A30-B0F3-77F48D2486E5 <9 7 6 5 4 3 2 1>
   30    1 0xffffff7f82ffc000 0x12000    0x12000    com.apple.driver.AppleUSBTDM (439.70.3) 1CE56483-ADE1-3C2B-ABF6-2F7BA1E3B14E <29 28 27 26 25 22 7 6 5 4 3 1>
   31    0 0xffffff7f83769000 0x64000    0x64000    com.apple.driver.AppleKeyStore (2) D3957B25-2814-309B-B9AF-780544562883 <30 29 24 23 19 18 9 7 6 5 4 3 1>
   32    5 0xffffff7f80ef2000 0x9b000    0x9b000    com.apple.iokit.IOUSBFamily (900.4.1) 2837DD62-F2B0-3357-A164-D44FEF6810E0 <22 21 12 7 5 4 3 1>
   33    0 0xffffff7f83812000 0x16000    0x16000    com.virtualhere.vhhcd (1.0.97) 58ECD881-BFD6-3477-B1F2-9C91F0A3D640 <32 5 4 3>
   34    2 0xffffff7f81e3e000 0xc000     0xc000     com.apple.driver.AppleEffaceableStorage (1.0) CA2AB641-FF39-316A-BA4F-5AF561C66AC7 <7 5 4 3 1>
   35    0 0xffffff7f8384f000 0xc000     0xc000     com.apple.driver.AppleFDEKeyStore (28.30) 691E35C5-E9F8-33D8-8492-9731EC9D2A6D <34 9 7 6 5 4 3 1>
   36    0 0xffffff7f83c76000 0x19000    0x19000    com.apple.driver.DiskImages (480.60.1) 7D6AED48-938B-315C-A581-356240778C5F <25 7 6 5 4 3 1>
[... cut ...]
  214    0 0xffffff7f83fb0000 0xc000     0xc000     com.apple.driver.usb.cdc.ecm (5.0.0) DD7CC27B-1AFC-3180-8B78-36E7F7821A59 <209 83 54 22 15 5 4 3 1>
  215    0 0xffffff7f83fbc000 0x17000    0x17000    com.paragon-software.lvm-for-mac (1) 603C496E-9F62-3FE9-981F-D5D85E9E411D <25 5 4 3 1>
  216    0 0xffffff7f83fd3000 0x8000     0x8000     com.apple.driver.AppleXsanScheme (3) 79D5E92F-789E-3C37-BE0E-7D1EAD697DD9 <25 5 4 3 1>
  217    0 0xffffff7f83fdb000 0xf000     0xf000     com.apple.filesystems.msdosfs (1.10) 8FE45584-8696-3CB8-BF53-6A105E3256D7 <7 5 4 1>
```

closed 2018-08-26 by samyk