Proxmark/proxmark3 #657: Issue with Flashing - does not work - devices seems bricked

Axlbuld, 2018-08-22:
Hi,

I just pulled the latest version and tried to flash it onto a proxmark3 easy as well as my proxmark3 RDV2 (before trying the RDV4). However, I have severe issues:
- flashing the bootrom works
- flashing the rest fails after one line of dots. Afterwards, the proxmark3 won't boot. It does not show up on ttyACM0 anymore.

I tried the master as well as iceman fork. no success.
I have found some very old very small special firmware (LF-standalone). That was mostly the only I could flash successfully afterwards. However, I had to hold the Button for more than 10 Seconds to get it into DFU mode and keep it flashed until the process has finished. I have to follow this procedure for bootrom and fullimage.

Now, I was able to flash an older icsom-fork image onto the RDV2 and get it to work (however, LED CHR is constantly flashing red).
I am using kali rolling 2.0, just updated and upgraded.

Any idea? Where did I fail?

Axlbuld, 2018-08-23:
Found the issue - thanks fto a friend. It stil is the ModemManager on Kali. Eventhough I have created the blacklisting entry suggested, it still interferes and resets the device (the old PM3 RDV2 is clicking every 20 seconds or so).
ModemManager killed, works fine.

closed 2018-08-23 by Axlbuld

iceman1001, 2018-08-23:
And how did you kill the modemmanager?   Update Wiki for Kalli?

Axlbuld, 2018-08-23:
Sorry, forgot to write.
As I don't know what the ModemManager is used for, I left it installed. I did not kill it permanently. So, just plain commands:
_ps -aux | grep ModemManager
kill [pid]_ - on my system stable 521

I know, it is not permanent and not intelligent or sophisticated, and you have to do it everytime before use. 
_apt-get purge ModemManager_ might do a better job as it is permanent, but I don't know, what I will break with that.

Axlbuld, 2018-08-23:
@Iceman: do you update the wiki? I am quite new to the github stuff and a bit preoccupied to break stuff.
And thanks for the good work.