Proxmark/proxmark3 #663: autobuild in appveyor

merlokk, 2018-08-30:
Hi,

I look into my CI config and see strange situation:

```
do.c:60:26: warning: implicit declaration of function '_longjmp'; did you mean 'longjmp'? [-Wimplicit-function-declaration]
 #define LUAI_THROW(L,c)  _longjmp((c)->b, 1)
                          ^
ldo.c:106:5: note: in expansion of macro 'LUAI_THROW'
     LUAI_THROW(L, L->errorJmp);  /* jump to it */
     ^~~~~~~~~~
ldo.c: In function 'luaD_rawrunprotected':
ldo.c:61:30: error: too few arguments to function '_setjmp'
 #define LUAI_TRY(L,c,a)  if (_setjmp((c)->b) == 0) { a }
                              ^
ldo.c:131:3: note: in expansion of macro 'LUAI_TRY'
   LUAI_TRY(L, &lj,
   ^~~~~~~~
In file included from ldo.c:8:0:
C:/ProxSpace/msys2/mingw32/i686-w64-mingw32/include/setjmp.h:227:63: note: declared here
   int __cdecl __attribute__ ((__nothrow__,__returns_twice__)) _setjmp(jmp_buf _Buf, void *_Ctx);
                                                               ^~~~~~~
ldo.c: In function 'luaD_throw':
ldo.c:122:1: warning: 'noreturn' function does return
 }
 ^
```

Has anyone encountered this behavior?

merlokk, 2018-10-13:
looks like the same
https://github.com/cesanta/mongoose/commit/63c5658a2b3b22a654bf90ef2a3290b3f8192b7a

merlokk, 2018-10-13:
and was the same with -D_XOPEN_SOURCE=500
in #685

pwpiwi, 2018-10-13:
One is Windows, the other is Linux...

merlokk, 2018-10-13:
we build in msys2. its not linux or windows(
im checking....   maybe it works...
but if it works it will need to check in all systems...

merlokk, 2018-10-13:
not works(

merlokk, 2018-10-13:
got rid of 
```
D:\ProxSpace\pm3\liblua\luaconf.h (1 hit)
	Line 72: #define LUA_USE_ULONGJMP
```
and...
```
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o lauxlib.o lauxlib.c
bash : lauxlib.c:229:10: fatal error: sys/wait.h: No such file or directory
At line:13 char:1
 #include <sys/wait.h>
          ^~~~~~~~~~~~
compilation terminated.
```

merlokk, 2018-10-13:
looks like gcc trying to make win platform instead of linux....

merlokk, 2018-10-13:
LUA_USE_POSIX is set up....

merlokk, 2018-10-14:
ok, it looks like gcc compiles in windows with flags from linux....

pwpiwi, 2018-10-14:
Those flags "configure" the C-Library. They are independet of the OS.

Everything compiles fine in mingw, You need to find out what is different in appveyor.

merlokk, 2018-10-22:
PR will be in several hours

merlokk, 2018-10-22:
done