Proxmark/proxmark3 #664: windows 10 64bit install - invalid pgp keys - make command not working

mudge1337, 2018-08-30:
**OS** 
---- windows 10 64 bit
**problem**:  
---- returning error from starting batch file "runme64" seems to not trust keys to load GCC tar file. making it unable to use "make" command. Continued to get error "bash: make: command not found"

**response**: 
mingw-w64-x86_64-gcc-8.2.0-1- 100%[=================================================>]  28.37M  2.22MB/s    in 12s

2018-08-30 17:20:20 (2.30 MB/s) - 'mingw-w64-x86_64-gcc-8.2.0-1-any.pkg.tar.xz' saved [29747228/29747228]

FINISHED --2018-08-30 17:20:20--
Total wall clock time: 16s
Downloaded: 2 files, 29M in 13s (2.20 MB/s)
error: mingw32: key "5F92EFC1A47D45A1" is unknown
error: key "5F92EFC1A47D45A1" could not be looked up remotely
error: mingw64: key "5F92EFC1A47D45A1" is unknown
error: key "5F92EFC1A47D45A1" could not be looked up remotely
error: msys: key "5F92EFC1A47D45A1" is unknown
error: key "5F92EFC1A47D45A1" could not be looked up remotely
error: database 'mingw32' is not valid (invalid or corrupted database (PGP signature))
error: database 'mingw64' is not valid (invalid or corrupted database (PGP signature))
error: database 'msys' is not valid (invalid or corrupted database (PGP signature))
loading packages...
error: failed to prepare transaction (invalid or corrupted database)
error: mingw32: key "5F92EFC1A47D45A1" is unknown
error: key "5F92EFC1A47D45A1" could not be looked up remotely
error: mingw64: key "5F92EFC1A47D45A1" is unknown
error: key "5F92EFC1A47D45A1" could not be looked up remotely
error: msys: key "5F92EFC1A47D45A1" is unknown
error: key "5F92EFC1A47D45A1" could not be looked up remotely
error: database 'mingw32' is not valid (invalid or corrupted database (PGP signature))
error: database 'mingw64' is not valid (invalid or corrupted database (PGP signature))
error: database 'msys' is not valid (invalid or corrupted database (PGP signature))

TomHarkness, 2018-08-30:
Strange. Are you using the latest ProxSpace release zip? I have not seen this and have set up many Win 10 environments.

merlokk, 2018-08-31:
Try to use zip, not `git clone`

pwpiwi, 2018-09-06:
@mudge1337 :  if the tipp from either @TomHarkness  or @merlokk fixed your problem, then please let us know and close this issue, please.

closed 2018-10-02 by pwpiwi