Proxmark/proxmark3 #670: HF 14a sim with MiFare Desfire - wrong ATS?

Axlbuld, 2018-09-13:
Hi,

I am playing with the proxmark3 and a MiFare Desfire EV1 card. We have an application that seems to just use the UID using ISO 14443a. When I simulate with a proxmark, the card in the appliation device  (printer) reads stay's dead.
Comparing the read card with the simulation on the proxmark3 I can see that only the ATS is different: 
The card shows:  ATS : 06 75 77 81 02 80 02 F0  
The sim produces:  ATS : 04 58 80 02 13 CE   

Looking here: http://nfc-tools.org/index.php/ISO14443A it says, that the proxmark is wrong.
Card                            | ATQA | SAK| ATS                       | UID
MIFARE DESFire EV1 | 03 44 | 20    | 75  77  81  02  80 | 7 bytes

Having very limited programming skills I change the code in iso14443a.c accordingly, but then I can't read the sim with the second proxmark3. It says, card failed.
Besides, I don't understand the 06 and I guess 02 F0 are crc16, so I should not have them set up manually. I guess the crc does not calculate to 02 F0, but I did not verify so far.

Any good idea?

merlokk, 2018-09-13:
here ATR list
https://www.eftlab.co.uk/index.php/site-map/knowledge-base/171-atr-list-full
here tool for ATR parsing
https://smartcard-atr.appspot.com/

the same for ATS (it needs to put here ATS with CRC)
https://www.javacardos.com/tools/ats.html

and you can change it to anything you want...

pwpiwi, 2018-09-13:
04 and 06 are the size of the ATS (including the size byte, excluding the 2 CRC bytes at the end.

To change it in iso14443a.c you need to modify two lines:
```
uint8_t response6[] = { 0x06, 0x75, 0x77, 0x81, 0x02, 0x80, 0x00, 0x00 };
ComputeCrc14443(CRC_14443_A, response6, 6, &response6[6], &response6[7]);
```
where the 2nd line calculates the CRC for you.

pwpiwi, 2018-09-13:
BTW: the third byte (0x77) tells the reader that the card supports all bitrates (106kBaud, 212kBaud, 424kBaud) for sending and receiving. The reader may subsequently decide to go for a higher bitrate than the Proxmark can support (106kBaud only).

Axlbuld, 2018-09-13:
thanks. Works now. Great.
@merlokk: great links. Thanks.

closed 2018-09-13 by Axlbuld