Proxmark/proxmark3 #675: EMV Exec against ApplePay fails to decode some replies

drandreas, 2018-09-17:
@merlokk First of all I'm really impressed by your implementation. It gives a very detailed view of the exchanged data. Moreover the transaction kind of succeeds (The watch shows a checkmark).
But the log shows some errors. Since I don't understand all bits and bytes yet I take the liberty to only post snippets. 

I've run `hf emv exec -s -a -t -c`

I see CDCVM as expected for ApplePay
```
	--9f6c[02] 'Card Transaction Qualifiers (CTQ)':
		Byte 1 (00)
		Byte 2 (80)
			1....... - 'Consumer Device CVM Performed'
		00: 00 80                                           |..
```
and an unknown record
```
	--9f6e[04] 'Unknown ???':
		00: 23 88 00 00                                     |#...
```

Later on in the transaction I see an CVR decoding error:
```
IAD: 1f 4a 63 31 a0 00 00 00 00 10 03 02 73 00 00 00 00 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00           
	Key index:  0x4a          
	Crypto ver: 0x63(099)          
	CVR:          
	--20[1d] 'Cardholder Verification Results (CVR)':
		INVALID length!
		00: 31 a0 00 00 00 00 10 03 02 73 00 00 00 00 40 00 |1........s....@.
		10: 00 00 00 00 00 00 00 00 00 00 00 00 00          |.............
```

merlokk, 2018-09-18:
ill add this
9F6E | Form Factor Indicator

`INVALID length!` its not an error because it dont knows crypto version `99`

merlokk, 2018-09-18:
Transaction M/Chip or [x]VSDC is OK if card answers with something like that:
```
        --9f27[01] 'Cryptogram Information Data':
                AC1: ARQC (Online authorisation requested)
                00: 80                                              |.
```
Or it can return `TC` - transaction approved offline
or) if it return AAC - transaction declined by card.

if transaction type is MSD it returns Track2 eqivalent data and later terminal can check dCVV

drandreas, 2018-09-19:
I don't think the watch features MSD. It only advertises a single AID.

Thank you for adding `9F6E` and the hints / pointers. I'll investigate further and report back.

merlokk, 2018-09-19:
If you have documentation about crypto with ver 99 withouth NDA - please share) ill add decoder.

P.S. usually this part of information is available only after signing NDA and code cant be published into outsource(

drandreas, 2018-09-20:
I'm just getting into EMV. Apple Pay with Visa (thats what I tried) is supposedly Kernel 3:
https://support.apple.com/en-us/HT205747

"Contactless specification: VCPS 2.1.1" "EMV specification: Any version of Kernel 3"

merlokk, 2018-09-21:
https://www.emvco.com/wp-content/uploads/2017/05/Book_A_Architecture_and_General_Rqmts_v2_6_Final_20160422011856105.pdf
page 39. Kernel 3 is VISA

VCPS is available only gor NDA with VISA. I have an old version but it dont have crypto ver 99(

drandreas, 2018-10-07:
Since I don't see much potential for progress without breaking the crypto or obtaining NDA documents I'm closing the issue.

closed 2018-10-07 by drandreas