Proxmark/proxmark3 #685: ./proxmark3 Segmentation fault

iDoka, 2018-10-01:
I have the same behaviour on v3.0.1 and from lastest git version,
when I run `./proxmark3` with device path I get Segmentation fault:
```
$ ./proxmark3 /dev/ttyACM0
Segmentation fault (core dumped)
```

I add `-g -O0` to  in Makefile and run under `gdb`:

```
$ gdb --args ./proxmark3 /dev/ttyACM0
GNU gdb (GDB) Red Hat Enterprise Linux 7.6.1-100.el7_4.1
Copyright (C) 2013 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-redhat-linux-gnu".
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>...
Reading symbols from /home/doka/proj/RFID/proxmark3-3.0.1/client/proxmark3...done.
(gdb) run
Starting program: /home/doka/proj/RFID/proxmark3-3.0.1/client/./proxmark3 /dev/ttyACM0
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff584c921 in __strlen_sse2_pminub () from /lib64/libc.so.6
Missing separate debuginfos, use: debuginfo-install bzip2-libs-1.0.6-13.el7.x86_64 elfutils-libelf-0.168-8.el7.x86_64 elfutils-libs-0.168-8.el7.x86_64 expat-2.1.0-10.el7_3.x86_64 glib2-2.50.3-3.el7.x86_64 glibc-2.17-196.el7_4.2.x86_64 libX11-1.6.5-1.el7.x86_64 libXau-1.0.8-2.1.el7.x86_64 libXdamage-1.1.4-4.1.el7.x86_64 libXext-1.3.3-3.el7.x86_64 libXfixes-5.0.3-1.el7.x86_64 libXxf86vm-1.1.4-1.el7.x86_64 libattr-2.4.46-12.el7.x86_64 libcap-2.22-9.el7.x86_64 libdrm-2.4.74-1.el7.x86_64 libffi-3.0.13-18.el7.x86_64 libgcc-4.8.5-16.el7_4.1.x86_64 libgcrypt-1.5.3-14.el7.x86_64 libgpg-error-1.12-3.el7.x86_64 libicu-50.1.2-15.el7.x86_64 libpng-1.5.13-7.el7_2.x86_64 libselinux-2.5-12.el7.x86_64 libstdc++-4.8.5-16.el7_4.1.x86_64 libxcb-1.12-1.el7.x86_64 libxshmfence-1.2-1.el7.x86_64 mesa-libGL-17.0.1-6.20170307.el7.x86_64 mesa-libglapi-17.0.1-6.20170307.el7.x86_64 ncurses-libs-5.9-14.20130511.el7_4.x86_64 pcre-8.32-17.el7.x86_64 qt5-qtbase-5.6.2-1.el7.x86_64 qt5-qtbase-gui-5.6.2-1.el7.x86_64 readline-6.2-10.el7.x86_64 systemd-libs-219-42.el7_4.7.x86_64 xz-libs-5.2.2-1.el7.x86_64 zlib-1.2.7-17.el7.x86_64
(gdb) backtrace
#0  0x00007ffff584c921 in __strlen_sse2_pminub () at /lib64/libc.so.6
#1  0x00000000004129fd in wai_getExecutablePath (out=0x0, capacity=0, dirname_length=0x0) at whereami.c:198
#2  0x0000000000407fe2 in set_my_executable_path () at proxmark3.c:206
#3  0x00000000004081d1 in main (argc=2, argv=0x7fffffffd788) at proxmark3.c:243
(gdb) 
```

At build stage I get highlighted warning also associated with `wai_getExecutablePath`:

```
gcc -MT obj/whereami.o -MMD -MP -MF obj/whereami.Td -std=c99 -D_ISOC99_SOURCE -I. -I../include -I../common -I../zlib -I/opt/local/include -I../liblua -Wall  -g -O0 -DHAVE_GUI -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED  -c -o obj/whereami.o whereami.c
whereami.c: In function ‘wai_getExecutablePath’:
whereami.c:194:5: warning: implicit declaration of function ‘realpath’ [-Wimplicit-function-declaration]
     resolved = realpath(WAI_PROC_SELF_EXE, buffer);
     ^
whereami.c:194:14: warning: assignment makes pointer from integer without a cast [enabled by default]
     resolved = realpath(WAI_PROC_SELF_EXE, buffer);
              ^
whereami.c: In function ‘wai_getModulePath’:
whereami.c:271:20: warning: assignment makes pointer from integer without a cast [enabled by default]
           resolved = realpath(path, buffer);
                    ^
mv -f obj/whereami.Td obj/whereami.d
```
I'm try to replace `whereami.c` on lastest version from here: https://github.com/gpakosz/whereami/blob/master/src/whereami.c but issue by run the same.

How can I fix it?

_____________________
PS: my OS is CentOS 7.4 64bit

pwpiwi, 2018-10-01:
`realpath()` is usually defined in `stdlib.h` which is included by `whereami.c`. Isn't this the case for CentOS?

iDoka, 2018-10-01:
On my system `realpath()` defined inside preprocessing pragma:

```
$ grep -nA2 -B7 realpath /usr/include/stdlib.h
726-
727-#if defined __USE_BSD || defined __USE_XOPEN_EXTENDED
728-/* Return the canonical absolute name of file NAME.  If RESOLVED is
729-   null, the result is malloc'd; otherwise, if the canonical name is
730-   PATH_MAX chars or more, returns null with `errno' set to
731-   ENAMETOOLONG; if the name fits in fewer than PATH_MAX chars,
732-   returns the name in RESOLVED.  */
733:extern char *realpath (const char *__restrict __name,
734-		       char *__restrict __resolved) __THROW __wur;
735-#endif
```

Any ideas for workaround?

pwpiwi, 2018-10-01:
We would need something like
```
#ifdef CentOS
#define __USE_BSD
#endif
```
(or whatever is defined on CentOS).

iDoka, 2018-10-01:
here predefined macros in CentOS 7:

```
$ gcc -dM -E - </dev/null

#define __DBL_MIN_EXP__ (-1021)
#define __UINT_LEAST16_MAX__ 65535
#define __ATOMIC_ACQUIRE 2
#define __FLT_MIN__ 1.17549435082228750797e-38F
#define __UINT_LEAST8_TYPE__ unsigned char
#define __INTMAX_C(c) c ## L
#define __CHAR_BIT__ 8
#define __UINT8_MAX__ 255
#define __WINT_MAX__ 4294967295U
#define __ORDER_LITTLE_ENDIAN__ 1234
#define __SIZE_MAX__ 18446744073709551615UL
#define __WCHAR_MAX__ 2147483647
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_1 1
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_2 1
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_4 1
#define __DBL_DENORM_MIN__ ((double)4.94065645841246544177e-324L)
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_8 1
#define __GCC_ATOMIC_CHAR_LOCK_FREE 2
#define __FLT_EVAL_METHOD__ 0
#define __unix__ 1
#define __GCC_ATOMIC_CHAR32_T_LOCK_FREE 2
#define __x86_64 1
#define __UINT_FAST64_MAX__ 18446744073709551615UL
#define __SIG_ATOMIC_TYPE__ int
#define __DBL_MIN_10_EXP__ (-307)
#define __FINITE_MATH_ONLY__ 0
#define __GNUC_PATCHLEVEL__ 5
#define __UINT_FAST8_MAX__ 255
#define __DEC64_MAX_EXP__ 385
#define __INT8_C(c) c
#define __UINT_LEAST64_MAX__ 18446744073709551615UL
#define __SHRT_MAX__ 32767
#define __LDBL_MAX__ 1.18973149535723176502e+4932L
#define __UINT_LEAST8_MAX__ 255
#define __GCC_ATOMIC_BOOL_LOCK_FREE 2
#define __UINTMAX_TYPE__ long unsigned int
#define __linux 1
#define __DEC32_EPSILON__ 1E-6DF
#define __unix 1
#define __UINT32_MAX__ 4294967295U
#define __LDBL_MAX_EXP__ 16384
#define __WINT_MIN__ 0U
#define __linux__ 1
#define __SCHAR_MAX__ 127
#define __WCHAR_MIN__ (-__WCHAR_MAX__ - 1)
#define __INT64_C(c) c ## L
#define __DBL_DIG__ 15
#define __GCC_ATOMIC_POINTER_LOCK_FREE 2
#define __SIZEOF_INT__ 4
#define __SIZEOF_POINTER__ 8
#define __USER_LABEL_PREFIX__ 
#define __STDC_HOSTED__ 1
#define __LDBL_HAS_INFINITY__ 1
#define __FLT_EPSILON__ 1.19209289550781250000e-7F
#define __LDBL_MIN__ 3.36210314311209350626e-4932L
#define __DEC32_MAX__ 9.999999E96DF
#define __INT32_MAX__ 2147483647
#define __SIZEOF_LONG__ 8
#define __STDC_IEC_559__ 1
#define __STDC_ISO_10646__ 201103L
#define __UINT16_C(c) c
#define __DECIMAL_DIG__ 21
#define __gnu_linux__ 1
#define __LDBL_HAS_QUIET_NAN__ 1
#define __GNUC__ 4
[... 152 lines cut ...]
#define __DEC32_MANT_DIG__ 7
#define __DBL_MAX_10_EXP__ 308
#define __LDBL_DENORM_MIN__ 3.64519953188247460253e-4951L
#define __INT16_C(c) c
#define __STDC__ 1
#define __PTRDIFF_TYPE__ long int
#define __ATOMIC_SEQ_CST 5
#define __UINT32_TYPE__ unsigned int
#define __UINTPTR_TYPE__ long unsigned int
#define __DEC64_SUBNORMAL_MIN__ 0.000000000000001E-383DD
#define __DEC128_MANT_DIG__ 34
#define __LDBL_MIN_10_EXP__ (-4931)
#define __SIZEOF_LONG_LONG__ 8
#define __GCC_ATOMIC_LLONG_LOCK_FREE 2
#define __LDBL_DIG__ 18
#define __FLT_DECIMAL_DIG__ 9
#define __UINT_FAST16_MAX__ 18446744073709551615UL
#define __GNUC_GNU_INLINE__ 1
#define __GCC_ATOMIC_SHORT_LOCK_FREE 2
#define __UINT_FAST8_TYPE__ unsigned char
#define __ATOMIC_ACQ_REL 4
#define __ATOMIC_RELEASE 3
```

which one and where I should use for solve issue?

pwpiwi, 2018-10-01:
Hmm, I can't see anything specific for CentOS. Another option would be to define __USE_BSD for all Linux versions - hoping that it doesn't create other issues.

iDoka, 2018-10-01:
I added `-D__USE_BSD` to all `CFLAGS/CXXFLAGS` options in make
but issue isn't fix:
```
gcc -MT obj/whereami.o -MMD -MP -MF obj/whereami.Td  -std=c99 -D_ISOC99_SOURCE -I. -I../include -I../common -I../common/polarssl -I../zlib -I../uart -I/opt/local/include -I../liblua -I./jansson -Wall -g -O3 -D__USE_BSD  -DWITH_ISO14443a_StandAlone -DWITH_LF -DWITH_ISO15693 -DWITH_ISO14443a -DWITH_ISO14443b -DWITH_ICLASS -DWITH_LEGICRF -DWITH_HITAG -DWITH_CRC -DWITH_HFSNOOP -DWITH_GUI -DHAVE_GUI -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED  -c -o obj/whereami.o whereami.c
whereami.c: In function ‘wai_getExecutablePath’:
whereami.c:194:5: warning: implicit declaration of function ‘realpath’ [-Wimplicit-function-declaration]
     resolved = realpath(WAI_PROC_SELF_EXE, buffer);
     ^
whereami.c:194:14: warning: assignment makes pointer from integer without a cast [enabled by default]
     resolved = realpath(WAI_PROC_SELF_EXE, buffer);
              ^
whereami.c: In function ‘wai_getModulePath’:
whereami.c:271:20: warning: assignment makes pointer from integer without a cast [enabled by default]
           resolved = realpath(path, buffer);
                    ^
mv -f obj/whereami.Td obj/whereami.d
```

merlokk, 2018-10-01:
what lines in proxmark3.c do you have in your sources?
in master branch there are strange lines there....
```
#2  0x0000000000407fe2 in set_my_executable_path () at proxmark3.c:206
#3  0x00000000004081d1 in main (argc=2, argv=0x7fffffffd788) at proxmark3.c:243
```

iDoka, 2018-10-01:
@merlokk I try both versions: from lastest git and from here: https://github.com/Proxmark/proxmark3/archive/v3.0.1.tar.gz

Currently I use v3.0.1

pwpiwi, 2018-10-01:
Try with `_BSD_SOURCE ` instead.

iDoka, 2018-10-02:
@pwpiwi much thanks!!!
`_BSD_SOURCE` works for my case!

closed 2018-10-02 by iDoka

merlokk, 2018-10-03:
https://github.com/gpakosz/whereami/issues/15