Proxmark/proxmark3 #701: msys2 Windows10 maximum length of comamnd via readline is 254 chars.

merlokk, 2018-10-22:
from #697 
```
			rl_initialize();
					rl_extend_line_buffer(1024);
					cmd = readline(PROXPROMPT);
					struct readline_state sp;
					rl_save_state(&sp);
					printf("---len: %d %d\n", strlen(cmd), sp.buflen);
```
---len: 254 1280

test command here:
`hf fido auth 000102030405060708090a0b0c0d0e0f000102030405060708090a0b0c0d0e0f000102030405060708090a0b0c0d0e0f000102030405060708090a0b0c0d0e0f 000102030405060708090a0b0c0d0e0f000102030405060708090a0b0c0d0e0f 000102030405060708090a0b0c0d0e0f000102030405060708090a0b0c0d0e0f`

this works:
`proxmark3 com11 -c "hf fido auth 000102030405060708090a0b0c0d0e0f000102030405060708090a0b0c0d0e0f000102030405060708
090a0b0c0d0e0f000102030405060708090a0b0c0d0e0f 000102030405060708090a0b0c0d0e0f000102030405060708090a0b0c0d0e0f 00010203040506070
8090a0b0c0d0e0f000102030405060708090a0b0c0d0e0f"`

looks like this problem, but on C# https://stackoverflow.com/questions/5557889/console-readline-max-length

Gator96100, 2018-10-22:
It is definitely a readline issue. I created a simple test application with readline and this issue does appear.
`
#include <stdio.h>
#include <stdlib.h>
#include <string>
#include <readline/readline.h>
#include <readline/history.h>

int main(int argc, char** argv) {

  char* buf;
  while ((buf = readline(">> ")) != nullptr) {

    printf("[%s]\n", buf);

    // readline malloc's a new buffer every time.
    free(buf);
  }

  return 0;
}
`

Gator96100, 2018-10-22:
When downgrading to readline6 this error does not exist

pwpiwi, 2018-10-22:
I could successfully test on Windows 10. I am calling msys.bat without any parameters and I have then the following environment variables set:
```
COMSPEC='C:\WINDOWS\SysWOW64\cmd.exe'
MSYSCON=sh.exe
MSYSTEM=MINGW32
OS=Windows_NT
OSTYPE=msys
```

merlokk, 2018-10-22:
this `bat`?
https://github.com/Gator96100/ProxSpace/blob/master/msys2/msys2_shell.cmd
or you have another?

merlokk, 2018-10-22:
as i see for me 

```
COMSPEC=C:\Windows\system32\cmd.exe
MSYSCON=defterm
MSYSTEM=MINGW32
```

pwpiwi, 2018-10-22:
Another `bat`. I am still on msys, not msys2. Readline 5.0.

merlokk, 2018-10-22:
it looks it between readline and msys. but where...

merlokk, 2018-10-22:
as i see in msys2 cmd file set env variables and start bash.exe
nothing more...

Gator96100, 2018-10-23:
I rather fix it by finding the problem in readline. I think qt5 requires readline7.

merlokk, 2018-10-23:
I dont see sources for readline. ill try to find it in msys tomorrow.....
I raised issue in msys repo, but it ignored(

merlokk, 2018-10-24:
I see only
```
ProxSpace\msys2\mingw32\include\readline\readline.h
ProxSpace\msys2\mingw32\include\editline\readline.h
```
so it needs to recompile it with msys2 components....
https://github.com/Alexpux/MSYS2-packages/tree/master/readline
but there i dont see sources too(

merlokk, 2018-10-24:
https://tiswww.case.edu/php/chet/readline/rltop.html
http://git.savannah.gnu.org/cgit/readline.git/snapshot/readline-master.tar.gz
i dont have an idea how to change it in msys.
maybe let it be. 
lets wait reaction
https://github.com/Alexpux/MSYS2-packages/issues/1465

Gator96100, 2018-10-24:
If you are looking for the source code of readline you will find it here: [https://ftp.gnu.org/gnu/readline/readline-7.0.tar.gz](url)
How the package is build is defined here [https://github.com/Alexpux/MINGW-packages/blob/master/mingw-w64-readline/PKGBUILD](url)

merlokk, 2018-10-24:
but there no sources. there just updates...
it needs to generate update and somehow compile msys2?

Gator96100, 2018-10-26:
Not a Proxmark, ProxSpace or msys2 issue. A workaround is live and this issue was reported to the correct project. I guess this can be closed now.

closed 2018-10-27 by merlokk