Proxmark/proxmark3 #703: brew install --HEAD / git clone make halts on error

dmaij, 2018-10-23:
When running (on OSX Mojave)
```
brew tap proxmark/proxmark3
brew install --HEAD proxmark3
```
The following error occurs:
```
iso15693.c:928:13: error: 'DecodeReader.posCount' may be used uninitialized in this function [-Werror=maybe-uninitialized]
  if (DEBUG) Dbprintf("max behindby = %d, samples = %d, gotFrame = %d, Decoder: state = %d, len = %d, bitCount = %d, posCount = %d",
             ^
iso15693.c:864:17: note: 'DecodeReader.posCount' was declared here
  DecodeReader_t DecodeReader;
                 ^
cc1: all warnings being treated as errors
make[1]: *** [obj/iso15693.o] Error 1
make: *** [armsrc/all] Error 2
```

This is repeatable by checking out the git and building
```
git clone https://github.com/Proxmark/proxmark3.git
make clean; make
```
(same error, not repeated here)

This is easily fixed by adding a declaration:
```
diff --git a/armsrc/iso15693.c b/armsrc/iso15693.c
index 9479c3c..a5c5b78 100644
--- a/armsrc/iso15693.c
+++ b/armsrc/iso15693.c
@@ -838,6 +838,7 @@ static void DecodeReaderInit(uint8_t *data, uint16_t max_len, DecodeReader_t* De
        DecodeReader->byteCount = 0;
        DecodeReader->bitCount = 0;
        DecodeReader->shiftReg = 0;
+       DecodeReader->posCount = 0;
}
```

pwpiwi, 2018-10-23:
Please check latest HEAD.

dmaij, 2018-10-24:
Thanks, just confirmed building and brew install working. on (osx 10.14 mojave)
You may close this one.

closed 2018-10-24 by pwpiwi