Proxmark/proxmark3 #707: Flasing is broken (for some laptops!)

ikarus23, 2018-10-30:
Hi,
I have access to several laptops (with arch linux) but most of them are not able to flash a proxmark. They get stuck right in the middle of flashing (causing high CPU load).

```
./client/flasher /dev/ttyACM1 armsrc/obj/fullimage.elf
Loading ELF file 'armsrc/obj/fullimage.elf'...
Loading usable ELF segments:
0: V 0x00102000 P 0x00102000 (0x0002b660->0x0002b660) [R X] @0x94
1: V 0x00200000 P 0x0012d660 (0x00001928->0x00001928) [RW ] @0x2b6f4
Note: Extending previous segment from 0x2b660 to 0x2cf88 bytes

Waiting for Proxmark to appear on /dev/ttyACM1 ......
 Found.
Entering bootloader...
(Press and release the button only to abort)
Waiting for Proxmark to appear on /dev/ttyACM1 ..
 Found.

Flashing...
Writing segments for file: armsrc/obj/fullimage.elf
 0x00102000..0x0012ef87 [0x2cf88 / 360 blocks].....................Waiting for a response from the proxmark...          
You can cancel this operation by pressing the pm3 button   
```
However, another laptop (configured in the same way) is able to flash it. Are there timing issues with the USB? Does anyone else has the same issue? I got it for quite a while now...

pwpiwi, 2018-10-30:
Could be a modem manager issue. Please have a look here:  https://github.com/RfidResearchGroup/proxmark3/issues/35

ikarus23, 2018-10-31:
Damn, that's embarrassing. I know very well that modem manager is the source for some of these issues with proxmark. I even deactivated on the system that is causing the trouble. However, somehow it was running again (maybe due to an update or me playing around with network manager).

Now everything is back to normal. Thanks for pointing out the obvious! ;)

closed 2018-10-31 by ikarus23