Proxmark/proxmark3 #715: lets add topics to repository

merlokk, 2018-11-18:
hi

to have a better search visibility of proxmark repository lets add topics to repository header.
i suggest to add iso14443, u2f, fido2, nfc, etc...

pwpiwi, 2018-11-18:
Good idea, but afaik can only be done by the owner of the repository.

merlokk, 2018-11-18:
may you try?

pwpiwi, 2018-11-18:
No, I don't. (or I am looking in the wrong place)

merlokk, 2018-11-18:
it on the left down corner of img. it looks you cant(
one of my repo:
[image]

pwpiwi, 2018-11-19:
@0xffff: Can you help?

pwpiwi, 2019-04-25:
@0xffff: are you still the owner of this github project? Can you please add the topics?
nfc
rfid
mifare
proxmark3
proxmark
pm3
rdv40
iso15693
iso14443a
iso14443b
125khz
clone
simulate
iso7816
iclass
contactless
sniffer

0x2b3bfa0, 2020-02-27:
Also, maybe:

* `nested`
* `hardnested`