Proxmark/proxmark3 #719: forum registration is broken

sbma44, 2018-11-21:
authedmine.com has an invalid SSL cert, preventing the captcha from working. Confirmed in Chrome and Safari. The only workaround here is installing a busted cert which seems nonideal.

[image]

[image]

RfidResearchGroup, 2018-11-21:
Not ideal,  but it works very well against spam-bots.   Lets see if authedmine updates their SSL cert since it expires 24november 2018..

sbma44, 2018-11-21:
Actually, ignore me: looks like the network I'm on is running a Cisco firewall thingamajig that has authedmine.com marked as malware. The busted Cisco cert is me getting MITMed by my own LAN. I opened a VPN connection and the captcha works fine. Sorry for the false alarm!

closed 2018-11-21 by sbma44