Proxmark/proxmark3 #731: Compile Error on Ubuntu

tresacton, 2018-12-04:
Error when running ```make clean && make all``` on an Ubuntu based system.

```
360:~/proxmark3$ uname -ar
Linux 360 4.15.0-42-generic #45~16.04.1-Ubuntu SMP Mon Nov 19 13:02:27 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
```

I've also experienced this issue on other Ubuntu system.

Error is:
```
...

arm-none-eabi-gcc -c -I../include -I../common -I. -Wall -Werror -pedantic -std=c99 -Os -DON_DEVICE -fno-strict-aliasing -ffunction-sections -fdata-sections -DWITH_ISO14443a_StandAlone -DWITH_LF -DWITH_ISO15693 -DWITH_ISO14443a -DWITH_ISO14443b -DWITH_ICLASS -DWITH_LEGICRF -DWITH_HITAG -DWITH_CRC -DWITH_HFSNOOP -DWITH_GUI -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED -I../zlib -mthumb-interwork -o obj/mifareutil.o mifareutil.c 
mifareutil.c: In function 'mifare_ultra_auth':
mifareutil.c:299:2: error: missing braces around initializer [-Werror=missing-braces]
  mbedtls_des3_context ctx = { 0x00 };
  ^
mifareutil.c:299:2: error: (near initialization for 'ctx.sk') [-Werror=missing-braces]
cc1: all warnings being treated as errors
../common/Makefile.common:84: recipe for target 'obj/mifareutil.o' failed
make[1]: *** [obj/mifareutil.o] Error 1
make[1]: Leaving directory '/home/user/proxmark3/armsrc'
Makefile:31: recipe for target 'armsrc/all' failed
make: *** [armsrc/all] Error 2

```

pwpiwi, 2018-12-04:
What is the version of your arm-none-eabi-gcc?

pwpiwi, 2018-12-04:
Can you please test if PR #732 fixes this issue?

closed 2018-12-06 by pwpiwi