Proxmark/proxmark3 #733: Is this illegal?

CrimsonTorso, 2018-12-04:
Hello, I'm wondering if this tool is illegal to use. Could someone reply back with the laws it's breaking if I were to use this.

pwpiwi, 2018-12-04:
Well, this is the software repository. Every contribution to this repository and therefore the whole project allows further distribution under GPL2 or later.

For other none software related aspects the forum is the better place to ask (http://www.proxmark.org/forum/index.php).

cds333, 2019-04-09:
If you use it on your own equipment or you get permission from the owner then it is completely legal. If you use it on someone else's property without permission to commit a crime then it is illegal. It is no different than a crowbar in that sense; it all depends on what you use it for- there are not any laws that pertain specifically to the proxmark.

darconeous, 2019-10-28:
I am not a lawyer, but...

I would think that, much like locksmith tools, the legality depends largely on what jurisdiction you are operating in and what you intend to do with this tool.

For example, in Florida if you own a crowbar and have the intent of using it to commit a burglary or trespass, then that is "criminal possession of burglary tools", a third-degree felony:

> Under the law, if you possess any implement, tool, or machine with the intent to use the same to commit a burglary or trespass, or even allow the implement, tool, or machine to be used to commit a burglary or trespass, you can be found guilty of a third degree felony. Notice again the issue of intent, in the context of a charge for possession of burglary tools. For example, if you possess a hacksaw and the government says you intend to use it to cut through a lock on someone’s garage to steal something inside, you can be convicted of possession of burglary tools.

(quoted from [here](https://cobblawfirm.com/florida-criminal-law-can-it-really-be-illegal-to-possess-a-crowbar-hacksaw-or-even-a-simple-screwdriver-because-they-are-burglary-tools/))

In some places (Poland, Japan, etc) just *possessing* lock picking tools without having a locksmith license is considered criminal intent. Some creative interpretation of what a "lock picking tool" by the local prosecutor could lead to trouble if you get their attention.

Bottom line: Don't use the tool to do anything illegal, and **know the laws in your jurisdiction**.