Proxmark/proxmark3 #738: hf mf dump not using key B when trying to read from card

chuayupeng, 2018-12-18:
Was trying to clone a card that had only key B read access, and the dump command was failing because of that.

pwpiwi, 2018-12-19:
hf mf dump **does** use Key B when necessary. At least it should. If it doesn't there should be a much simpler fix than the proposed PR.

pwpiwi, 2018-12-20:
Can you please post the log (commands typed, screen output) from your failed dump process including the acquisition of dumpkeys.bin.

chuayupeng, 2018-12-21:
in comparison after my edits in my branch, this was the output.
```
proxmark3> hf mf dump 1
|-----------------------------------------|          
|------ Reading sector access bits...-----|          
|-----------------------------------------|          
#db# Authentication failed. Card timeout.          
#db# Auth error          
#db# READ BLOCK FINISHED          
#db# READ BLOCK FINISHED          
#db# Authentication failed. Error card response.          
#db# Auth error          
#db# READ BLOCK FINISHED          
#db# READ BLOCK FINISHED          
#db# Authentication failed. Card timeout.          
#db# Auth error          
#db# READ BLOCK FINISHED          
#db# READ BLOCK FINISHED          
#db# Authentication failed. Card timeout.          
#db# Auth error          
#db# READ BLOCK FINISHED          
#db# READ BLOCK FINISHED          
#db# Authentication failed. Card timeout.          
#db# Auth error          
#db# READ BLOCK FINISHED          
#db# READ BLOCK FINISHED          
#db# Authentication failed. Card timeout.          
#db# Auth error          
#db# READ BLOCK FINISHED          
#db# READ BLOCK FINISHED          
#db# READ BLOCK FINISHED          
#db# READ BLOCK FINISHED          
#db# READ BLOCK FINISHED          
#db# READ BLOCK FINISHED          
#db# READ BLOCK FINISHED          
#db# READ BLOCK FINISHED          
#db# READ BLOCK FINISHED          
#db# READ BLOCK FINISHED          
#db# READ BLOCK FINISHED          
#db# READ BLOCK FINISHED          
|-----------------------------------------|          
|----- Dumping all blocks to file... -----|          
|-----------------------------------------|          
#db# Authentication failed. Card timeout.          
#db# Auth error          
#db# READ BLOCK FINISHED          
#db# READ BLOCK FINISHED          
Successfully read block  0 of sector  0.          
#db# Authentication failed. Card timeout.          
#db# Auth error          
#db# READ BLOCK FINISHED          
#db# READ BLOCK FINISHED          
Successfully read block  1 of sector  0.          
#db# Authentication failed. Card timeout.          
#db# Auth error          
#db# READ BLOCK FINISHED          
#db# READ BLOCK FINISHED          
Successfully read block  2 of sector  0.          
#db# Authentication failed. Card timeout.          
#db# Auth error          
#db# READ BLOCK FINISHED          
#db# READ BLOCK FINISHED          
Successfully read block  3 of sector  0.          
#db# Authentication failed. Card timeout.          
#db# Auth error          
#db# READ BLOCK FINISHED          
#db# READ BLOCK FINISHED          
Successfully read block  0 of sector  1.          
#db# Authentication failed. Card timeout.          
#db# Auth error          
[... 151 lines cut ...]
#db# READ BLOCK FINISHED          
Successfully read block  1 of sector 13.          
#db# READ BLOCK FINISHED          
Successfully read block  2 of sector 13.          
#db# READ BLOCK FINISHED          
Successfully read block  3 of sector 13.          
#db# READ BLOCK FINISHED          
Successfully read block  0 of sector 14.          
#db# READ BLOCK FINISHED          
Successfully read block  1 of sector 14.          
#db# READ BLOCK FINISHED          
Successfully read block  2 of sector 14.          
#db# READ BLOCK FINISHED          
Successfully read block  3 of sector 14.          
#db# READ BLOCK FINISHED          
Successfully read block  0 of sector 15.          
#db# READ BLOCK FINISHED          
Successfully read block  1 of sector 15.          
#db# READ BLOCK FINISHED          
Successfully read block  2 of sector 15.          
#db# READ BLOCK FINISHED          
Successfully read block  3 of sector 15.          
Dumped 64 blocks (1024 bytes) to file dumpdata.bin   
proxmark3> 
```

pwpiwi, 2018-12-21:
Thanks. Obviously the Access Conditions cannot be read with KeyA which should always be possible. Is this really a Mifare Classic card? We need to investigate a bit further. Can you please try and send the result of
* hf 14a info
* hf mf rdbl 3 A  000000000000
* hf mf rdbl 3 B  b0b1b2b3b4b5

chuayupeng, 2018-12-21:
Sure thing! I'm new to this so not entirely sure as well, here are the results of the commands.

```
proxmark3> hf 14a info
 UID : db 32 c2 77           
ATQA : 00 04          
 SAK : 08 [2]          
TYPE : NXP MIFARE CLASSIC 1k | Plus 2k SL1          
proprietary non iso14443-4 card found, RATS not supported          
No chinese magic backdoor command detected          
Prng detection: WEAK          
proxmark3> 
```

```
proxmark3> hf mf rdbl 3 A 000000000000
--block no:3, key type:A, key:00 00 00 00 00 00            
#db# Authentication failed. Card timeout.          
#db# Auth error          
#db# READ BLOCK FINISHED          
isOk:00          
proxmark3> 
```

```
proxmark3> hf mf rdbl 3 B b0b1b2b3b4b5
--block no:3, key type:B, key:b0 b1 b2 b3 b4 b5            
#db# READ BLOCK FINISHED          
isOk:01 data:00 00 00 00 00 00 7f 07 88 69 00 00 00 00 00 00           
Trailer decoded:          
Access block 0: read AB; write AB; increment AB; decrement transfer restore AB          
Access block 1: read AB; write AB; increment AB; decrement transfer restore AB          
Access block 2: read AB; write AB; increment AB; decrement transfer restore AB          
Access block 3: write A by B; read ACCESS by AB; write ACCESS by B; write B by B          
UserData: 69
```

Hope this helps!

marshmellow42, 2018-12-21:
Doesn't his nested cmd indicate a different keyA?  "hf mf nested 1 0 A
a0a1a2a3a4a5 d"

chuayupeng, 2018-12-24:
I got the original a0a1a2a3a4a5 key from running `hf mf mifare` and used it to run the nested command. Perhaps its an error in the nested command when giving the output? See below for output of both.
```
proxmark3> hf mf rdbl 0 A 000000000000
--block no:0, key type:A, key:00 00 00 00 00 00            
#db# Authentication failed. Card timeout.          
#db# Auth error          
#db# READ BLOCK FINISHED          
isOk:00  
```

```
proxmark3> hf mf rdbl 0 A a0a1a2a3a4a5
--block no:0, key type:A, key:a0 a1 a2 a3 a4 a5            
#db# READ BLOCK FINISHED          
isOk:01 data:db 32 c2 77 5c 88 04 00 c0 8e 5c d2 65 80 49 14  
```

chuayupeng, 2018-12-26:
Not too sure, how do I check that? Does the information given below when starting the program help?

```
Prox/RFID mark3 RFID instrument          
bootrom: master-rysc/v3.0.1 2017-09-21 16:36:44
os: master-rysc/v3.0.1 2017-09-21 16:36:45
LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
HF FPGA image built for 2s30vq100 on 2017/05/17 at 17:48:26
          
uC: AT91SAM7S512 Rev B          
Embedded Processor: ARM7TDMI          
Nonvolatile Program Memory Size: 512K bytes. Used: 192856 bytes (37%). Free: 331432 bytes (63%).          
Second Nonvolatile Program Memory Size: None          
Internal SRAM Size: 64K bytes          
Architecture Identifier: AT91SAM7Sxx Series          
Nonvolatile Program Memory Type: Embedded Flash Memory
```

pwpiwi, 2018-12-26:
Yes, this helps. Your firmware is from September 2017. You need to update it. See the Wiki (https://github.com/Proxmark/proxmark3/wiki/Windows#flashing-your-proxmark3-to-the-latest-firmware)

chuayupeng, 2018-12-27:
I'm using Kali Linux, and following this wiki (https://github.com/Proxmark/proxmark3/wiki/Kali-Linux#upgrading-proxmark3-firmware), but the firmware does not update. Added symbolic links to make sure the proxmark3 device is always detected because it kept switching between ttyACM0 and ttyACM1, but was awaiting response for over an hour. Is there any way to resolve this issue?

```
./flasher /dev/proxmark3 ../armsrc/obj/fullimage.elf 
Loading ELF file '../armsrc/obj/fullimage.elf'...
Loading usable ELF segments:
0: V 0x00102000 P 0x00102000 (0x0002c040->0x0002c040) [R X] @0x94
1: V 0x00200000 P 0x0012e040 (0x00001968->0x00001968) [RW ] @0x2c0d4
Note: Extending previous segment from 0x2c040 to 0x2d9a8 bytes

Waiting for Proxmark to appear on /dev/proxmark3 .
 Found.
Entering bootloader...
(Press and release the button only to abort)
Waiting for Proxmark to appear on /dev/proxmark3 ...................
 Found.
Waiting for a response from the proxmark...          
You can cancel this operation by pressing the pm3 button 
```

chuayupeng, 2018-12-27:
Turns out was an issue with ModemManger on Kali, fixed it and it works perfectly now! Thanks!

closed 2018-12-27 by chuayupeng