Proxmark/proxmark3 #764: Firmware version

H4kk3r3z, 2019-01-19:
How do I know what is the latest firmware?

Do you not have every firmware version going back to 1.0 or beta with its release date.

Would be nice to confirm before I brick my device

pwpiwi, 2019-01-20:
I am not sure if I correctly understand the question. If you compile (HEAD of) this repository, you will get the latest firmware. If you checkout any of the previous commits and compile, you will get the firmware of that point in time. If you download one of the releases and compile, you will get the firmware at the release date.

H4kk3r3z, 2019-01-24:
Fair enough... It would be nice if current version was written on front page.
Nevermind, Im being a big baby - Thanks

closed 2019-01-24 by H4kk3r3z

iceman1001, 2019-01-24:
I understand,  its quite frustrating not actually knowing how to identify the client version.
Since the firmware / bootrom has nice timestamps / version but the client in the offical repo lacks it.

There are releases here on Github, which is the recommened points in time for the source code.
Always use the same client/firmware from the same build.  If you download a release, make sure to keep all code seperate in different folders.  Like for instance **v2.2,  v3.1,  latest,**   

Do feel free to contribute,  an improved version text for the client is a good start for you :)