Proxmark/proxmark3 #767: 'emv search' finds same AID multiple times

pwpiwi, 2019-01-29:
```
proxmark3> emv search -s
 Search completed.
|------------------|--------|-------------------------|
|    AID           |Priority| Name                    |
|------------------|--------|-------------------------|
|a000000025010801  |   01   |AMERICAN EXPRESS         |
|a000000025010801  |   01   |AMERICAN EXPRESS         |
|a000000025010801  |   01   |AMERICAN EXPRESS         |
|------------------|--------|-------------------------|
proxmark3>
```
Probably caused by returning AID which only partially match. Searching for "A000000025", "A00000002501" and "A000000025010801" will all find the latter AID.

merlokk, 2019-01-29:
it looks that PSE have several records
may you put here result `emv pse -st` ?

pwpiwi, 2019-01-29:
'emv search' doesn't check the PPSE but tries each AID from the list. See the result of 'emv search -st':
```
proxmark3> emv search -sat
 >>>> 00 a4 04 00 09 a0 00 00 00 03 05 07 60 10 00
 <<<< 6a 82
ERROR: APDU(00a4) ERROR: [6A82] File not found
 >>>> 00 a4 04 00 07 a0 00 00 00 03 10 10 00
 <<<< 6a 82
ERROR: APDU(00a4) ERROR: [6A82] File not found
 >>>> 00 a4 04 00 08 a0 00 00 00 03 10 10 01 00
 <<<< 6a 82
ERROR: APDU(00a4) ERROR: [6A82] File not found
 >>>> 00 a4 04 00 08 a0 00 00 00 03 10 10 02 00
 <<<< 6a 82
ERROR: APDU(00a4) ERROR: [6A82] File not found
 >>>> 00 a4 04 00 07 a0 00 00 00 03 20 10 00
 <<<< 6a 82
ERROR: APDU(00a4) ERROR: [6A82] File not found
 >>>> 00 a4 04 00 07 a0 00 00 00 03 20 20 00
 <<<< 6a 82
ERROR: APDU(00a4) ERROR: [6A82] File not found
 >>>> 00 a4 04 00 07 a0 00 00 00 03 30 10 00
 <<<< 6a 82
ERROR: APDU(00a4) ERROR: [6A82] File not found
 >>>> 00 a4 04 00 07 a0 00 00 00 03 40 10 00
 <<<< 6a 82
ERROR: APDU(00a4) ERROR: [6A82] File not found
 >>>> 00 a4 04 00 07 a0 00 00 00 03 50 10 00
 <<<< 6a 82
ERROR: APDU(00a4) ERROR: [6A82] File not found
 >>>> 00 a4 04 00 07 a0 00 00 00 03 60 10 00
 <<<< 6a 82
ERROR: APDU(00a4) ERROR: [6A82] File not found
 >>>> 00 a4 04 00 07 a0 00 00 00 03 60 20 00
 <<<< 6a 82
ERROR: APDU(00a4) ERROR: [6A82] File not found
 >>>> 00 a4 04 00 07 a0 00 00 00 03 80 02 00
 <<<< 6a 82
ERROR: APDU(00a4) ERROR: [6A82] File not found
 >>>> 00 a4 04 00 07 a0 00 00 00 03 80 10 00
 <<<< 6a 82
ERROR: APDU(00a4) ERROR: [6A82] File not found
 >>>> 00 a4 04 00 07 a0 00 00 00 03 90 10 00
 <<<< 6a 82
ERROR: APDU(00a4) ERROR: [6A82] File not found
 >>>> 00 a4 04 00 08 a0 00 00 00 03 99 99 10 00
 <<<< 6a 82
ERROR: APDU(00a4) ERROR: [6A82] File not found
 >>>> 00 a4 04 00 05 a0 00 00 00 98 00
 <<<< 6a 82
ERROR: APDU(00a4) ERROR: [6A82] File not found
 >>>> 00 a4 04 00 07 a0 00 00 00 98 08 48 00
 <<<< 6a 82
ERROR: APDU(00a4) ERROR: [6A82] File not found
 >>>> 00 a4 04 00 06 a0 00 00 00 04 01 00
 <<<< 6a 82
ERROR: APDU(00a4) ERROR: [6A82] File not found
 >>>> 00 a4 04 00 07 a0 00 00 00 04 10 10 00
 <<<< 6a 82
ERROR: APDU(00a4) ERROR: [6A82] File not found
 >>>> 00 a4 04 00 09 a0 00 00 00 04 10 10 12 13 00
 <<<< 6a 82
ERROR: APDU(00a4) ERROR: [6A82] File not found
 >>>> 00 a4 04 00 09 a0 00 00 00 04 10 10 12 15 00
 <<<< 6a 82
ERROR: APDU(00a4) ERROR: [6A82] File not found
 >>>> 00 a4 04 00 07 a0 00 00 00 04 20 10 00
 <<<< 6a 82
ERROR: APDU(00a4) ERROR: [6A82] File not found
 >>>> 00 a4 04 00 07 a0 00 00 00 04 30 10 00
[... 95 lines cut ...]
 >>>> 00 a4 04 00 07 a0 00 00 00 42 50 10 00
 <<<< 6a 82
ERROR: APDU(00a4) ERROR: [6A82] File not found
 >>>> 00 a4 04 00 06 a0 00 00 00 65 10 00
 <<<< 6a 82
ERROR: APDU(00a4) ERROR: [6A82] File not found
 >>>> 00 a4 04 00 07 a0 00 00 00 65 10 10 00
 <<<< 6a 82
ERROR: APDU(00a4) ERROR: [6A82] File not found
 >>>> 00 a4 04 00 07 a0 00 00 01 54 44 42 00
 <<<< 6a 82
ERROR: APDU(00a4) ERROR: [6A82] File not found
 >>>> 00 a4 04 00 07 f0 00 00 00 03 00 01 00
 <<<< 6a 82
ERROR: APDU(00a4) ERROR: [6A82] File not found
 >>>> 00 a4 04 00 07 a0 00 00 05 24 10 10 00
 <<<< 6a 82
ERROR: APDU(00a4) ERROR: [6A82] File not found
 >>>> 00 a4 04 00 07 d5 78 00 00 02 10 10 00
 <<<< 6a 82
ERROR: APDU(00a4) ERROR: [6A82] File not found
 Search completed.
proxmark3>
```
BTW: doesn't print the table with '-sat' ???

merlokk, 2019-01-29:
yes, you are right) needs to check if field exists.

it not prints table with `-t`

pwpiwi, 2019-02-06:
fixed by PR #776

closed 2019-02-06 by pwpiwi