Proxmark/proxmark3 #773: FIDO U2F authenticator not detected anymore since PR #769

Fl0-0, 2019-02-02:
Since PR #769 my NFC FIDO2 is not detected: `FIDO2 version not exists (6700 - Wrong length)`

 ```
FIDO U2F authenricator detected. Version: U2F_V2          
FIDO2 version not exists (6700 - Wrong length). 
```

Reverted to  #437035a75b2737d0b36886bc2bacaeb0219c6325 it works:

```
FIDO U2F authenricator detected. Version: U2F_V2          
FIDO2 version CBOR decoded:          
Map[
  1 (versions): Array[
    U2F_V2
    FIDO_2_0
  ]
  2 (extensions): Array[
    hmac-secret
  ]
...
```

I think the cause  is the changes made in client/fido/fidocore.c .

merlokk, 2019-02-02:
I'll look.  What authenticator do you have?

Fl0-0, 2019-02-02:
I use a YubiKey 5 NFC.

merlokk, 2019-02-02:
works

Fl0-0, 2019-02-03:
Yes ! Thanks @merlokk

closed 2019-02-04 by pwpiwi