Proxmark/proxmark3 #779: Error flashin cdc on unbuntu

Zer0-Case, 2019-02-06:
So when i go to do the flashin for cdc it (the specific command is client/flasher /dev/ttyACM0 -b bootrom/obj/bootrom.elf armsrc/obj/fullimage.elf) and it tells me "bash: client/flasher: No such file or directory" and i have no idea why. I tried running the command in the proxmark3 director and outside of it to no avail. I'm sure i'm just making a silly error but i'm really at a lose.

pwpiwi, 2019-02-06:
Your working directory may not be in your PATH environment. Try `./client/flasher`

pwpiwi, 2019-02-09:
Can be closed?

Zer0-Case, 2019-02-12:
No such luck i still get the same error.

pwpiwi, 2019-02-12:
Most probably this is not an issue with the code. Please check the forum and then ask your question there (http://www.proxmark.org/forum/index.php).

closed 2019-02-21 by pwpiwi