Proxmark/proxmark3 #785: PM3 USB undetected after full image flash

Squishy33, 2019-02-15:
Hi,
after flashing my PM3 my laptop cannot detect it anymore and `dmesg` shows:

> [15322.091286] usb 3-2: new full-speed USB device number 52 using xhci_hcd
> [15339.453992] usb 3-2: new full-speed USB device number 53 using xhci_hcd
> [15356.812779] usb 3-2: new full-speed USB device number 54 using xhci_hcd
> [15374.175611] usb 3-2: new full-speed USB device number 55 using xhci_hcd
> [15391.538327] usb 3-2: new full-speed USB device number 56 using xhci_hcd
> [15408.905157] usb 3-2: new full-speed USB device number 57 using xhci_hcd
> [15426.263979] usb 3-2: new full-speed USB device number 58 using xhci_hcd

and `grep -i xhci /boot/config-$(uname -r)`

> CONFIG_USB_XHCI_HCD=m
> CONFIG_USB_XHCI_PCI=m
> CONFIG_USB_XHCI_PLATFORM is not set

PM3 device doesn't show on `lsub, lsusb -t`, i would like to know if you can help me in this. Or is there a way to reroll (that is if my laptop is able to detect it). thanx

Fl0-0, 2019-02-17:
Hi @Squishy33, you might have corrupted the firmware...
https://github.com/Proxmark/proxmark3/wiki/flashing

Which command do you use for flashing ?
You must kill daemons which can interfere with the `tty` peripherals, on Kali:
-  `systemctl stop ModemManager`
-  `systemctl disable ModemManager`

Next:
1. Try to reflash it by holding the button on the Proxmark while connecting it to your computer until flashing is complete.
2. If the booloader is corrupted you must recover it using the JTAG procedure.

Squishy33, 2019-02-18:
Hi,
I might have corrupted the firmware while doing the update and this time i read through the flashing procedure and also used the 1st method you have proposed which worked and now it is up and running again. Im glad that the bootloader wasnt corrupted.

closed 2019-02-18 by iceman1001