Proxmark/proxmark3 #793: wrong hash calculation in 'emv scan -w'

pwpiwi, 2019-02-21:
'emv scan -w ' results in 
```
[...]

INFO: -->Recovering certificates.
CA Public Key index 0xf
Verifying CA Public Key for a0:00:00:00:25 IDX 0f 1408 bits...OK
 Issuer PK recovered. RID a0:00:00:00:25 IDX 0f CSN 01:00:06
ERROR: Calculated wrong hash

INFO: decoded:    1f 99 41 d3 25 5b 44 2f 22 f8 62 ff 67 d1 c4 65 0c f2 68 2d

INFO: calculated: f2 54 ba 5b d6 74 01 f0 21 e7 9f 50 5d f9 ae 26 78 7b cd cb

 ICC PK recovered. RID a0:00:00:00:25 IDX 0f CSN 01:01:01

 File `C:\MinGW\msys\1.0\home\udr100\github\proxmark3\client\test.json` saved.
```

merlokk, 2019-02-22:
Its ok because it just scans card and not make full transaction. 
https://github.com/Proxmark/proxmark3/blob/master/client/emv/cmdemv.c#L1711-L1713
maybe it needs to suppress errors? or let it be?

pwpiwi, 2019-02-26:
Is this the same issue as in 'emv Roca'? Which I had fixed with PR #781?

merlokk, 2019-02-26:
yes. can be fixed by ODAiList