Proxmark/proxmark3 #794: Your bootloader does not understand the new START_FLASH command

bleepnetworks, 2019-02-24:
Hi Proxmark,

I have a very old devkit which is running the HID firmware:

[  186.338042] usb 1-2: new full-speed USB device number 3 using ohci-pci
[  186.602874] usb 1-2: New USB device found, idVendor=9ac4, idProduct=4b8f
[  186.602879] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  186.602881] usb 1-2: Product: ProxMark-3 RFID Instrument
[  186.602882] usb 1-2: Manufacturer: J. Westhues
[  196.650896] hid-generic 0003:9AC4:4B8F.0002: hiddev0,hidraw1: USB HID v1.00 Device [J. Westhues ProxMark-3 RFID Instrument] on usb-0000:00:06.0-2/input0

I flashed the old HID firmware as per https://github.com/Proxmark/proxmark3/wiki/Ubuntu-Linux#Flashing_HID which looked good initially:

```
./flasher -b ../../bootrom/obj/bootrom.elf 
Loading ELF file '../../bootrom/obj/bootrom.elf'...
Loading usable ELF segments:
0: V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94
1: V 0x00200000 P 0x00100200 (0x00000d88->0x00000d88) [RWX] @0x298

Waiting for Proxmark to appear on USB...
Connected units:
	1. SN: ? [001/007]
 Found.
Entering bootloader...
Press and hold down button NOW if your bootloader requires it.
Waiting for Proxmark to reappear on USB...........................
Connected units:
	1. SN: ? [001/008]
 Found.
Note: Your bootloader does not understand the new START_FLASH command
      It is recommended that you update your bootloader

Flashing...
Writing segments for file: ../../bootrom/obj/bootrom.elf
 0x00100000..0x001001ff [0x200 / 2 blocks].. OK
 0x00100200..0x00100f87 [0xd88 / 14 blocks].............. OK

Resetting hardware...
All done.

Have a nice day!
```
But I can't seem to complete the second step and flash the main firmware image
```
./flasher ../../armsrc/obj/fullimage.elf 
Loading ELF file '../../armsrc/obj/fullimage.elf'...
Loading usable ELF segments:
0: V 0x00102000 P 0x00102000 (0x00037ec0->0x00037ec0) [R X] @0x94
1: V 0x00200000 P 0x00139ec0 (0x000012f8->0x000012f8) [RW ] @0x37f54
Note: Extending previous segment from 0x37ec0 to 0x391b8 bytes

Waiting for Proxmark to appear on USB..........
Connected units:
	1. SN:  [001/015]
 Found.
write failed: error sending control message: Connection timed out!
Trying to reopen device...

Connected units:
	1. SN:  [001/015]
proxmark3> ReceiveCommandPoll returned 0
Error: Unknown Proxmark mode 
```
Any ideas?

Fl0-0, 2019-02-26:
Hello @bleepnetworks,

With your command you only have flashed the bootloader, so the bootloeder is USB cdc now and no more USB HID.

Flash the firmware the 'new cdc' flasher tool now.

You need to reflash it by holding the button on the Proxmark while connecting it to your computer until flashing is complete (because you flashed only the bootrom before ...)

in the client directory (not in hid-flasher)
`./flasher /dev/ttyACM0 ../armsrc/obj/fullimage.elf`

pwpiwi, 2019-02-26:
Makes sense to me. If this is confirmed, the Wiki definitely needs an update.

bleepnetworks, 2019-02-27:
thanks for the replies, the proxmark is up to date now cheers :D

closed 2019-02-27 by bleepnetworks

pwpiwi, 2019-02-27:
Can you please tell us how the issue could be resolved? Others may want to know as well...

bleepnetworks, 2019-02-28:
Could have been an issue with my virtualised environment, as flashed from my colleagues’ laptop that is known to be good.

Probably just running the wrong flasher as @Fl0-0 pointed out though.