Proxmark/proxmark3 #798: Hitag2 issues in current repo - i2c related?

jslawek, 2019-03-09:
Hi,

hitag2 does not work for me in current repo version (tried RRG too btw).
"lf hitag reader 26" does not recognize tag id (no result in client), reading blocks or simulating also does not work.
I have tried several commits back, and it works for me in https://github.com/Proxmark/proxmark3/commit/968ad672808c4bd4ae0d413bbe877f263e3f8228 but does not work in
5a28b51 (I could not compile these few in between to check).
I have further pinpointed the problem to @pwpiwi commit https://github.com/Proxmark/proxmark3/commit/8d7d7b618777fddcde8897945f2ec42eb85095e2, more specifically " "update i2c.c from RfidResearchGroup repository".
After switching i2c.c in 5a28b51 to the previous one from 968ad67, hitag commands (lf hitag) work for me again.

I confirmed this on Elechouse RDV2 as well as another AT91SAMS512 clone. Interestingly, the RDV4 behaves in a different way however. The exact same fw that recognizes hitag2 UID for RDV2, flashed on RDV4 ends with error:
```code
proxmark3> lf hitag reader 26
#db# Uknown frame length: 2
```

Also, even when the lf hitag commands work, the "lf search" does not recognize the tag anyway. And after this "lf search" with empty result - "lf hitag" commands do not work any more, until the Proxmark device re-plug. But I think this may be another, unrelated issue.

pwpiwi, 2019-03-13:
I don't have a Hitag card at hand but I think that @iceman1001 is right. I have added TC0 initialization to Hitag (and HitagS) functions in PR #800. Please give it a try.

iceman1001, 2019-03-13:
yup yup,  those TC0 inits made the **lf hitag reader 26** come alive again.   It can identify my card.

```
pm3 --> lf hita reader 26
#db# Configured for hitag2 reader
Valid Hitag2 tag found - UID: ac20a810
pm3 --> lf hita list
recorded activity (TraceLen = 512 bytes):
 ETU     :nbits: who bytes
---------+-----+----+-----------
 +      0:    5:     c0
 +    203:   32: TAG ac! 20  a8  10
```

jslawek, 2019-03-13:
Yes, I confirm it works! Not only "lf hitag reader 26" but also reading the whole card and simulating it to the reader, and moreover - "lf search" also recognizes the tag again! 
I am still having difficulties on RDV4 though, but I guess it is unrelated. I begin to suspect it may be somehow related to signal quality (clipping?). It gets various length of data (or no data at all), depending how far I place the tag from antenna:
```
proxmark3> lf hitag reader 26
#db# Uknown frame length: 25          
proxmark3> lf hitag reader 26
#db# Uknown frame length: 31          
proxmark3> lf hitag reader 26
#db# Uknown frame length: 25          
proxmark3> lf hitag reader 26
#db# Uknown frame length: 1          
proxmark3> lf hitag reader 26
proxmark3> lf hitag reader 26
```

pwpiwi, 2019-03-13:
merged.

jslawek, 2019-03-13:
Thanks for such quick resolution! Closing as it works for most Proxmark hardware devices, and the RDV4 signal quality issues (not strictly related to this) moved to RRG repo: https://github.com/RfidResearchGroup/proxmark3/issues/124

closed 2019-03-13 by jslawek