Proxmark/proxmark3 #809: Proxmark3 rdv4.0 create the dump Mifare 4k but when eload again the dump, card type if unknown

hornyd, 2019-03-29:
Proxmark3 rdv4.0 create the dump Mifare 4k but when eload again the dump, card type if unknown

I follow the procedure of proxmark and I got all the keys of the Mifare Classic card. I can read all blocks, I can also create the dumpkeys and data file. all works as expected.

When I am trying to reload the dump.eml file to the proxmark as Mifare 4K and read the trace, it is identified as Mf 1K. Also if I send the sim command, it is recognized as mf 1k

Any suggestion?

pwpiwi, 2019-03-30:
Please post the sequence of commands and their results.

hornyd, 2019-04-01:
pm3 ~$ ./client/proxmark3.exe com11
Prox/RFID mark3 RFID instrument
bootrom: master/v3.1.0-83-g1f61f19-dirty-suspect 2019-03-29 13:47:33
os: master/v3.1.0-83-g1f61f19-dirty-suspect 2019-03-29 13:47:43
fpga_lf.bit built for 2s30vq100 on 2015/03/06 at 07:38:04
fpga_hf.bit built for 2s30vq100 on 2019/03/20 at 08:08:07
SmartCard Slot: available

uC: AT91SAM7S512 Rev B
Embedded Processor: ARM7TDMI
Nonvolatile Program Memory Size: 512K bytes. Used: 205758 bytes (39%). Free: 318530 bytes (61%).
Second Nonvolatile Program Memory Size: None
Internal SRAM Size: 64K bytes
Architecture Identifier: AT91SAM7Sxx Series
Nonvolatile Program Memory Type: Embedded Flash Memory
proxmark3> hf search

 UID : 46 f5 6c e1
ATQA : 00 02
 SAK : 18 [2]
TYPE : NXP MIFARE Classic 4k | Plus 4k SL1
proprietary non iso14443-4 card found, RATS not supported
No chinese magic backdoor command detected
Prng detection: WEAK

Valid ISO14443A Tag Found - Quiting Search

proxmark3> hf mf chk *4 ? d default_keys.dic
--chk keys. sectors:40, block no:  0, key type:B, eml:n, dmp=y checktimeout=471 us
File content error. '´╗┐# Default' must include 12 HEX symbols
chk custom key[ 0] 000000000000
chk custom key[ 1] c0d0c4db331f
chk custom key[ 2] 431c68fe9d40
chk custom key[ 3] 6c6830064652
chk custom key[ 4] 50edff5e4d62
chk custom key[ 5] 33b668ec04e8
chk custom key[ 6] 8782abcdcc1e
chk custom key[ 7] a0fbf5e4587b
….
proxmark3> hf mf dump 4 m
..
#db# READ BLOCK FINISHED
Successfully read block 15 of sector 39.
Dumped 256 blocks (4096 bytes) to file dumpdata.bin
proxmark3> 
proxmark3> hf mf sim u B6FDD1A7
mf 1k sim uid: b6 fd d1 a7 , numreads:0, flags:2 (0x02)
#db# 4B UID: b6fdd1a7

Sequence of the commands:
- hf mf chk *4 ? d default_keys.dic
- hf mf dump 4 m
- hf mf esave
- hf mf sim

Initially the card is detected as 4K, then the dump is successful and all data are stored. But when I try to emulate the memory, it simulate as 1K and SAK 44 00

pwpiwi, 2019-04-01:
Yes, indeed. hf mf sim only simulates 1k cards. Give me a few days to come back to my PM...

iceman1001, 2019-04-08:
@hornyd  There is support for 1k, 2k, 4k simulation in the RRG fork since yesterday,   https://github.com/RfidResearchGroup/proxmark3   Haven't tried it yet.

pwpiwi, 2019-04-08:
Yes, I am back. Please try PR #812.

pwpiwi, 2019-04-09:
The 'm' option is still there. Nothing has been changed for the dump command.

hornyd, 2019-04-09:
I used the link provided and everything went well. The driver installed and firmware also updated.

I got the key file generated but see below when I use the command:

pm3 --> hf mf dump 4 m
[!] Unknown parameter 'm'

Usage:   hf mf dump [card memory] k <name> f <name>
  [card memory]: 0 = 320 bytes (Mifare Mini), 1 = 1K (default), 2 = 2K, 4 = 4K
  k <name>     : key filename, if no <name> given, UID will be used as filename
  f <name>     : data filename, if no <name> given, UID will be used as filename

Examples:
         hf mf dump
         hf mf dump 4
pm3 -->

iceman1001, 2019-04-09:
there is no **"m"** param in an iceman based fork...

hornyd, 2019-04-09:
thank you for your quick answer, clear now.

How can I read the card in the case when blocks 2 to 20 are unused?

pwpiwi, 2019-04-10:
There has been work done in parallel. So both RRG and this repository now support 4K cards. The 'm' option is still available in this repository but not in RRG. 

Changes in RRG have already been committed, in this branch they are in PR #812 for review.

pwpiwi, 2019-04-10:
I have pushed more changes to PR #812 to improve the simulation. NXP Taginfo now recognizes the simulated tag most of the time. I think that a slight speedup of the simulation is still required.

closed 2019-06-21 by pwpiwi