Proxmark/proxmark3 #810: HF 14a raw command is not working fine

chaytoun911, 2019-04-02:
I'm Trying to change UID for a mifare desfire which I lately procured from LAB401 using the described command on the website : 

hf 14a raw -s -c 02 00 ab 00 00 07 11 22 33 44 55 66 77

however I'm receiving a below message from proxmark3

proxmark3> hf 14a raw -s -c 02 00 ab 00 00 07 11 22 33 44 55 66 77
Card selected. UID[7]:          
00 00 00 00 00 00 00           
timeout while waiting for reply.

Did anybody faced this issue before? 

thanks for your support.

pwpiwi, 2019-04-27:
Sorry for late response, this somehow escaped me. I will provide a fix this weekend.

pwpiwi, 2019-04-27:
Hmmm. I cannot find anything to fix. 'hf 14a raw' works as expected. Are you sure that this card is supposed to respond at all to this command? Did you check if its UID has changed as expected?

I propose to ask this question in the forum. You will reach a much wider audience.

pwpiwi, 2019-05-22:
any news? Can we close this issue?

closed 2019-06-21 by pwpiwi