Proxmark/proxmark3 #813: "bash: make: command not found" on windows 10

cds333, 2019-04-09:
This is what I get when I run runme.bat after following the instructions located here: https://github.com/Proxmark/proxmark3/wiki/Windows

The same steps on my windows 7 machine work perfectly.

error: mingw32: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: mingw64: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: msys: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: package 'mingw-w64-x86_64-gcc-libs' was not found
error: mingw32: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: mingw64: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: msys: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: database 'mingw32' is not valid (invalid or corrupted database (PGP signature))
error: database 'mingw64' is not valid (invalid or corrupted database (PGP signature))
error: database 'msys' is not valid (invalid or corrupted database (PGP signature))
loading packages...
error: '/var/cache/pacman/pkg/mingw-w64-x86_64-gcc-libs-8.2.0-3-any.pkg.tar.xz': invalid or corrupted package (PGP signature)
error: mingw32: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: mingw64: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: msys: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: package 'mingw-w64-x86_64-gcc' was not found
error: mingw32: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: mingw64: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: msys: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: database 'mingw32' is not valid (invalid or corrupted database (PGP signature))
error: database 'mingw64' is not valid (invalid or corrupted database (PGP signature))
error: database 'msys' is not valid (invalid or corrupted database (PGP signature))
loading packages...
error: '/var/cache/pacman/pkg/mingw-w64-x86_64-gcc-8.2.0-3-any.pkg.tar.xz': invalid or corrupted package (PGP signature)
error: mingw32: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: mingw64: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: msys: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: package 'mingw-w64-x86_64-readline' was not found
error: mingw32: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: mingw64: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: msys: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: database 'mingw32' is not valid (invalid or corrupted database (PGP signature))
error: database 'mingw64' is not valid (invalid or corrupted database (PGP signature))
error: database 'msys' is not valid (invalid or corrupted database (PGP signature))
loading packages...
error: '/var/cache/pacman/pkg/mingw-w64-x86_64-readline-6.3.008-1-any.pkg.tar.xz': invalid or corrupted package (PGP signature)
error: mingw32: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: mingw64: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: msys: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: package 'git' was not found
error: mingw32: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: mingw64: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: msys: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: database 'mingw32' is not valid (invalid or corrupted database (PGP signature))
error: database 'mingw64' is not valid (invalid or corrupted database (PGP signature))
error: database 'msys' is not valid (invalid or corrupted database (PGP signature))
error: mingw32: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: mingw64: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: msys: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: package 'make' was not found
error: mingw32: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: mingw64: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: msys: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: database 'mingw32' is not valid (invalid or corrupted database (PGP signature))
error: database 'mingw64' is not valid (invalid or corrupted database (PGP signature))
error: database 'msys' is not valid (invalid or corrupted database (PGP signature))
error: mingw32: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: mingw64: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: msys: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: package 'pkg-config' was not found
error: mingw32: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: mingw64: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: msys: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: database 'mingw32' is not valid (invalid or corrupted database (PGP signature))
error: database 'mingw64' is not valid (invalid or corrupted database (PGP signature))
error: database 'msys' is not valid (invalid or corrupted database (PGP signature))
error: mingw32: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: mingw64: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: msys: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: package 'mingw-w64-x86_64-jansson' was not found
error: mingw32: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: mingw64: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: msys: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: database 'mingw32' is not valid (invalid or corrupted database (PGP signature))
error: database 'mingw64' is not valid (invalid or corrupted database (PGP signature))
error: database 'msys' is not valid (invalid or corrupted database (PGP signature))
error: mingw32: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: mingw64: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: msys: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: package 'mingw-w64-x86_64-qt5' was not found
error: mingw32: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: mingw64: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: msys: signature from "Alexey Pavlov (Alexpux) <alexpux@gmail.com>" is unknown trust
error: database 'mingw32' is not valid (invalid or corrupted database (PGP signature))
error: database 'mingw64' is not valid (invalid or corrupted database (PGP signature))
error: database 'msys' is not valid (invalid or corrupted database (PGP signature))

pwpiwi, 2019-04-09:
Your mingw installation is not OK. Do you use @Gator96100 Proxspace?

pwpiwi, 2019-04-26:
You should raise an issue there. Works perfect on my Windows 10 machines.

Gator96100, 2019-04-26:
> 
> 
> Yes; https://github.com/Gator96100/ProxSpace/archive/master.zip
>I found another thread msys2/MINGW-packages#3580 which says this is a server error. I guess it is not unreasonable that the server would be down for maintenance at 01:00
>
>I'll try again soon

Does this error still occur? Did you test it again?

pwpiwi, 2019-05-22:
@cds333 : it would be nice to answer to questions regarding your issue...

closed 2019-06-21 by pwpiwi

AskAlice, 2020-07-15:
if anyone sees this happen, you probably moved your ProxSpace folder after running runme.bat, and need to re-extract your proxspace archive and start fresh