Proxmark/proxmark3 #814: error while loading shared libraries

doortyx, 2019-04-10:
Hi there!
Have a problem with proxmark3:
tried to run: ./proxmark3 /dev/ttyACM0
but I get:
./proxmark3: error while loading shared libraries: libreadline.so.6: cannot open shared object file: No such file or directory
Linux 4.18.0-17-generic #18~18.04.1-Ubuntu x86_64 x86_64 x86_64 GNU/Linux
Any ideas anyone?

Fl0-0, 2019-04-11:
Hello,

You need to install libraries and tools before compiling and using proxmark3 client, for Ubuntu:

https://github.com/Proxmark/proxmark3/wiki/Ubuntu-Linux

in your case: `apt install libreadline5 libreadline-dev` should get rid of this error.

doortyx, 2019-04-11:
Hi, thanks for your help.
The problem in my case is as follows:
libreadline-dev is already the newest version (7.0-3)
libreadline5 is already the newest version (5.2+dfsg-3build1)
Should I uninstall libreadline7 and try to install 5 instead?

doortyx, 2019-04-11:
I managed to solve it using this video:
https://www.youtube.com/watch?v=_bae1v0o_JA
Worked in my case.

closed 2019-04-11 by doortyx