Proxmark/proxmark3 #819: ProxMark3 bricked after flashing fullimage.elf

BlackFlag1337, 2019-05-07:
Hi, 
I have problem...
Yesterday i wanted to pass my pm3 in v3.0.1, arrived at this command: sudo ./flasher / dev / ttyACM0 -b ../armsrc/obj/fullimage.elf
My pm3 is no longer detected and the led "c" orange remains lit, thank you very much for your answers.
i paid pm3 easy to aliexpress and he was walking before the flash :(

MrButlerTron, 2019-05-07:
> sudo ./flasher / dev / ttyACM0 **-b** ../armsrc/obj/fullimage.elf
The -b says that you are flashing the bootloader but you gave it the fullimage(FPGA+something else).  When you flashed it the second time you omitted the -b and gave it the full image which would be correct, but you still may have the wrong code in the bootloader.

Try
`sudo ./flasher /dev/ttyACM -b ../bootrom/obj/bootrom.elf ../armsrc/obj/fullimage.elf`

iceman1001, 2019-05-07:
If you messed up bootrom,   you will need to JTAG it.

BlackFlag1337, 2019-05-08:
> The -b says that you are flashing the bootloader but you gave it the fullimage(FPGA+something else). When you flashed it the second time you omitted the -b and gave it the full image which would be [...]

i try and result : 
root@bf:~/proxmark3/client# sudo ./flasher /dev/ttyACM -b ../bootrom/obj/bootrom.elf ../armsrc/obj/fullimage.elf
Loading ELF file '../bootrom/obj/bootrom.elf'...
Loading usable ELF segments:
0: V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94
1: V 0x00200000 P 0x00100200 (0x00000c84->0x00000c84) [R X] @0x298

Loading ELF file '../armsrc/obj/fullimage.elf'...
Loading usable ELF segments:
0: V 0x00102000 P 0x00102000 (0x0002ecb0->0x0002ecb0) [R X] @0x94
1: V 0x00200000 P 0x00130cb0 (0x000018c8->0x000018c8) [RW ] @0x2ed44
Note: Extending previous segment from 0x2ecb0 to 0x30578 bytes

Waiting for Proxmark to appear on /dev/ttyACM ........................................................................................................................
ERROR: invalid serial port
Could not find Proxmark on /dev/ttyACM.

BlackFlag1337, 2019-05-08:
> If you messed up bootrom, you will need to JTAG it.

no way to debrick without jtag?

pwpiwi, 2019-05-08:
If your Proxmark isn't **very** old, there is no need to risk flashing the bootloader at all. Most of the time it is sufficient to flash the firmware (fullimage.elf) only.

In your last try you had `/dev/ttyACM` instead of `/dev/ttyACM0` (if this is the serial port of your proxmark). 

If you press the PM3 button before connecting it to the PC (i.e. during power up) and keep it pressed, it will go directly to flash mode without starting the firmware. This is required when your PM3 firmware (fullimage.elf) is broken. It may be worth to check which serial port it connects when you do this.

If you really destroyed the bootloader, then yes, you will need to fix it via JTAG.

MrButlerTron, 2019-05-08:
Triple check the commands before you run them, it looks like you keep omitting parts of the command before running it. 

Type it out, check it, check it again, and then try to connect the Proxmark with the button held down as @pwpiwi suggested.

pwpiwi, 2019-05-08:
> The -b says that you are flashing the bootloader

Not true. The `-b` only enables flashing the bootloader. Fullimage.elf will not overwrite the bootloader.

pwpiwi, 2019-05-08:
Did you read the Wiki? ModemManager? HID-device?

BlackFlag1337, 2019-05-08:
> If you really destroyed the bootloader, then yes, you will need to fix it via JTAG.

okay do u have link to fix via jtag ?

MrButlerTron, 2019-05-08:
> > The -b says that you are flashing the bootloader
> 
> Not true. The `-b` only enables flashing the bootloader. Fullimage.elf will not overwrite the bootloader.

Ah, my mistake.  thank you for correcting me.

pwpiwi, 2019-05-08:
https://github.com/Proxmark/proxmark3/wiki/Debricking-Proxmark3-with-buspirate
https://github.com/Proxmark/proxmark3/wiki/De-Bricking-Segger

micolous, 2019-05-10:
I had saw the same symptoms on a PM3 bought from a Chinese online marketplace that came with `Modify by Willok(willok@163.com)` firmware (`2016-11-14 03:06:26`) + bootloader (`2016-11-09 00:59:56`): flashing v3.1.0 fullimage.elf resulted in the device no longer present over USB.

I fixed it by putting into bootloader mode, holding down the button and keep holding it while bootloader flashing was in progress.  It worked fine after the v3.1.0 bootloader was on there.

I don't know what bootloader they ship these things with, but it's clearly totally broken. They also don't provide source code as required by the license, or respond to requests for source code, so you have plenty of rights to open a dispute IMHO.

It's really trivial to build the firmware properly and comply with the license, I don't understand how these places mess it up.

pwpiwi, 2019-05-22:
@BlackFlag1337  : how did you proceed?

closed 2019-06-21 by pwpiwi