Proxmark/proxmark3 #820: Proxmark 3 can't read new iClass 2000 cards

Supercodegames, 2019-05-10:
It seems the newer iClass 2000 cards (very slim, not clamshell)  cannot be read by the Proxmark3
When trying to read, the  voltage drop when doing `hw tune`, but it returns "no known card found".

Is there a planned fix for this?

Threads with this issue:
http://www.proxmark.org/forum/viewtopic.php?id=5307
http://www.proxmark.org/forum/viewtopic.php?id=5798
http://www.proxmark.org/forum/viewtopic.php?id=6223
http://www.proxmark.org/forum/viewtopic.php?id=6311
http://www.proxmark.org/forum/viewtopic.php?id=6144

pwpiwi, 2019-05-10:
First someone needs to identify the root cause. Only then we can think about a fix.

Looks like no one has a clue up to now.

Supercodegames, 2019-05-12:
It seems its mainly the new cards with the serial number with a +
The old ones with a * work fine

Supercodegames, 2019-05-12:
HID surprised the Proxmark community when it quietly refreshed its iClass card.

We all assumed it was a "bug" but its seems its a whole new card card that the proxmark can't read. As of now, the Proxmark only supports the legacy card.

The legacy cards start with a *
The newer cards start with a +

The new mac calculation might be what's causing the issue (don't know for sure though).

pwpiwi, 2019-05-14:
I thought that iClass uses ISO15693 communication, instead the table talks about ISO14443B? Do we know at all which communication standard is used by the new "+" cards?

Supercodegames, 2019-05-17:
I think iClass support both, but maybe the + only support ISO15693

pwpiwi, 2019-05-17:
... but then it should work with the PM3.

Supercodegames, 2019-05-20:
Updated OP. Its not new cards, it happens by certain programmed and configured card running under this “non-ISO14443B”.  Someone has to figure out the codec...

pwpiwi, 2019-05-21:
Can you please check if the card responds to an ISO15693 inventory request? Try
```
hf 15 cmd raw -c 26 01 00
hf plot
```
and check the graph. If it contains more than noise, please post it here.

pwpiwi, 2019-06-21:
any results?

vpr1982, 2019-07-04:
Hi pwpiwi! 
Since TS had seemingly abandoned this issue here's my result.

Looks like some problem with the command.

proxmark3> hf 15 cmd raw -c 26 01 00
received 0 octets          
          
proxmark3> hf plot
#db# unknown command:: 0x0801          
Waiting for a response from the proxmark...          
You can cancel this operation by pressing the pm3 button

marshmellow42, 2019-07-04:
It should be 
data plot

vpr1982, 2019-07-04:
I can tell it should. But that's what it is in my case.

pwpiwi, 2019-07-04:
It is indeed "hf plot". You have to update your firmware if it is yet unknown.

vpr1982, 2019-07-04:
proxmark3> hf 15 cmd raw -c 26 01 00
received 0 octets          
          
proxmark3> hf plot
proxmark3> 

Where do I look for the graph?

pwpiwi, 2019-07-11:
In the graph window. The same window as with 'data plot'.

patrickdk77, 2019-07-18:
Not sure if this is the same card, the only id is HID iCLASS Px G9P XT, but it doesn't show up in proxmark3 with anything I have done
[image]

pwpiwi, 2019-07-25:
Did you scroll to the right?

edlovesiraq, 2019-10-15:
Any further work on this? I think i have the same issue. I can see the card under HF tune. writing on the side is HID iCLASS DL, but when I do hf search, or hf iclass reader i get nothing. I tried the commands listed but they don't seem to be the right syntax in the latest build. 
proxmark3> hf 15 cmd raw -c 26 01 00
received 0 octets

proxmark3> hf plot
proxmark3>
I tried in pm3, and tried with proxmark3, no dice either time.  Open to testing this issue please let me know what else I can do. New to the community and hoping to be value added.

pwpiwi, 2019-10-21:
OK, I am a bit further down the road. Please try latest changes including PR #876. And I stand corrected, the correct commands to check for an iClass ISO15693 card is 
```
hf 15 cmd raw 0c
hf plot
```
and you should see at least the tag response to the right and eventually the reader command or parts of it to the left - if the card answers at all.

You may also try 'hf iclass reader' again. I have improved it a lot. Be aware what is lying on your table because the PM3 now even reads iClass cards lying *beside* the antenna. The RDV4 needs a minimum distance between antenna and card (maybe because of the wild oscillations caused by the high Q antenna - if you look at the plot you know what I mean).

[image]

pwpiwi, 2019-10-23:
Again wrong command. Previous post edited. Correct command is
```
hf 15 cmd raw 0a
hf plot
```

patrickdk77, 2019-11-12:
Sorry, I bricked my proxmark attempting to load this firmware somehow. just got it fixed up.

I'm not seeing any changes in the hf plot on mine but the results of the commands with the PR #876 are:
proxmark3> hf 15 cmd raw 0a
received SOF only. Maybe Picopass/iCLASS?         

proxmark3> hf iclass reader
   CSN: b8 c4 7c 0f ff ff 12 e0           
    CC: ff ff ff ff 63 ff ff ff           
  Mode: Application [Locked]          
Coding: ISO 14443-2 B/ISO 15693          
 Crypt: Secured page, keys not locked          
    RA: Read access not enabled          
   Mem: 16 KBits/16 App Areas (255 * 8 bytes) [1F]          
   AA1: blocks 06-12          
   AA2: blocks 13-FF          
 AppIA: ff ff ff ff ff ff ff ff           
      : Possible iClass (legacy tag)          
   CSN: b8 c4 7c 0f ff ff 12 e0           
    CC: ff ff ff ff 63 ff ff ff           
  Mode: Application [Locked]          
Coding: ISO 14443-2 B/ISO 15693          
 Crypt: Secured page, keys not locked          
    RA: Read access not enabled          
   Mem: 16 KBits/16 App Areas (255 * 8 bytes) [1F]          
   AA1: blocks 06-12          
   AA2: blocks 13-FF          
 AppIA: ff ff ff ff ff ff ff ff           
      : Possible iClass (legacy tag)

pwpiwi, 2019-11-12:
This looks pretty much like a standard legacy iClass card. I am wondering why this should have created issues?

aaronmaxlevy, 2019-11-12:
@pwpiwi HID is no longer using PicoPass chips for iCLASS, as they have been discontinued. If you take a look at the chip in a current card, it looks like something custom from NXP, likely running a “PicoPass Emulator” of sorts from HID. My guess is perhaps there are differences in timing, etc. Between those and real PicoPass cards.

patrickdk77, 2019-11-12:
My issue might be unrelated, but when I initially looked up the issue I had, it seemed like this was related.
My card says on it
HID iCLASS Px G9P XT
I can read the hid lf part fine, but have not been able to read the iclass part of the card.

pwpiwi, 2019-11-27:
I am not sure about the current status of this issue and if it can be closed. @patrickdk77: you report "have not been able to read the iclass part of the card" but shows a perfect read two posts above. Does it mean that the issue is fixed for you?

Any feedback from @Supercodegames, @vpr1982 or @edlovesiraq ?

steve-embling, 2019-11-27:
So I also have a card that suffers from this and can confirm that this patch (for me at least) fixes
```
hf search
```
and 
```
hf iclass readblk
```
But I haven't had a chance to confirm other read commands or writing yet.

steve-embling, 2020-01-14:
@pwpiwi In response to the other thread, I did have a card that could not be read with proxmark (0/20 hf search detected the card) to one that reliably reads (10/10) on the same card, after updating to the version given here.

I can't say for sure my issue was the same as the original poster, but it looks to be closed to me as the scope of this issue only covered reading of the cards.

Also - thank you for all the work on this.

closed 2020-01-14 by pwpiwi

vpr1982, 2020-03-04:
This patch fixes this problem. New issue #907