Proxmark/proxmark3 #821: Added lf paradox clone to clone a Paradox card

adk47, 2019-05-14:
For some reason the "lf paradox clone" commands are missing, but in your change log you are saying it is there.

marshmellow42, 2019-05-14:
Make sure you don't mix different client and firmware versions.

That error is a common indication that this is what you may have done.  I see that command exists on the current codebase.

adk47, 2019-05-14:
I only use the command line tool.  The new boot loader doesn’t work in the latest master repo, but the image I have imaged on PM3 is definitely the latest master version.  I see your code in master but no reference to the new command you created.  Can you help me realize what I’ve done wrong?

marshmellow42, 2019-05-14:
This is better suited discussion for the pm3 forum.  It seems you have some learning to do, and the forum is the best place for that.

Ps. You are not using the command properly.

adk47, 2019-05-14:
The command doesn’t exist.  

proxmark3> hw version
Prox/RFID mark3 RFID instrument          
bootrom: master/v3.1.0-68-g0b6efd0-suspect 2019-05-13 23:55:38
os: /-suspect 2019-05-13 23:43:15
fpga_lf.bit built for 2s30vq100 on 2015/03/06 at 07:38:04
fpga_hf.bit built for 2s30vq100 on 2019/03/20 at 08:08:07
SmartCard Slot: not available
          
uC: AT91SAM7S512 Rev B          
Embedded Processor: ARM7TDMI          
Nonvolatile Program Memory Size: 512K bytes. Used: 206245 bytes (39). Free: 318043 bytes (61).          
Second Nonvolatile Program Memory Size: None          
Internal SRAM Size: 64K bytes          
Architecture Identifier: AT91SAM7Sxx Series          
Nonvolatile Program Memory Type: Embedded Flash Memory          
proxmark3> lf paradox
help             This help          
demod            Demodulate a Paradox FSK tag from the GraphBuffer          
read             Attempt to read and Extract tag data from the antenna          
proxmark3> lf paradox clone
help             This help          
demod            Demodulate a Paradox FSK tag from the GraphBuffer          
read             Attempt to read and Extract tag data from the antenna  

Aaron Kaufman
CEO & Chairman
OpenMed, Inc.
78 SW 7th St. Suite 500
Miami, FL 33131

786-300-5289
aaron@openmed.com

adk47, 2019-05-14:
[image]

https://cl.ly/66d0bac23a64

Command doesn't exist @marshmellow42

adk47, 2019-05-14:
@marshmellow42 i double checked everything all the way through to your code.  I have the correct bootloader, firmware for the latest master clone and the "clone" command is not available.

https://github.com/Proxmark/proxmark3/blob/5f84531b82309c5cbb72ea5bdaaee3a1be734eb5/client/cmdlfparadox.c#L124

Compile the master branch, load firmware on PMS and you will not see the clone command as you have in your list of commands.  I think your new commands were not tied into the main app.

iceman1001, 2019-05-14:
just for testing,
Try pulling latest source from this repo,   compile,  flash,   and run using the compiled client.

adk47, 2019-05-14:
@iceman1001 thanks for the advice - I use a mac, the only client I have is the terminal.  Am I doing something wrong.  Pretty sure the connections for a few functions are not wired up in main.
[image]

pwpiwi, 2019-05-14:
'lf paradox clone' is not in the 3.1.0 release. You have to tell homebrew to use the latest (yet unreleased) code:
```
brew install --HEAD proxmark3
```

adk47, 2019-05-14:
You will see from the following I cloned the latest Master in this thread.
Can you take one more look - if you compile from head manually (like i did)
and see the hw version the command still isn't available.

pwpiwi, 2019-05-15:
I am compiling from head on a daily basis. The command is there:
```
~/github/proxmark3$ client/proxmark3.exe com6
Prox/RFID mark3 RFID instrument
bootrom: mod_list_mf/v3.0.1-341-g9bdd402-dirty-suspect 2018-02-25 11:29:01
os: update_mfu/v3.1.0-89-gd59a643-dirty-suspect 2019-05-08 20:14:37
fpga_lf.bit built for 2s30vq100 on 2015/03/06 at 07:38:04
fpga_hf.bit built for 2s30vq100 on 2019/03/20 at 08:08:07
SmartCard Slot: not available

uC: AT91SAM7S512 Rev B
Embedded Processor: ARM7TDMI
Nonvolatile Program Memory Size: 512K bytes. Used: 204938 bytes (39%). Free: 319350 bytes (61%).
Second Nonvolatile Program Memory Size: None
Internal SRAM Size: 64K bytes
Architecture Identifier: AT91SAM7Sxx Series
Nonvolatile Program Memory Type: Embedded Flash Memory
proxmark3> lf paradox clone 1234567
#db# DONE!
proxmark3>
```
I agree with @marshmellow42 that you should ask this question in the forum. And you should show us there what you are doing. Otherwise no one would be able to find out what you are doing wrong.

closed 2019-05-15 by pwpiwi