Proxmark/proxmark3 #827: SIM/SAM Reader

cefedrific, 2019-05-28:
Hello all.is the sim / sam reader compatible with this product?
https://www.aliexpress.com/item/Upgrade-proxmark3-develop-suit-Kits-5-0-proxmark-RDV4-NFC-RFID-reader-writer-for-rfid-card/32986761218.html?spm=2114.search0104.3.24.30d3544cHU6kW8&ws_ab_test=searchweb0_0,searchweb201602_7_10065_10130_10068_10890_10547_319_10546_317_10548_10696_10190_453_10084_454_10083_10618_10307_10820_10303_537_536_10059_10884_10887_321_322_10103,searchweb201603_51,ppcSwitch_0&algo_expid=e5bd9231-8339-408f-91dd-4302364aaa85-3&algo_pvid=e5bd9231-8339-408f-91dd-4302364aaa85

and if so i would like to know if i can decrypt lyovel smart cards for coffee?

assasinfil, 2019-05-28:
sim card slot not included

cefedrific, 2019-05-29:
hello and thank you for answering me. I know that the sim / sam reader is not included but if I buy it apart is it compatible with this product?

merlokk, 2019-05-29:
short answer - no. its not and rdv4 device and it dont have integrated reader.
long - this repository compatible with any of pcsc reader.

cefedrific, 2019-05-29:
sorry answer too short I do not understand well.The sim reader is not compatible so I understood correctly.
 Could someone give me a link to buy a proxmark3 of good quality allowing me to read several frequencies and lyovel chips (coffee) if possible but not a link to the last pm3 rdv4 because it is too pricey for me :(  although I like it a lot. I am looking for a proxmark3 correct but it is difficult with the many version that I find on aliexpress or amazon I do not know which to choose. thank you for your answers.

pwpiwi, 2019-05-31:
The SIM/SAM extension for the Proxmark RDV4 is just an adapter to connect Smartcards (Contact Cards) to the integrated SIM slot. Only the RDV4 has an integrated SIM slot.

You can use a standard USB Smartcard Reader with the Proxmark software. The reader needs to support the PCSC protocol. You don't need any Proxmark Hardware for the Smartcard functionality. If price is an issue and you only want to work on Smartcards, you can buy all you need for less than 12€.

cefedrific, 2019-05-31:
Bonjour et merci de votre reponse.
un lecteur de carte a puce standard pourrait lire les donnéés mais je  ne pourrais pas ecrire des nouvelles por changer le montant?
Je possede le acr122u depuis quelque année et je modifie le montant des mifares et vu que les machines ont changeé ce n'est  plus des tags mais des cartes a puces donc j aimerais effectuer le meme travail.
je pensais que pour faire cela il fallait le proxmark3 mais a priori non.
avez-vous un lien pour l achat de ce lecteur?

cefedrific, 2019-05-31:
lecteur Gemalto IDBridge CT30 + gscriptor software?

pwpiwi, 2019-05-31:
It is up to you to research. The Software allows you to send raw commands (APDUs) to the card and see the answer. There are no other commands to write to smartcards yet.

Google for PCSC reader.

Side note: I don't think that you can manipulate a properly designed smartcard. And if you can, you should be able to earn enough money in your job to afford some coffees.

cefedrific, 2019-05-31:
merci  et  je pense qu il est possible de modifier le montant d une simple carte a puce pour le café pour un simple ouvrier passionné par l informatiquesi biensur si les données sont stockees coté client sinon j attaquerai le serveur ;)

pwpiwi, 2019-06-21:
Is your question answered? Can we close this issue?

closed 2019-08-12 by pwpiwi