Proxmark/proxmark3 #834: T55xx finalise

assasinfil, 2019-06-13:
Need finalise option for t55xx chip.
My intercom does not accept a clone, apparently checks the chip for overwriting.

assasinfil, 2019-06-13:
Or need instruction for set password

mwalker33, 2019-06-13:
Have a look at the data sheet for specific.
Set the password in block 7 page 0, normal write.
Set the max blocks as needed and make the password active by setting the 
 bits in the config block 0 page 0
Once set remember to use the password for future writes.

merlokk, 2019-06-13:
how to set password in the t5577 datasheet. some systems checks for clones in very strange way. sometimes needs to change data in block0 and card will work as original.
to check - needs to record exchange between t5577 and reader

mwalker33, 2019-06-24:
@assasinfil did you get this working as needed ? I dont think there is an issue with the software, just what you may need to do to make the reader happy with the T5577.

the-d-kid, 2019-06-30:
to activate the password on T5577 use this comands
lf t55xx write b 7 d [password(8hex symbols)]
lf t55xx write b 0 d 00148050
if you need to disable pass use it
lf t55xx write b 0 d 00148040 p [password]

mwalker33, 2019-06-30:
the block 0 data will need to match that needed for the application.  so I would read the current block 0 and set the password bit.

assasinfil, 2019-07-06:
> to activate the password on T5577 use this comands
> lf t55xx write b 7 d [password(8hex symbols)]
> lf t55xx write b 0 d 00148050
> if you need to disable pass use it
> lf t55xx write b 0 d 00148040 p [password]

That is, for a block of 0, I just read what is written there and overwrite with a password?

mwalker33, 2019-07-06:
On a t5577 page 0 block 7 is where the password must be stored if used.  If you need to send out data from block 7 then you wont be able to set a password.
e.g.
… Set the password you want to use (and remember it)
lf t55xx write b 7 d [password(8hex symbols)]

… Tell the t5577 to use the password. This is the block 0 write.  But an example that may or may not be what you need for your chip setup. 
lf t55xx write b 0 d 00148050

if you need to disable pass use it
lf t55xx write b 0 d 00148040 p [password]

The password is enabled by setting is the 5th last bit (28) , the other bits need to be set for you application.
Can I suggest you look at the datasheet (page 9 to start)
http://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-9187-RFID-ATA5577C_Datasheet.pdf

Then try to work out what to bit to change

If unsure, please post the output from
lf t55xx read b 0
And what you think it should look like to enable the password, someone will confirm for you.
(this way, you will learn what to do for other projects)

assasinfil, 2019-07-07:
And when will the em cloning with a password appear?

assasinfil, 2019-07-07:
It’s just that the whole problem started with the fact that I wanted to copy my key for the intercom (em 410) on t5577, but the intercom refuses to accept it, although they are identical. Intercom Fermax

mwalker33, 2019-07-08:
I understand your challenge.  
First up, I don't think this is a software issue and more a how to question.
There are many unknowns here so what will and wont work will be found through testing and trying some things.
A few key unknowns.
1. What is the original "key".  you call it an EM, but not more data.  What does the proxmark detect it as (lf search) an "EM4100" or something different ?
2. Why wont the reader accept the card.
- Does the original card look the same as the clone on the proxmark from the lf search of both the original and clone.
3. While I believe this is about learning and less about telling people how to do something, we still cant tell you the exact command to set the password (which might help), as we don't know the config data in block 0, sending a "random" config to block 0 may make the problem worse.
so, post the output from "lf t55xx detect" and lf t55 read b 0 

From my perspective, the T5577 cards don't have a finalize as such.  What you can do is change the config, to make it harder to detect it as a T5577, but your challenge may be that your clone is not setup up correctly yet.  i.e. We dont know how it was setup or how it is meant to look.

We are trying to help you work the problem and find the solution.

assasinfil, 2019-07-08:
can move this question to the forum?

assasinfil, 2019-07-08:
Or maybe the key fob with the number on the case is read as t5577?

mwalker33, 2019-07-08:
That does not look right.  
1. You got info with at password and it says password is set.
2. Some of the bits in the config should be 0 (fixed) and they are 1, again indication not correct.
Most likely the lf t55 detect was not run prior.

So I repeat again:
post the output from "lf t55xx detect" and "lf t55 read b 0"

assasinfil, 2019-07-08:
I tried the LF sim mode and let me intercom, but when I copy the key fob, the intercom does not allow.

merlokk, 2019-07-08:
needs to sniff how intercom see that here not its key.
basically, it needs to change some parameters in block0 of t5577 and it not see this) but what parameters - depends on what intercom to do to check

assasinfil, 2019-07-08:
My intercom is FERMAX, but lf sim working to open door

merlokk, 2019-07-08:
and `lf t55 dump`

assasinfil, 2019-07-08:
```
proxmark3> lf t55 dump
Reading Page 0:
blk | hex data | binary
----+----------+---------------------------------
  0 | 836252FF | 10000011011000100101001011111111
  1 | 013EC1B1 | 00000001001111101100000110110001
  2 | F1C8004F | 11110001110010000000000001001111
  3 | 1297FC72 | 00010010100101111111110001110010
  4 | FB06C4A5 | 11111011000001101100010010100101
  5 | 40027D83 | 01000000000000100111110110000011
  6 | FFC72001 | 71111111110001110010000000000001
  7 | 36252FF8 | 00110110001001010010111111111000
Reading Page 1:
blk | hex data | binary
----+----------+---------------------------------
  0 | 09F60D89 | 00001001111101100000110110001001
  1 | 1C8004FB | 00011100100000000000010011111011
  2 | 297FC720 | 00101001011111111100011100100000
  3 | 60D894BF | 01100000110110001001010010111111
proxmark3> lf t55 dump
Reading Page 0:
blk | hex data | binary
----+----------+---------------------------------
  0 | 836252FF | 10000011011000100101001011111111
  1 | 013EC1B1 | 00000001001111101100000110110001
  2 | F1C8004F | 11110001110010000000000001001111
  3 | 1297FC72 | 00010010100101111111110001110010
  4 | FB06C4A5 | 11111011000001101100010010100101
  5 | 40027D83 | 01000000000000100111110110000011
  6 | FFC72001 | 71111111110001110010000000000001
  7 | 6C4A5FF1 | 01101100010010100101111111110001
Reading Page 1:
blk | hex data | binary
----+----------+---------------------------------
  0 | 13EC1B12 | 00010011111011000001101100010010
  1 | 1C8004FB | 00011100100000000000010011111011
  2 | 297FC720 | 00101001011111111100011100100000
  3 | C1B1297F | 11000001101100010010100101111111
proxmark3> lf t55 dump
Reading Page 0:
blk | hex data | binary
----+----------+---------------------------------
  0 | C8004FB0 | 11001000000000000100111110110000
  1 | 97FC7200 | 10010111111111000111001000000000
  2 | 06C4A5FF | 00000110110001001010010111111111
  3 | 013EC1B1 | 00000001001111101100000110110001
  4 | F1C8004F | 11110001110010000000000001001111
  5 | 1297FC72 | 00010010100101111111110001110010
  6 | F60D894B | 11110110000011011000100101001011
  7 | 8004FB06 | 10000000000001001111101100000110
Reading Page 1:
blk | hex data | binary
----+----------+---------------------------------
  0 | 7FC72001 | 01111111110001110010000000000001
  1 | 6C4A5FF1 | 01101100010010100101111111110001
  2 | 13EC1B12 | 00010011111011000001101100010010
  3 | 390009F6 | 00111001000000000000100111110110
proxmark3> lf t55 dump
Reading Page 0:
blk | hex data | binary
----+----------+---------------------------------
  0 | 297FC720 | 00101001011111111100011100100000
  1 | B06C4A5F | 10110000011011000100101001011111
  2 | 0027D836 | 00000000001001111101100000110110
  3 | FE390009 | 11111110001110010000000000001001
  4 | 6252FF8E | 01100010010100101111111110001110
  5 | 9F60D894 | 10011111011000001101100010010100
  6 | C8004FB0 | 11001000000000000100111110110000
  7 | 2FF8E400 | 00101111111110001110010000000000
[... 97 lines cut ...]
----+----------+---------------------------------
  0 | 04FB06C4 | 00000100111110110000011011000100
  1 | 8E40027D | 10001110010000000000001001111101
  2 | 94BFE390 | 10010100101111111110001110010000
  3 | B06C4A5F | 10110000011011000100101001011111
proxmark3> lf t55 dump
Reading Page 0:
blk | hex data | binary
----+----------+---------------------------------
  0 | 0009F60D | 00000000000010011111011000001101
  1 | FF8E4002 | 11111111100011100100000000000010
  2 | D894BFE3 | 11011000100101001011111111100011
  3 | 27D83625 | 00100111110110000011011000100101
  4 | 390009F6 | 00111001000000000000100111110110
  5 | 52FF8E40 | 01010010111111111000111001000000
  6 | 60D894BF | 01100000110110001001010010111111
  7 | 004FB06C | 00000000010011111011000001101100
Reading Page 1:
blk | hex data | binary
----+----------+---------------------------------
  0 | FC720013 | 11111100011100100000000000010011
  1 | C4A5FF1C | 71000100101001011111111100011100
  2 | 7D836252 | 01111101100000110110001001010010
  3 | 90009F60 | 10010000000000001001111101100000
```

assasinfil, 2019-07-08:
lf t55 detect not working if password setuped on the key

assasinfil, 2019-07-08:
And how do I record the communication between the intercom and the key?

merlokk, 2019-07-08:
`lf sniff`
if you set a password for t5577 - just remove it or put it to dump command. (very carefully!!!!)

assasinfil, 2019-07-08:
[lfsnoopwitchkeyandintercom(raw).zip](https://github.com/Proxmark/proxmark3/files/3369649/lfsnoopwitchkeyandintercom.raw.zip)

assasinfil, 2019-07-08:
its not zip, its dump from data dump

the-d-kid, 2019-07-08:
lol, I have the same doorphones, but everything works fine

assasinfil, 2019-07-08:
how you clone to t5577?

the-d-kid, 2019-07-08:
yes, i use T5577
the sequence of my actions is
lf em 410xread (read original tag)
lf em 410xwrite [id] 1 (write clone)
and set pass:
lf t55xx write b 7 d [password(8hex symbols)]
lf t55xx write b 0 d 00148050
To be honest, I did not check if the copy works without a password, but I write on T5577
maybe my intercom has other settings or another model
here is a photo of my panel: https://pp.userapi.com/c858120/v858120643/f433/1r9vLZUiV1s.jpg

assasinfil, 2019-07-08:
I did exactly the same thing, but the intercom does not accept, the panel looks like the photo.
[http://www.fermax-russia.ru/scripts/timthumb.php?src=//images/site/7387.jpg&w=326&h=490&zc=2](domophone)

assasinfil, 2019-07-08:
```
proxmark3> lf t55 dump 65857569 o
Reading Page 0:
blk | hex data | binary
----+----------+---------------------------------
Safety Check Overriden - proceeding despite risk
  0 | 00148050 | 00000000000101001000000001010000
Safety Check Overriden - proceeding despite risk
  1 | FF8E4002 | 11111111100011100100000000000010
Safety Check Overriden - proceeding despite risk
  2 | 7D836252 | 01111101100000110110001001010010
Safety Check Overriden - proceeding despite risk
  3 | FFFFFFFF | 11111111111111111111111111111111
Safety Check Overriden - proceeding despite risk
  4 | FFFFFFFF | 11111111111111111111111111111111
Safety Check Overriden - proceeding despite risk
  5 | FFFFFFFF | 11111111111111111111111111111111
Safety Check Overriden - proceeding despite risk
  6 | FFFFFFFF | 11111111111111111111111111111111
Safety Check Overriden - proceeding despite risk
  7 | B2C2BAB4 | 10110010110000101011101010110100
Reading Page 1:
blk | hex data | binary
----+----------+---------------------------------
Safety Check Overriden - proceeding despite risk
  0 | 00148050 | 00000000000101001000000001010000
Safety Check Overriden - proceeding despite risk
  1 | E03900D0 | 11100000001110010000000011010000
Safety Check Overriden - proceeding despite risk
  2 | B82D4773 | 10111000001011010100011101110011
Safety Check Overriden - proceeding despite risk
  3 | 00A00003 | 00000000101000000000000000000011
```

assasinfil, 2019-07-08:
maybe because of page 1, the intercom does not start up, because it sees another code?
```
proxmark3> lf t55 dump
Reading Page 0:
blk | hex data | binary
----+----------+---------------------------------
  0 | 7D836252 | 01111101100000110110001001010010
  1 | 7D836252 | 01111101100000110110001001010010
  2 | 7D836252 | 01111101100000110110001001010010
  3 | 7D836252 | 01111101100000110110001001010010
  4 | 7D836252 | 01111101100000110110001001010010
  5 | 7D836252 | 01111101100000110110001001010010
  6 | 7D836252 | 01111101100000110110001001010010
  7 | 7D836252 | 01111101100000110110001001010010
Reading Page 1:
blk | hex data | binary
----+----------+---------------------------------
  0 | B82D4773 | 10111000001011010100011101110011
  1 | B82D4773 | 10111000001011010100011101110011
  2 | B82D4773 | 10111000001011010100011101110011
  3 | B82D4773 | 10111000001011010100011101110011
```

merlokk, 2019-07-08:
00148040 RF/64, Manchester. all is ok
ill look tomorrow at lf snoop

mwalker33, 2019-07-08:
it looks like you did an lf em 410xwrite <id> 1, so a standard em4100
You set a password, so while that is set, you need to provide a password for every lf t55 command.
I would NOT expect to see any config in block 3 page 1
As shown after a valid t55 detect and t55 dump (both with password)
block 3 page 1 :  00A00003 - but that looks invalid, so lets clear (even though it should not be active).

Assuming the password is still set and is 65857569.
lf t55 write b 3 1 d 00000000 p  65857569 

Is the ID : 19004FC3C9
It also my be sending out the Page 1 data an not the page 0 data for some reason/condition, so lets set that to the same EM ID.  (hinted by the data in the dump without the password, when a password is set and you use a t5577 command without the password, it will send out the "default read" data, which in this case should be the EM data)
Again, assuming the password is still set
lf t55 write b 1 1 d FF8E4002 p  65857569 
lf t55 write b 2 1 d 7D836252 p  65857569 

So three commands to run (and why).  Then check the writes are ok with a new card dump
lf t55 detect p  65857569
lf t55 dump p  65857569 o

if that looks ok, (block 1  page 1 = block 1 page 0, and block 2 page 1 = block 2 page 0)
Check the ID is correct  via lf search, run this on the original AND the clone, the results should be 100% the same, if not show both outputs

Place the original on the pm3
lf search
Place the clone on the pm3
lf search

If all looks ok, they on the reader.

assasinfil, 2019-07-08:
In theory, now everything is correct and should work.
```
proxmark3> lf t55 dump
Reading Page 0:
blk | hex data | binary
----+----------+---------------------------------
  0 | 7D836252 | 01111101100000110110001001010010
  1 | 7D836252 | 01111101100000110110001001010010
  2 | 7D836252 | 01111101100000110110001001010010
  3 | 7D836252 | 01111101100000110110001001010010
  4 | 7D836252 | 01111101100000110110001001010010
  5 | 7D836252 | 01111101100000110110001001010010
  6 | 7D836252 | 01111101100000110110001001010010
  7 | 7D836252 | 01111101100000110110001001010010
Reading Page 1:
blk | hex data | binary
----+----------+---------------------------------
  0 | 7D836252 | 01111101100000110110001001010010
  1 | 7D836252 | 01111101100000110110001001010010
  2 | 7D836252 | 01111101100000110110001001010010
  3 | 7D836252 | 01111101100000110110001001010010
proxmark3> lf t55 dump 65857569 o
Reading Page 0:
blk | hex data | binary
----+----------+---------------------------------
Safety Check Overriden - proceeding despite risk
  0 | 00148050 | 00000000000101001000000001010000
Safety Check Overriden - proceeding despite risk
  1 | FF8E4002 | 11111111100011100100000000000010
Safety Check Overriden - proceeding despite risk
  2 | 7D836252 | 01111101100000110110001001010010
Safety Check Overriden - proceeding despite risk
  3 | FFFFFFFF | 11111111111111111111111111111111
Safety Check Overriden - proceeding despite risk
  4 | FFFFFFFF | 11111111111111111111111111111111
Safety Check Overriden - proceeding despite risk
  5 | FFFFFFFF | 11111111111111111111111111111111
Safety Check Overriden - proceeding despite risk
  6 | FFFFFFFF | 11111111111111111111111111111111
Safety Check Overriden - proceeding despite risk
  7 | B2C2BAB4 | 10110010110000101011101010110100
Reading Page 1:
blk | hex data | binary
----+----------+---------------------------------
Safety Check Overriden - proceeding despite risk
  0 | 00148050 | 00000000000101001000000001010000
Safety Check Overriden - proceeding despite risk
  1 | FF8E4002 | 11111111100011100100000000000010
Safety Check Overriden - proceeding despite risk
  2 | 7D836252 | 01111101100000110110001001010010
Safety Check Overriden - proceeding despite risk
  3 | FFFFFFFF | 11111111111111111111111111111111
```

assasinfil, 2019-07-08:
Could you give a link to the documentation about 1 sheet, I figured out the first sheet, there is a detailed analysis on the forum: [http://www.proxmark.org/forum/viewtopic.php?id=1767](t5577)

merlokk, 2019-07-08:
https://store.dangerousthings.com/wp-content/uploads/doc_xEM_Atmel-9187-RFID-ATA5577C_Datasheet.pdf

mwalker33, 2019-07-08:
or from the vendors site (who make the chip) as I already posted above.
http://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-9187-RFID-ATA5577C_Datasheet.pdf

assasinfil, 2019-07-08:
And then you can add an article on the forum about 1 page?

assasinfil, 2019-07-08:
That is, I should always duplicate 3 blocks to 1 page out of 0?

mwalker33, 2019-07-08:
Before we start drawing conclusions about what should or should not be set, we need to confirm if the card works.  In my opinion that changes I suggested should not make a difference to a standard read of an em4100 tag.  I was trying to remove them as they MAY be an issue under some conditions which are unknown.  
The config in block 0 page 0 is in block 0 page 1 and you cant change that (they will always match).
Block 1 and 2 of page 1 default hold the chip trackability data from the vendor.  Normally used for quality control. i.e. a bad batch could be IDed by that data.  But once in production you can store data there as needed.  And I could see it was send out data from that area in the dump without a password (when a password was needed).
The block 3 page 1 config is more for advanced needs, and should only be set when you know you need it.   For block 3 page 1 config to be active the first 4 bits should be set to 0110 or 1001 (read the datasheet for why).  Any other value should mean any data there is not used.  I was more interested in WHY there was invalid data there.

I think you are confusing what the EM4100 tag is and what the T5577 is. They are two very different cards.  The EM4100 tag is a read only tag with a fixed ID that a system can use.
The T5577 is a card that can emulate other codes (e.g. the EM4100).  So to setup you need to know what the original card needs.  The proxmark3 coders have done this work for you for a stock EM4100 (in this example), but if there is something weird it may need tweaking.

assasinfil, 2019-07-08:
I will write when I check the key, it is very strange that the usual em4100 cloning did not work with my intercom

merlokk, 2019-07-08:
it not strange. many intercom checks if you place a clone. and way they check very different.

assasinfil, 2019-07-09:
Everything works fine, apparently there was a problem in the configuration of the first page.  Why lf em 410x write does not clear the first page when cloning?

merlokk, 2019-07-09:
Because not all t5577 have it

mwalker33, 2019-07-09:
Good to hear you got it working.
Thanks merlokk, I would have said not all T55x7 have it, e.g. the T5567 looks to be just a page 0 chip.

Can I suggest that if its working, then this be closed and move questions about what it can and cant do to the forum.

assasinfil, 2019-07-09:
I think I need to check the lf t55 wipe function (I suspect that it doesn’t clear 1 page) and the em 410xwrite cloning function with the t5577 mode.

closed 2019-07-09 by assasinfil