Proxmark/proxmark3 #839: Clone MF PLus Tag with 18 Sectors

pittihomme, 2019-06-26:
Hi all , i try to copy my tag with my proxmark3. It seems that the new generation of tags added 2 more blocs .
UID : xx xx xx xx      
ATQA : 00 04          
 SAK : 08 [2]          
TYPE : NXP MIFARE CLASSIC 1k | Plus 2k SL1          
proprietary non iso14443-4 card found, RATS not supported          
No chinese magic backdoor command detected          
Prng detection: HARDENED (hardnested) 

When i do a chk i find all the keys (that i sniffed many time ago) but only from bloc 0 to 15 

|---|----------------|---|----------------|---|          
|sec|key A           |res|key B           |res|          
|---|----------------|---|----------------|---|          
|000|  484558414354  | 1 |  a22ae129c013  | 1 |          
|001|  484558414354  | 1 |  49fae4e3849f  | 1 |          
|002|  484558414354  | 1 |  38fcf33072e0  | 1 |          
|003|  484558414354  | 1 |  8ad5517b4b18  | 1 |          
|004|  484558414354  | 1 |  509359f131b1  | 1 |          
|005|  484558414354  | 1 |  6c78928e1317  | 1 |          
|006|  484558414354  | 1 |  aa0720018738  | 1 |          
|007|  484558414354  | 1 |  a6cac2886412  | 1 |          
|008|  484558414354  | 1 |  62d0c424ed8e  | 1 |          
|009|  484558414354  | 1 |  e64a986a5d94  | 1 |          
|010|  484558414354  | 1 |  8fa1d601d0a2  | 1 |          
|011|  484558414354  | 1 |  89347350bd36  | 1 |          
|012|  484558414354  | 1 |  66d2b7dc39ef  | 1 |          
|013|  484558414354  | 1 |  6bc1e1ae547d  | 1 |          
|014|  484558414354  | 1 |  22729a9bd40f  | 1 |          
|015|  484558414354  | 1 |  484558414354  | 1 |          
|---|----------------|---|----------------|---| 

After using the harnested attack i found to 2 last A and B sectors keys . ( i renamed them in 16 and 17 )

|016|  5c8ff9990da2  | 1 |  d01afeeb890a  | 1 | 
|017|  75ccb59c9bed | 1 |  4b791bea7bcc  | 1 | 

I think they just added an "extra" security by adding 2 blocs ...

Anyway , when i dump / restore my tag , there is only 16 sectors instead of the 18 needed . 

My 2 questions are :

Is it possible to read / dump a tag witch has more than 16 sectors with mifare / mifare plus ?

Is there any writable tag with more than 16 sectors ont the market ?

Thanks for your answers

iceman1001, 2019-06-26:
18 sectors Classic card???   Never heard of.  Do you have more information about this card?

pittihomme, 2019-06-26:
Sure , it is a HEXACT tags , used in France in more and more cities . What do you need exactly ?

pittihomme, 2019-06-26:
proxmark3> hf mf hardnested 0 A 484558414354 68 B
--target block no: 68, target key type:B, known target key: 0x000000000000 (not set), file action: none, Slow: No, Tests: 0           
Using AVX2 SIMD core.          

          
 time    | #nonces | Activity                                                | expected to brute force          
         |         |                                                         | #states         | time           
------------------------------------------------------------------------------------------------------          
       0 |       0 | Start using 4 threads and AVX2 SIMD core                |                 |          
       0 |       0 | Brute force benchmark: 503 million (2^28,9) keys/s      | 140737488355328 |    3d          
       1 |       0 | Using 235 precalculated bitflip state tables            | 140737488355328 |    3d          
       5 |     112 | Apply bit flip properties                               |    168506916864 |  6min          
       6 |     223 | Apply bit flip properties                               |     98179530752 |  3min          
       7 |     335 | Apply bit flip properties                               |     60757483520 |  2min          
       8 |     446 | Apply bit flip properties                               |     33304799232 |   66s          
       9 |     557 | Apply bit flip properties                               |     33304799232 |   66s          
      10 |     668 | Apply bit flip properties                               |     32756439040 |   65s          
      11 |     779 | Apply bit flip properties                               |     31782246400 |   63s          
      12 |     891 | Apply bit flip properties                               |     29834909696 |   59s          
      12 |    1003 | Apply bit flip properties                               |     29741737984 |   59s          
      13 |    1113 | Apply bit flip properties                               |     29741737984 |   59s          
      14 |    1224 | Apply bit flip properties                               |     29741737984 |   59s          
      14 |    1334 | Apply bit flip properties                               |     29741737984 |   59s          
      16 |    1445 | Apply Sum property. Sum(a0) = 136                       |      2707263232 |    5s          
      16 |    1554 | Apply bit flip properties                               |      2707263232 |    5s          
      17 |    1663 | Apply bit flip properties                               |      1855177600 |    4s          
      18 |    1770 | Apply bit flip properties                               |      1855177600 |    4s          
      19 |    1770 | (Ignoring Sum(a8) properties)                           |      1855177600 |    4s          
      22 |    1770 | Starting brute force...                                 |      1855177600 |    4s          
      26 |    1770 | Brute force phase completed. Key found: 4b791bea7bcc    |               0 |    0s

pwpiwi, 2019-06-26:
> Plus 2k SL1

A 2k card would have 32 sectors. Are thereore than 18?

pittihomme, 2019-06-26:
It's what i think but when i try to attack the sector 19 and above i have an error "auth2 error"

pwpiwi, 2019-06-27:
Can you read the sectors 16 and 17 with "hf mf rdsc" ?

pittihomme, 2019-06-27:
I got an error 04

hf mf rdsc 17 A 75ccb59c9bed
--sector no:17 key type:A key:75 cc b5 9c 9b ed            
           
#db# Cmd Error: 04          
#db# Read sector 17 block  0 error          
#db# READ SECTOR FINISHED          
isOk:00          

I got the same error for 16 A or B and 17 A or B . I don't understand ... If it is not sectors , what is it ?

Same error with rdbl

hf mf rdbl 64 A 5c8ff9990da2
--block no:64, key type:A, key:5c 8f f9 99 0d a2            
#db# Cmd Error: 04          
#db# Read block error          
#db# READ BLOCK FINISHED

pittihomme, 2019-06-27:
I can find the block by using harnested attack but not with rdbl ? I'm lost ...

pittihomme, 2019-06-27:
i tried all sectors to be sure and sector 17 B works ...

proxmark3> hf mf rdsc 17 B 4b791bea7bcc
--sector no:17 key type:B key:4b 79 1b ea 7b cc            
           
#db# READ SECTOR FINISHED          
isOk:01          
data   : 48 4d 41 34 35 35 19 39 16 77 e7 00 00 00 00 00           
data   : 4a ce ac 9f 3e 4f 7c 45 9c d2 51 ff 95 0a dc 15           
data   : ee f3 7a 87 be b7 18 7c 45 7a b8 78 43 fc f1 8a           
trailer: 00 00 00 00 00 00 70 f0 f8 69 00 00 00 00 00 00           
Trailer decoded:          
Access block 68: read B          
Access block 69: read B          
Access block 70: read B          
Access block 71: read ACCESS by AB          
UserData: 69

pittihomme, 2019-06-27:
The last block of sector 17 A is readable but not the 68 , 69 and 70 

proxmark3> hf mf rdbl 71 A 75ccb59c9bed
--block no:71, key type:A, key:75 cc b5 9c 9b ed            
#db# READ BLOCK FINISHED          
isOk:01 data:00 00 00 00 00 00 70 f0 f8 69 00 00 00 00 00 00           
Trailer decoded:          
Access block 68: read B          
Access block 69: read B          
Access block 70: read B          
Access block 71: read ACCESS by AB          
UserData: 69

So sectors 16 A and B and theirs blocks are impossible to read .
Sector 17 A can't be read but only block 71 ( the last one can be found)
Sector 17 B is readable (sectors and blocks)

pwpiwi, 2019-06-27:
You should be able to read block 67 as well.

pwpiwi, 2019-06-27:
I propose that you post your (very interesting) findings in the forum in order to reach a much wider audience.

pittihomme, 2019-06-27:
Hmmm , where can i post , the forum seems quite big .

pwpiwi, 2019-06-27:
"Unknown tags" would be a good category.

pwpiwi, 2019-06-27:
Did you try block 67?

pittihomme, 2019-06-27:
i have an error on A and B

proxmark3> hf mf rdbl 67 A 5c8ff9990da2
--block no:67, key type:A, key:5c 8f f9 99 0d a2            
#db# Cmd Error: 04          
#db# Read block error          
#db# READ BLOCK FINISHED          
isOk:00          
proxmark3> hf mf rdbl 67 B d01afeeb890a 
--block no:67, key type:B, key:d0 1a fe eb 89 0a            
#db# Cmd Error: 04          
#db# Read block error          
#db# READ BLOCK FINISHED          
isOk:00

pittihomme, 2019-06-29:
Ok guys , it seems that i have nos answers in the forum . i still try to find a way to copy all the keys in a dump. Is it possible to add into the program a way to scan the sectors from 0 to 17 instead of 15 ? I'm looking for a way to read the blocs from sector 16 and 17 . Do you have any ideas ?

iceman1001, 2019-06-29:
do it manually with read / write commands.

pittihomme, 2019-06-29:
I did it but only 17B sector can be read . 16 A and B and 17 A give me an error . Is there an other way to do it ?

pwpiwi, 2019-06-29:
I have the feeling that we miss something. Which software are you using on which operating system? What is the output of "hw version"?

pittihomme, 2019-06-30:
Last version of proxmark / Kali .

proxmark3> hw version
Prox/RFID mark3 RFID instrument          
bootrom: master/v3.1.0-74-g3d057cf-suspect 2019-03-19 12:42:54
os: master/v3.1.0-75-gfdd9395-suspect 2019-03-19 13:02:11
fpga_lf.bit built for 2s30vq100 on 2015/03/06 at 07:38:04
fpga_hf.bit built for 2s30vq100 on 2019/02/15 at 20:40:32
SmartCard Slot: not available
          
uC: AT91SAM7S512 Rev A          
Embedded Processor: ARM7TDMI          
Nonvolatile Program Memory Size: 512K bytes. Used: 205329 bytes (39%). Free: 318959 bytes (61%).          
Second Nonvolatile Program Memory Size: None          
Internal SRAM Size: 64K bytes          
Architecture Identifier: AT91SAM7Sxx Series     

I can send one of my keys to one of you if you want in order to go faster ...

pittihomme, 2019-06-30:
I tryied to dump my tag but i have errors . 18 sectors like i said . 
proxmark3> hf mf chk *2 ? clef.dic

--chk keys. sectors:32, block no:  0, key type:B, eml:n, dmp=n checktimeout=471 us          
chk custom key[ 0] 62d0c424ed8e          
chk custom key[ 1] a22ae129c013          
chk custom key[ 2] 484558414354          
chk custom key[ 3] 38fcf33072e0          
chk custom key[ 4] 49fae4e3849f          
chk custom key[ 5] 8ad5517b4b18          
chk custom key[ 6] 509359f131b1          
chk custom key[ 7] 6c78928e1317          
chk custom key[ 8] aa0720018738          
chk custom key[ 9] a6cac2886412          
chk custom key[10] e64a986a5d94          
chk custom key[11] 8fa1d601d0a2          
chk custom key[12] 89347350bd36          
chk custom key[13] 66d2b7dc39ef          
chk custom key[14] 6bc1e1ae547d          
chk custom key[15] 22729a9bd40f          
chk custom key[16] 5c8ff9990da2          
chk custom key[17] 75ccb59c9bed          
chk custom key[18] d01afeeb890a          
chk custom key[19] 4b791bea7bcc          

To cancel this operation press the button on the proxmark...          
--o          
|---|----------------|---|----------------|---|          
|sec|key A           |res|key B           |res|          
|---|----------------|---|----------------|---|          
|000|  484558414354  | 1 |  a22ae129c013  | 1 |          
|001|  484558414354  | 1 |  49fae4e3849f  | 1 |          
|002|  484558414354  | 1 |  38fcf33072e0  | 1 |          
|003|  484558414354  | 1 |  8ad5517b4b18  | 1 |          
|004|  484558414354  | 1 |  509359f131b1  | 1 |          
|005|  484558414354  | 1 |  6c78928e1317  | 1 |          
|006|  484558414354  | 1 |  aa0720018738  | 1 |          
|007|  484558414354  | 1 |  a6cac2886412  | 1 |          
|008|  484558414354  | 1 |  62d0c424ed8e  | 1 |          
|009|  484558414354  | 1 |  e64a986a5d94  | 1 |          
|010|  484558414354  | 1 |  8fa1d601d0a2  | 1 |          
|011|  484558414354  | 1 |  89347350bd36  | 1 |          
|012|  484558414354  | 1 |  66d2b7dc39ef  | 1 |          
|013|  484558414354  | 1 |  6bc1e1ae547d  | 1 |          
|014|  484558414354  | 1 |  22729a9bd40f  | 1 |          
|015|  484558414354  | 1 |  484558414354  | 1 |          
|016|  5c8ff9990da2  | 1 |  d01afeeb890a  | 1 |          
|017|  75ccb59c9bed  | 1 |  4b791bea7bcc  | 1 |          
|018|  ffffffffffff  | 0 |  ffffffffffff  | 0 |          
|019|  ffffffffffff  | 0 |  ffffffffffff  | 0 |          
|020|  ffffffffffff  | 0 |  ffffffffffff  | 0 |          
|021|  ffffffffffff  | 0 |  ffffffffffff  | 0 |          
|022|  ffffffffffff  | 0 |  ffffffffffff  | 0 |          
|023|  ffffffffffff  | 0 |  ffffffffffff  | 0 |          
|024|  ffffffffffff  | 0 |  ffffffffffff  | 0 |          
|025|  ffffffffffff  | 0 |  ffffffffffff  | 0 |          
|026|  ffffffffffff  | 0 |  ffffffffffff  | 0 |          
|027|  ffffffffffff  | 0 |  ffffffffffff  | 0 |          
|028|  ffffffffffff  | 0 |  ffffffffffff  | 0 |          
|029|  ffffffffffff  | 0 |  ffffffffffff  | 0 |          
|030|  ffffffffffff  | 0 |  ffffffffffff  | 0 |          
|031|  ffffffffffff  | 0 |  ffffffffffff  | 0 |          
|---|----------------|---|----------------|---| 

When i try to dump , i got an error with the sector 16 , sector 17 seems readeable 

proxmark3> hf mf dump 2 m
|-----------------------------------------|          
|------ Reading sector access bits...-----|          
|-----------------------------------------|          
#db# READ BLOCK FINISHED          
#db# READ BLOCK FINISHED          
#db# READ BLOCK FINISHED          
#db# READ BLOCK FINISHED          
#db# READ BLOCK FINISHED          
#db# READ BLOCK FINISHED          
#db# READ BLOCK FINISHED          
#db# READ BLOCK FINISHED          
#db# READ BLOCK FINISHED          
#db# READ BLOCK FINISHED          
#db# READ BLOCK FINISHED          
#db# READ BLOCK FINISHED          
#db# READ BLOCK FINISHED          
#db# READ BLOCK FINISHED          
#db# READ BLOCK FINISHED          
#db# READ BLOCK FINISHED          
#db# Cmd Error: 04          
#db# Read block error          
#db# READ BLOCK FINISHED          
#db# Cmd Error: 04          
#db# Read block error          
#db# READ BLOCK FINISHED          
#db# Cmd Error: 04          
#db# Read block error          
#db# READ BLOCK FINISHED          
Could not get access rights for sector 16. Trying with defaults...          
#db# READ BLOCK FINISHED          
#db# Auth error          
#db# READ BLOCK FINISHED          
#db# Auth error          
#db# READ BLOCK FINISHED          
#db# Auth error          
#db# READ BLOCK FINISHED          
Could not get access rights for sector 18. Trying with defaults...          
#db# Auth error          
#db# READ BLOCK FINISHED          
#db# Auth error          
#db# READ BLOCK FINISHED          
#db# Auth error          
#db# READ BLOCK FINISHED

iceman1001, 2019-06-30:
Ok, I am curious enough now.   Do please send one tag/keyfob to me.
email me.

mwalker33, 2019-07-01:
I am wondering if the errors are due to a sector trailer with invalid permissions.
On one of my test mifare classic 1k cards I managed to set invalid permissions.  end result is I could check and find the passwords but could not read the sector or blocks.
In my case I get a Cmd Error: 00 (not 04 as here), then Read Block Error.
I know 100% what the keys are and I know 100% the permission bytes were wrong (my fault).

Just a thought.

pittihomme, 2019-07-01:
How can i do that ? I think there is maybe a key inside the reader who can be the only able to read the blocks . I just don't know how to proceed

mwalker33, 2019-07-01:
I am just thinking.   To do what I am saying you would need to write to the sector trailer
Just for a test, I took a magic card.  Set all the sector trailers to 
ffffffffffff7F078800ffffffffffff
I could then read and dump all sectors as per normal.
I then wrote  invalid permission bytes ffffff
ffffffffffffffffff00ffffffffffff
then tried to read 
[usb] pm3 --> hf mf rdsc 15 a ffffffffffff
--sector no:15 key type:A key:FF FF FF FF FF FF

#db# Cmd Error: 04
#db# Read sector 15 block  0 error
isOk:00
then (being a magic card) wrote a valid sector trailer back and attempted to read again.
[usb] pm3 --> hf mf rdsc 15 a ffffffffffff
--sector no:15 key type:A key:FF FF FF FF FF FF

isOk:01
data   : 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
data   : 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
data   : 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
trailer: 00 00 00 00 00 00 7F 07 88 00 00 00 00 00 00 00
Trailer decoded:
Access block 60: rdAB wrAB incAB dectrAB
Access block 61: rdAB wrAB incAB dectrAB
Access block 62: rdAB wrAB incAB dectrAB
Access block 63: wrAbyB rdCbyAB wrCbyB wrBbyB
UserData: 00

I was thinking back to some copy protection where they used to put errors on the discs.  so if you copied  it the error would not be there and the software would know it was a copy as "no errors".
So what if the vendor wont some invalid permissions and knows its a copy if they are not their.

In order to test you would need to create the same thing.  BUT if done to a valid card you wont be able to recover if it does not work.  Do they make 2 K UID cards ?

pittihomme, 2019-07-01:
I think i see what you mean . Maybe the errors are here for a good reason ... Anyway i don't have any 2K card to try it . I try to modify my dumkeys.bin or dumpdata.bin but i don't know how to open it . In order to dump the errors i need to read them first . That's interesting . I really think that the errors can be read by the reader with a specific uid or key inside the program . I did a try  1 month ago to sniff the communication and i had nothing . I will try again in a few weeks . What if the blocs are not hexadecimal ? Why not to put a key with 12*x or any letter not hexadecimal ? The bloc will show an error when you read it ...

mwalker33, 2019-07-01:
I don't know if its valid, just an idea.  I want to stress that if you do write invalid permissions you WONT recover it, that sector will be bricked.  So if not correct you will need a new card.

That said,, and assuming a 2k card (32 sectors)
Read what you can with the keys found and write to make a 1:1.
then for any sectors that you know the keys (as per chk ) then build the sector trailers based on that.
e.g. <12 hex digit key A>FFFFFF00<12 hex digit key B>

Again, I will stress this WILL brick that sector on a normal card (a UID magic card that supports the c commands can be recovered so would be a good start for testing).

pittihomme, 2019-07-01:
Just have to find a 2k magic card ...

iceman1001, 2019-07-06:
Thanks @pittihomme   I got the tag.

Its VIGIK tag for sure and there is two sectors 16,17 answering.

Looking at readsector code,  it takes and tries to actually read that sector.  No "wrapping" protection etc in code.   As verified with tracelist.

Also the keyA doesn't work,   as verified with the Access rights.

`[usb] pm3 --> hf mf rdsc 17 b 4b791bea7bcc
--sector no:17 key type:B key:4B 79 1B EA 7B CC            
          
isOk:01          
data   : 48 48 57 36 33 38 11 44 15 AD A8 00 00 00 00 00           
data   : 70 C7 2B EF C4 A0 6E 6B 16 DA B9 69 5E 6E EC 09           
data   : 33 90 64 91 E4 C9 ED E8 E3 F6 86 58 CA B0 41 5A           
trailer: 00 00 00 00 00 00 70 F0 F8 69 00 00 00 00 00 00           
Trailer decoded:          
Access block 68: rdB          
Access block 69: rdB          
Access block 70: rdB          
Access block 71: rdCbyAB          
UserData: 69          
[usb] pm3 --> hf mf list
[+] Recorded Activity (TraceLen = 314 bytes)          
[=]           
Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer          
iso14443a - All times are in carrier periods (1/13.56Mhz)          
          
      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation          
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------          
          0 |        992 | Rdr |52                                                                       |     | WUPA          
       2244 |       4612 | Tag |04  00                                                                   |     |           
       7040 |       9504 | Rdr |93  20                                                                   |     | ANTICOLL          
      10692 |      16580 | Tag |d3  a8  bb  62  a2                                                       |     |           
      19456 |      29920 | Rdr |93  70  d3  a8  bb  62  a2  9b  0f                                       |  ok | SELECT_UID          
      31172 |      34692 | Tag |08  b6  dd                                                               |     |           
      37120 |      41824 | Rdr |61  44  0d  66                                                           |  ok | AUTH-B(68)          
      46660 |      51396 | Tag |f4  98  4e  d0                                                           |     | AUTH: nt           
      60672 |      69984 | Rdr |bd  06  0c  ed  0d  62  90  87                                           |     | AUTH: nr ar (enc)          
      71236 |      75972 | Tag |c5! 6b  7f  ae                                                           |     | AUTH: at (enc)          
      81920 |      86688 | Rdr |a1  60  b5  78                                                           |     |           
            |            |  *  |                                              key 4b791bea7bcc prng HARD  |     |          
            |            |  *  |30  44  22  AC                                                           |  ok | READBLOCK(68)          
      87876 |     108740 | Tag |09! 17! 8c! 6d! 03  7a  09! 9f  65  ae! 37! f1! 58  03  b8  18  30! 84!  |     |           
            |            |  *  |48  48  57  36  33  38  11  44  15  AD  A8  00  00  00  00  00  D2  45   |  ok |           
     121600 |     126368 | Rdr |ac  ee  cc  2a                                                           |     |           
            |            |  *  |30  45  AB  BD                                                           |  ok | READBLOCK(69)          
     127556 |     148356 | Tag |8b  2a  b7! 42  2d  1d  ee! b5! bd! be  b6! a0! 63! b0  3c  78  2d  88   |     |           
            |            |  *  |70  C7  2B  EF  C4  A0  6E  6B  16  DA  B9  69  5E  6E  EC  09  A3  3A   |  ok |           
     161280 |     166048 | Rdr |14  28  8d  74                                                           |     |           
            |            |  *  |30  46  30  8F                                                           |  ok | READBLOCK(70)          
     167236 |     188036 | Tag |19  3d  85! 64  b0! b7  71  34! 9d  cc! 5f! 68  ba  cf! ed  d6! db! f5   |     |           
            |            |  *  |33  90  64  91  E4  C9  ED  E8  E3  F6  86  58  CA  B0  41  5A  F7  E9   |  ok |           
     200960 |     205728 | Rdr |1b  d4  33  b7                                                           |     |           
            |            |  *  |30  47  B9  9E                                                           |  ok | READBLOCK(71)          
     206916 |     227780 | Tag |96! 4d  fa  16! 6a! 71! 61  8c  b5  a1! 5e! a2! 7e  da  c9  04! 35! e3   |     |           
            |            |  *  |00  00  00  00  00  00  70  F0  F8  69  00  00  00  00  00  00  29  C7   |  ok |           
     240512 |     245216 | Rdr |9e  65  82  82                                                           |     |           
            |            |  *  |50  00  57  CD                                                           |  ok | HALT          
`

So it must be a Mifare Plus 2K tag,   someone told me this week that there is a mixed mode in Mifare Plus,   where it can have S1/S3  where some blocks answers to crypto1 and rest AES.

Using the hf mfp info commands gets some more details.

`[usb] pm3 --> hf mfp inf
 UID : D3 A8 BB 62           
ATQA : 00 04          
 SAK : 08 [2]          
TYPE : NXP MIFARE CLASSIC 1k | Plus 2k SL1 | 1k Ev1          
[=] proprietary non iso14443-4 card found, RATS not supported          
[=] Answers to magic commands: NO           
[+] Prng detection: HARD           
----------------------------------------------          
Mifare Plus info:          
[=] ATQA: Mifare Plus 2k 4bUID          
[=] SAK: Mifare Plus 2k 7bUID          
[=] Mifare Plus SL mode: SL1          
`

pwpiwi, 2019-07-11:
Still weird.
> proprietary non iso14443-4 card found, RATS not supported

A Mifare Plus should be iso14443-4 compliant.
> Could not get access rights for sector 16

Access rights can always be read with KeyA. There are no valid access conditions which would prevent that. I think this is an indication for @mwalker33's assumption, that sector 16 has (probably deliberately) invalid access condition, which makes it unreadable.
> there is a mixed mode in Mifare Plus, where it can have S1/S3 where some blocks answers to crypto1 and rest AES.

There is an optional AES authentication possibility in S1. But this is for the card, not per sector. In S3, classic Mifare authentication doesn't work at all.

pwpiwi, 2019-07-16:
I didn't look close enough. It is a Mifare Classic 1k EV1. Sector 17 contains the originality signature.

pittihomme, 2019-07-16:
I still have one tag ( dfferent of yours ) , maybe we can try to see if it is the same sectors 16 and 17 ?

pwpiwi, 2019-07-16:
The data in sector 17 will be different because the originality signature depends on the UID. But key B should be the same.

pittihomme, 2019-07-16:
Is there a way to read sector 16 ? I'm not enought skilled for the next steps ... 
If it can be read , how to restore it and on witch tag ? a 1k or 2k ?

pittihomme, 2019-07-18:
I tried already to use only 16 sectors and the reader didn't read the tag. So the sectors 16 and 17 are here for something.

pwpiwi, 2019-07-19:
Did you try with a magic 4k? Copy sector 17 and brick sector 16.

pittihomme, 2019-07-19:
Do you know where i Can buy thiz tag ? Il found cards at 30+ € , seems expensive for what i need to do

pwpiwi, 2019-07-19:
No. Time to close here and switch to the forum. I think we agree that there is no issue with the code.

closed 2019-08-12 by pwpiwi

sn0wst0rm, 2023-08-28:
Hi! I was doing some research on my university (NTNU) badge, I wanted to clone but then I encountered some strange behaviour I didn't ever see in a classic Mifare tag.

Searching through the web I came across this issue that is talking exactly about the same tag (same behaviour, two additional sector, 16 and 17, same keys, only sector 17 readable with key B). moreover, the same badge hosts another chip, a LF EM410x ( RF/64 ), never saw such an arrangement in a normal badge.

Reading through this issue I found out that the conversation was shifted to the forum but unfortunately I am struggling finding the topic also with the search.

Was it deleted? If someone could point me tho re right resources it would be awesome!

pittihomme, 2023-08-28:
Hi Riccardo, can you send me a mail directly ? mw [at] bfnice.com

CarlosJaramillo2, 2024-01-05:
> Hi all , i try to copy my tag with my proxmark3. It seems that the new generation of tags added 2 more sectors . UID : xx xx xx xx ATQA : 00 04 SAK : 08 [2] TYPE : NXP MIFARE CLASSIC 1k | Plus 2k SL1 proprietary non iso14443-4 card found, RATS not supported No chinese magic backdoor command detected Prng detection: HARDENED (hardnested)
> 
> When i do a chk i find all the keys (that i sniffed many time ago) but only from sector 0 to 15
> 
> |---|----------------|---|----------------|---|
> 
> sec	key A	res	key B	res
> 000	[484558414354](tel:484558414354)	1	a22ae129c013	1
> 001	[484558414354](tel:484558414354)	1	49fae4e3849f	1
> 002	[484558414354](tel:484558414354)	1	38fcf33072e0	1
> 003	[484558414354](tel:484558414354)	1	8ad5517b4b18	1
> 004	[484558414354](tel:484558414354)	1	509359f131b1	1
> 005	[484558414354](tel:484558414354)	1	6c78928e1317	1
> 006	[484558414354](tel:484558414354)	1	aa0720018738	1
> 007	[484558414354](tel:484558414354)	1	a6cac2886412	1
> 008	[484558414354](tel:484558414354)	1	62d0c424ed8e	1
> 009	[484558414354](tel:484558414354)	1	e64a986a5d94	1
> 010	[484558414354](tel:484558414354)	1	8fa1d601d0a2	1
> 011	[484558414354](tel:484558414354)	1	89347350bd36	1
> 012	[484558414354](tel:484558414354)	1	66d2b7dc39ef	1
> 013	[484558414354](tel:484558414354)	1	6bc1e1ae547d	1
> 014	[484558414354](tel:484558414354)	1	22729a9bd40f	1
> 015	[484558414354](tel:484558414354)	1	[484558414354](tel:484558414354)	1
> ---	----------------	---	----------------	---
> After using the harnested attack i found to 2 last A and B sectors keys . ( i renamed them in 16 and 17 )
> 
> |016| 5c8ff9990da2 | 1 | d01afeeb890a | 1 | |017| 75ccb59c9bed | 1 | 4b791bea7bcc | 1 |
> 
> I think they just added an "extra" security by adding 2 sectors / 8 blocs ...
> 
> Anyway , when i dump / restore my tag , there is only 16 sectors instead of the 18 needed .
> 
> My 2 questions are :
> 
> Is it possible to read / dump a tag witch has more than 16 sectors with mifare / mifare plus ?
> 
> Is there any writable tag with more than 16 sectors ont the market ?
> 
> Thanks for your answers

Hello! Did you make it works? Wich type of tag did you use to make the new copy? 
I have same issue and I’m kind of new in this