Proxmark/proxmark3 #840: How to script a simple proxmark3 command?

N2Network, 2019-07-03:
I'm trying to connect a proxmark3 to my raspberry pi 3 b+ so that when it's powered on it will simply read any rfid that it comes in contact with. 

I can do this manually by executing /home/pi/proxmark3/client/proxmark3 /dev/ttyACM0 -c "lf hid read" However, this quickly opens then closes. I'm curious if someone can point me in the right direction to configure this. If possible it would be nice to trunk out to a text file the results of the read as well. Unless they are automatically already stored in a log file?

pwpiwi, 2019-07-11:
You can run a single command (like you did) or you can run a sequence of commands which you would prepare in a text file. However, there are no loop commands (which you would need to create an endless loop). A possibility I can think of: write a lua script which does what you want. Then execute this script on startup.

N2Network, 2019-07-11:
I have no Lua programming experience. Any examples how this type of script would work?

pwpiwi, 2019-07-11:
> I have no Lua programming experience.

Me too.
> Any examples how this type of script would work?

You may have a look at the scripts in client/scripts. You would start a script (located in this directory) on startup with '/home/pi/proxmark3/client/proxmark3 /dev/ttyACM0 -l example_lua_script_name'.

mwalker33, 2019-07-15:
I am trying to teach myself the lua integration now.  I think the lua "calls" are at the USB comms level.
i.e. If you were to look at the "lf hid read" command, you will find it has a call "id" and some arguments. So you need to build that in the lua script.  The call ID name/value are found in the usb_cmd.lua file in the client/lualibs folder.
What i have not got to yet is how to read and demod the response.  So any tips would be good.

merlokk, 2019-07-15:
logic here
https://github.com/Proxmark/proxmark3/blob/master/client/scripting.c
lua can send usb commands and can execute command line commands (and something else)

mwalker33, 2019-07-16:
This is very much a play, but should get you started.
I did not give things too much thought, just found something that seemed to work.
```
local cmds = require('commands')
local getopt = require('getopt')
local bin = require('bin')
local utils = require('utils')
local format=string.format
local floor=math.floor

function sleep (a) 
    local sec = tonumber(os.clock() + a); 
    while (os.clock() < sec) do 
    end 
end

local function main(args)
	print( string.rep('--',30) )
	print( string.rep('--',30) )

	-- core.console will go async when waiting for the proxmark to return
	-- so need to wait for a response (sleep)
	core.console('lf hid read')
	sleep(30) -- run for 30 seconds
	
	-- Need some what to tell the proxmark to stop waiting for HID tags
	core.console('lf search')

	print( string.rep('--',30) )

end
main(args)
```

I had the script it a test folder (windows), so a relative link to the script.

command line : 
D:\pm3\gitfiles\proxmark3\client>proxmark3.exe com10 -c "script run ../../../../read_hid"

output (card id removed)
```
Execute command from commandline: script run ../../../../read_hid
Prox/RFID mark3 RFID instrument
bootrom: master/v3.1.0-88-g9ebbfd8-suspect 2019-04-23 09:39:46
os: master/v3.1.0-118-g59f0b03-dirty-suspect 2019-07-12 00:52:55
fpga_lf.bit built for 2s30vq100 on 2015/03/06 at 07:38:04
fpga_hf.bit built for 2s30vq100 on 2019/03/20 at 08:08:07
SmartCard Slot: not available

uC: AT91SAM7S512 Rev A
Embedded Processor: ARM7TDMI
Nonvolatile Program Memory Size: 512K bytes. Used: 206926 bytes (39%). Free: 317362 bytes (61%).
Second Nonvolatile Program Memory Size: None
Internal SRAM Size: 64K bytes
Architecture Identifier: AT91SAM7Sxx Series
Nonvolatile Program Memory Type: Embedded Flash Memory
proxmark3> script run ../../../../read_hid
--- Executing: ../../../../read_hid.lua, args ''
------------------------------------------------------------
------------------------------------------------------------
#db# TAG ID: ********** (xxxxx)
#db# TAG ID: ********** (xxxxx)
#db# TAG ID: ********** (xxxxx)
#db# TAG ID: ********** (xxxxx)
#db# TAG ID: ********** (xxxxx)
#db# TAG ID: ********** (xxxxx)
#db# Stopped
NOTE: some demods output possible binary
  if it finds something that looks like a tag
False Positives ARE possible

Checking for known tags:

No Data Found! - maybe not an LF tag?

------------------------------------------------------------

-----Finished
```

Since this just runs the command as you would inside the client, the output goes to the screen.
So a redirect might capture it for you.

If you are going to leave it running and reading, remove the sleep and create an infinite loop. (Not sure it the unit will get hot or not)

Hope its helpful.

pwpiwi, 2019-08-12:
@N2Network : can we close this?

closed 2019-09-21 by pwpiwi

abetusk, 2020-03-28:
I hope folks don't mind me performing a little necromancy on this thread.

For others (like me) that wanted to use a LUA script to continuously monitor a low frequency read, the following slightly modified script from @mwalker33 works without the active poll, issuing a system `sleep` call to help alleviate resources:

```
local cmds = require('commands')
local getopt = require('getopt')
local bin = require('bin')
local utils = require('utils')
local format=string.format
local floor=math.floor
local os = require("os")

local function main(args)
  print( string.rep('--',30) )
  print( string.rep('--',30) )

  -- core.console will go async when waiting for the proxmark to return
  -- so need to wait for a response (sleep)
  core.console('lf hid read')
  os.execute("sleep 30")

  print( string.rep('--',30) )

end
main(args)
```

The `sleep` command above can be replaced with `sleep infinity` to allow for indefinite sleep.

The above should be put into the `scripts` directory from where the `proxmark3` program is run.

The following command will buffer lines and `tee` it out to a log file:

```
stdbuf -eL -oL ./proxmark3 /dev/ttyACM0 -l lf-hid-read.lua | tee lf-hid-read.log
```