Proxmark/proxmark3 #841: Reading mifare 4K card when blocks are unused

hornyd, 2019-07-05:
I am reading a mifare 4k card where blocks from 1 to 21 are unused, the dump command fails:

|-----------------------------------------|
|----- Dumping all blocks to file... -----|
|-----------------------------------------|
#db# READ BLOCK FINISHED
Successfully read block  0 of sector  0.
#db# READ BLOCK FINISHED
Successfully read block  1 of sector  0.
#db# READ BLOCK FINISHED
Successfully read block  2 of sector  0.
#db# READ BLOCK FINISHED
Successfully read block  3 of sector  0.
Access rights do not allow reading of sector  1 block   0
Command execute timeout when trying to read block  0 of sector  1.

hornyd, 2019-07-07:
I tried to fix the library in the meantime, but I am still unable to get the dumb file being generated. I can only make the dump file manually

pwpiwi, 2019-07-11:
Not sure what you mean with "unused" blocks. Are you sure that you have both keys for sectors 1 to 21?

hornyd, 2019-07-11:
I mean that the blocks from 1 to 21 are just set to 0, no keys are used. we just used sectors from 22 to 39 with keys. (FYI I used iceman version and I can get the binary image with skipping blocks 1 to 21).

mwalker33, 2019-07-11:
@hornyd, are you saying that when using the iceman version, if it "errors" it will skip over and fill the dump with 00's for the failed block.  If so, what does it put in for block 7 (the key/permission block of sector 1)?

hornyd, 2019-07-11:
Indeed, the iceman version if it is "errors" it will just skip over and skip the whole block and try the next block. The binary will be filled only with the valid blocks (in my case from 22 to 39).

pwpiwi, 2019-07-11:
Huh? This is an Ultralight dump. Not a Mifare 4K dump. ?????????????

pwpiwi, 2019-07-11:
> I mean that the blocks from 1 to 21 are just set to 0, no keys are used.

There are *always* keys used. You cannot access a Mifare card block without authentication.

pwpiwi, 2019-07-11:
> Indeed, the iceman version if it is "errors" it will just skip over and skip the whole block and try the next block. The binary will be filled only with the valid blocks (in my case from 22 to 39).

... which would not be a good idea. Without an indication which blocks had been skipped the file would be useless.

hornyd, 2019-07-11:
[hf-mf-560F59E1-data.zip](https://github.com/Proxmark/proxmark3/files/3383599/hf-mf-560F59E1-data.zip)
@pwpiwi excuse me, i put the wrong file. The keys for the 1 to 21 blocks are just the regular Key B. But in principal if we have 0 or key it will just be ignored.

pwpiwi, 2019-07-12:
The dump contains sector trailers with all 0. This includes the access conditions which are invalid. This results in a bricked sector when you restore it to a blank card.

In your first post we see
> Access rights do not allow reading of sector 1 block 0

which is probably the issue for all data blocks in sectors 1 to 21? What does it say when it tries reading the sector trailers for these sectors (e.g. block 3 of sector 1)? Had there been any issue reading the access conditions? Maybe it would help if you post the complete command and results. Please add as appendix or mark it as code (add \`\`\` in a separate line before and after).

hornyd, 2019-07-15:
which is probably the issue for all data blocks in sectors **2** to 21? ==> It is as per the scheme, block 1 should be readble with the card issuer key

What does it say when it tries reading the sector trailers for these sectors (e.g. block 3 of sector 1)? Had there been any issue reading the access conditions? 
See below the block 3
`proxmark3> hf mf rdbl 3 B XXXXXXXXXXXX
--block no:3, key type:B, key: XXXXXXXXXXXX
#db# READ BLOCK FINISHED
isOk:01 data:XX XX XX XX XX XX 78 77 88 XX XX XX XX XX XX
Trailer decoded:
Access block 0: read AB; writeB
Access block 1: read AB; writeB
Access block 2: read AB; writeB
Access block 3: write A by B; read ACCESS by AB; write ACCESS by B; write B by B
UserData: 00`

However, If i use block 4 with the master key:
`proxmark3> hf mf rdbl 4 B XXXXXXXXXXX
--block no:4, key type:B, key: XXXXXXXXXXX
#db# Authentication failed. Card timeout.
#db# Auth error
#db# READ BLOCK FINISHED
isOk:00`

Which is the attempted behavior.
In regards to restoring to a card, I dont know the impact of this. I am only simulating the card for now, did not need to restore to a card

mwalker33, 2019-07-15:
In the block 4 rdbl where did the key come from? i.e. did you use a nested attack to find it ?
e.g.
```
 hf mf hardnested 0 A FFFFFFFFFFFF 4 A
```
where 0 is the block with the known A Key of FFFFFFFFFFFF and 4 is the block where we are trying to find the A key for, so swap FFFFFFFFFFFF with the valid known A key for block 3

hornyd, 2019-07-15:
I know the keys, I have the card issuer keys and the A and B keys. I can calculate the keys for each block. I created a local file with all the keys so when I check the keys it successfully authentication the blocks

`hf mf chk *4 ? d ./client/card_keys.dic`

pwpiwi, 2019-07-15:
You get an auth error when trying to read block 4. I.e. you do **not** have the correct key. What do you mean with "Master Key"? There are only A and B keys for a Mifare Classic card.

I had asked about the result when dump is trying the sector trailer of sector 1 (note: this is block 3 of sector 1, or block 7 in total).

Again: please post the commands you typed and the respective responses. The partial outputs and your comments are quite confusing.

pwpiwi, 2019-08-12:
Closed due to inactivity.

closed 2019-08-12 by pwpiwi