Proxmark/proxmark3 #843: hf mf chk - t Doesnt save to emulator memory

0x646e78, 2019-07-07:
When I run the following I don't see the key in memory:

```
hf mf chk 0 A t
<snip>
chk default key[16] 533cb6c723f6
chk default key[17] 8fd0a4f256e9

Found valid key:[0:A]a0a1a2a3a4a5
Found keys have been transferred to the emulator memory
```

```
hf mf ekeyprn
|---|----------------|----------------|
|sec|key A           |key B           |
|---|----------------|----------------|
|000|  ffffffffffff  |  ffffffffffff  |
|001|  ffffffffffff  |  ffffffffffff  |
|002|  ffffffffffff  |  ffffffffffff  |
|003|  ffffffffffff  |  ffffffffffff  |
|004|  ffffffffffff  |  ffffffffffff  |
|005|  ffffffffffff  |  ffffffffffff  |
|006|  ffffffffffff  |  ffffffffffff  |
|007|  ffffffffffff  |  ffffffffffff  |
|008|  ffffffffffff  |  ffffffffffff  |
|009|  ffffffffffff  |  ffffffffffff  |
|010|  ffffffffffff  |  ffffffffffff  |
|011|  ffffffffffff  |  ffffffffffff  |
|012|  ffffffffffff  |  ffffffffffff  |
|013|  ffffffffffff  |  ffffffffffff  |
|014|  ffffffffffff  |  ffffffffffff  |
|015|  ffffffffffff  |  ffffffffffff  |
|---|----------------|----------------|
```

if I run for all sectors I get some, but not all in memory:

    hf mf chk * ? t

```
proxmark3> hf mf ekeyprn
|---|----------------|----------------|
|sec|key A           |key B           |
|---|----------------|----------------|
|000|  a0a1a2a3a4a5  |  b0b1b2b3b4b5  |
|001|  ffffffffffff  |  ffffffffffff  |
|002|  ffffffffffff  |  ffffffffffff  |
|003|  ffffffffffff  |  ffffffffffff  |
|004|  ffffffffffff  |  ffffffffffff  |
|005|  ffffffffffff  |  ffffffffffff  |
|006|  ffffffffffff  |  ffffffffffff  |
|007|  ffffffffffff  |  ffffffffffff  |
|008|  ffffffffffff  |  ffffffffffff  |
|009|  ffffffffffff  |  ffffffffffff  |
|010|  ffffffffffff  |  ffffffffffff  |
|011|  ffffffffffff  |  ffffffffffff  |
|012|  ffffffffffff  |  ffffffffffff  |
|013|  ffffffffffff  |  ffffffffffff  |
|014|  a0a1a2a3a4a5  |  b0b1b2b3b4b5  |
|015|  ffffffffffff  |  ffffffffffff  |
|---|----------------|----------------|
```

I can verify that Sector 0, key A was in the dict, and is not the value ffffffffffff as shown in the key memory print.

```
hf mf chk 1 A t
--chk keys. sectors: 0, block no:  1, key type:A, eml:y, dmp=n checktimeout=471 us
No key specified, trying default keys
chk default key[ 0] ffffffffffff
chk default key[ 1] 000000000000
chk default key[ 2] a0a1a2a3a4a5
chk default key[ 3] b0b1b2b3b4b5
chk default key[ 4] aabbccddeeff
chk default key[ 5] 1a2b3c4d5e6f
chk default key[ 6] 123456789abc
chk default key[ 7] 010203040506
chk default key[ 8] 123456abcdef
chk default key[ 9] abcdef123456
chk default key[10] 4d3a99c351dd
chk default key[11] 1a982c7e459a
chk default key[12] d3f7d3f7d3f7
chk default key[13] 714c5c886e97
chk default key[14] 587ee5f9350f
chk default key[15] a0478cc39091
chk default key[16] 533cb6c723f6
chk default key[17] 8fd0a4f256e9

Found valid key:[1:A]a0a1a2a3a4a5
Found keys have been transferred to the emulator memory
```

But again, that run of an individual key is not saved to mem:

```
hf mf ekeyprn
|---|----------------|----------------|
|sec|key A           |key B           |
|---|----------------|----------------|
|000|  a0a1a2a3a4a5  |  b0b1b2b3b4b5  |
|001|  ffffffffffff  |  ffffffffffff  |
|002|  ffffffffffff  |  ffffffffffff  |
|003|  ffffffffffff  |  ffffffffffff  |
|004|  ffffffffffff  |  ffffffffffff  |
|005|  ffffffffffff  |  ffffffffffff  |
|006|  ffffffffffff  |  ffffffffffff  |
|007|  ffffffffffff  |  ffffffffffff  |
|008|  ffffffffffff  |  ffffffffffff  |
|009|  ffffffffffff  |  ffffffffffff  |
|010|  ffffffffffff  |  ffffffffffff  |
|011|  ffffffffffff  |  ffffffffffff  |
|012|  ffffffffffff  |  ffffffffffff  |
|013|  ffffffffffff  |  ffffffffffff  |
|014|  a0a1a2a3a4a5  |  b0b1b2b3b4b5  |
|015|  ffffffffffff  |  ffffffffffff  |
|---|----------------|----------------|
``` 

I'm still learning, so perhaps I've missed somthing or messed somethng up, but it seems like an issue.

0x646e78, 2019-07-07:
I think actually I've gotten something wrong here, or something else unexpected has happened.  I've just noticed  that my `hf mf chk 1 A t` actually returns the key for sector 0?

mwalker33, 2019-07-07:
hf mf chk 0 A t
This will check for a key for block "0"
Block 0 will have the same key as 1 2 and 3 (i.e. All blocks in Sector 0)
So to check for the A key for the 2nd Sector (Sector 1)
hf mf chk 4 A t

0x646e78, 2019-07-07:
I've juest realised my mistake in confusing sectors and blocks when checking. I think still the keys are not being saved to memory properly, as seen in the first example?

mwalker33, 2019-07-08:
I have had a quick look at the code for the hf mf chk
In short, the chk <block> <key id> is for checking for a single key.  Since there is no information about the size of the card (number of sectors)  it does not know what to do to setup and transfer (since the t option was set, it tries to run the transfer with 0 sectors, so displays the message "Found keys have been transferred..." but not true.
So, as is, the t option on that command results in a number of sectors of 0, thus nothing moved to memory.

So yes some room for improvement.  

In the mean time you could use 
 hf mf chk *1 ?  t <key dic>
then it will transfer the keys, as it knows its a 1k card (1 being the example here).

I have had a play with the code and got it to work where it will check a Single Key (A or B) for a single block (sector) and transfer just that one to emulator.  Needs full testing.
e.g.
hf mf chk 4 a t d:\pm3\keys.dic

|---|----------------|---|----------------|---|
|sec|key A           |res|key B           |res|
|---|----------------|---|----------------|---|
|001|  bbbbbbbbbbbb  | 1 |  ffffffffffff  | 0 |
|---|----------------|---|----------------|---|
and  hf mf ekeyprn shows the correct key updated.
I repeated with the b key and it then added just the b key.

As I am working on a current pull request and we need to make sure that correct. I don't wont to work on this at the some time.
If anyone else wants to see my code (roughed in) let me know where you would like me to post the CmdHF14AMfChk function.

mwalker33, 2019-07-12:
Sample run with changes to : CmdHF14AMfChk

Show a "blank" eumlator key store

```
proxmark3> hf mf ekeyprn
|---|----------------|----------------|
|sec|key A           |key B           |
|---|----------------|----------------|
|000|  000000000000  |  000000000000  |
|001|  000000000000  |  000000000000  |
|002|  000000000000  |  000000000000  |
|003|  000000000000  |  000000000000  |
|004|  000000000000  |  000000000000  |
|005|  000000000000  |  000000000000  |
|006|  000000000000  |  000000000000  |
|007|  000000000000  |  000000000000  |
|008|  000000000000  |  000000000000  |
|009|  000000000000  |  000000000000  |
|010|  000000000000  |  000000000000  |
|011|  000000000000  |  000000000000  |
|012|  000000000000  |  000000000000  |
|013|  000000000000  |  000000000000  |
|014|  000000000000  |  000000000000  |
|015|  000000000000  |  000000000000  |
|---|----------------|----------------|
```

Search/Find a single Key and transfer to emulator

```
proxmark3> hf mf chk 4 a t d:\pm3\keys.dic
|---|----------------|---|----------------|---|
|sec|key A           |res|key B           |res|
|---|----------------|---|----------------|---|
|001|  bbbbbbbbbbbb  | 1 |  ffffffffffff  | 0 |
|---|----------------|---|----------------|---|
1 keys(s) found have been transferred to the emulator memory

proxmark3> hf mf ekeyprn
|---|----------------|----------------|
|sec|key A           |key B           |
|---|----------------|----------------|
|000|  000000000000  |  000000000000  |
|001|  bbbbbbbbbbbb  |  000000000000  |
|002|  000000000000  |  000000000000  |
|003|  000000000000  |  000000000000  |
|004|  000000000000  |  000000000000  |
|005|  000000000000  |  000000000000  |
|006|  000000000000  |  000000000000  |
|007|  000000000000  |  000000000000  |
|008|  000000000000  |  000000000000  |
|009|  000000000000  |  000000000000  |
|010|  000000000000  |  000000000000  |
|011|  000000000000  |  000000000000  |
|012|  000000000000  |  000000000000  |
|013|  000000000000  |  000000000000  |
|014|  000000000000  |  000000000000  |
|015|  000000000000  |  000000000000  |
|---|----------------|----------------|
```

Perform a full check, but with sector 1 keys not found
 
```
proxmark3> hf mf chk *1 ? t
|---|----------------|---|----------------|---|
|sec|key A           |res|key B           |res|
|---|----------------|---|----------------|---|
|000|  ffffffffffff  | 1 |  ffffffffffff  | 1 |
|001|  ffffffffffff  | 0 |  ffffffffffff  | 0 |
|002|  ffffffffffff  | 1 |  ffffffffffff  | 1 |
|003|  ffffffffffff  | 1 |  ffffffffffff  | 1 |
|004|  ffffffffffff  | 1 |  ffffffffffff  | 1 |
|005|  ffffffffffff  | 1 |  ffffffffffff  | 1 |
|006|  ffffffffffff  | 1 |  ffffffffffff  | 1 |
|007|  ffffffffffff  | 1 |  ffffffffffff  | 1 |
|008|  ffffffffffff  | 1 |  ffffffffffff  | 1 |
|009|  ffffffffffff  | 1 |  ffffffffffff  | 1 |
|010|  ffffffffffff  | 1 |  ffffffffffff  | 1 |
|011|  ffffffffffff  | 1 |  ffffffffffff  | 1 |
|012|  ffffffffffff  | 1 |  ffffffffffff  | 1 |
|013|  ffffffffffff  | 1 |  ffffffffffff  | 1 |
|014|  ffffffffffff  | 1 |  ffffffffffff  | 1 |
|015|  ffffffffffff  | 1 |  ffffffffffff  | 1 |
|---|----------------|---|----------------|---|
30 keys(s) found have been transferred to the emulator memory

proxmark3> hf mf ekeyprn
|---|----------------|----------------|
|sec|key A           |key B           |
|---|----------------|----------------|
|000|  ffffffffffff  |  ffffffffffff  |
|001|  bbbbbbbbbbbb  |  000000000000  |
|002|  ffffffffffff  |  ffffffffffff  |
|003|  ffffffffffff  |  ffffffffffff  |
|004|  ffffffffffff  |  ffffffffffff  |
|005|  ffffffffffff  |  ffffffffffff  |
|006|  ffffffffffff  |  ffffffffffff  |
|007|  ffffffffffff  |  ffffffffffff  |
|008|  ffffffffffff  |  ffffffffffff  |
|009|  ffffffffffff  |  ffffffffffff  |
|010|  ffffffffffff  |  ffffffffffff  |
|011|  ffffffffffff  |  ffffffffffff  |
|012|  ffffffffffff  |  ffffffffffff  |
|013|  ffffffffffff  |  ffffffffffff  |
|014|  ffffffffffff  |  ffffffffffff  |
|015|  ffffffffffff  |  ffffffffffff  |
|---|----------------|----------------|
```

Check for the sector 1 B key

```
proxmark3> hf mf chk 4 b t d:\pm3\keys.dic
|---|----------------|---|----------------|---|
|sec|key A           |res|key B           |res|
|---|----------------|---|----------------|---|
|001|  ffffffffffff  | 0 |  cccccccccccc  | 1 |
|---|----------------|---|----------------|---|
1 keys(s) found have been transferred to the emulator memory

proxmark3> hf mf ekeyprn
|---|----------------|----------------|
|sec|key A           |key B           |
|---|----------------|----------------|
|000|  ffffffffffff  |  ffffffffffff  |
|001|  bbbbbbbbbbbb  |  cccccccccccc  |
|002|  ffffffffffff  |  ffffffffffff  |
|003|  ffffffffffff  |  ffffffffffff  |
|004|  ffffffffffff  |  ffffffffffff  |
|005|  ffffffffffff  |  ffffffffffff  |
|006|  ffffffffffff  |  ffffffffffff  |
|007|  ffffffffffff  |  ffffffffffff  |
|008|  ffffffffffff  |  ffffffffffff  |
|009|  ffffffffffff  |  ffffffffffff  |
|010|  ffffffffffff  |  ffffffffffff  |
|011|  ffffffffffff  |  ffffffffffff  |
|012|  ffffffffffff  |  ffffffffffff  |
|013|  ffffffffffff  |  ffffffffffff  |
|014|  ffffffffffff  |  ffffffffffff  |
|015|  ffffffffffff  |  ffffffffffff  |
|---|----------------|----------------|
```

So seems to work as needed.

Should we write to the dump file if only a single key check is used ?
i.e. we dont know this size of the card/number of sectors so will be 1 to sector needed, with all unknown keys being FF.

 
Comments ?

pwpiwi, 2019-07-15:
> Should we write to the dump file if only a single key check is used ?

I think we shouldn't because we have no idea on the number of keys to dump.

mwalker33, 2019-07-24:
@0x646e78 
My updates have now been merged (thanks pwpiwi for your help)
Can you update and confirm its now working as expected.

pwpiwi, 2019-08-02:
@0x646e78 : any feedback?

0x646e78, 2019-08-14:
Sorry folks, been afk and out of the country, I'll have a go tomorrow evening.

0x646e78, 2019-08-21:
Hi folks,

Just reflashed and had a play. Looks really good to me now, and I learned a thing also - `chk *<1-4>`

thanks!

closed 2019-08-21 by 0x646e78