Proxmark/proxmark3 #866: Broken Proxmark3 kit after flashing

colourful-land, 2019-09-10:
**Describe the bug**
Flashing the device with the latest code (cloned today) causes the device to power-cycle.

**To Reproduce**
Steps to reproduce the behavior:
1. flash the bootrom
````
$ sudo client/flasher /dev/ttyACM0 -b bootrom/obj/bootrom.elf 
Loading ELF file 'bootrom/obj/bootrom.elf'...
Loading usable ELF segments:
0: V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94
1: V 0x00200000 P 0x00100200 (0x00000c84->0x00000c84) [R X] @0x298

Waiting for Proxmark to appear on /dev/ttyACM0 .
 Found.
#db# unknown command:: 0xc10d410d          
#db# unknown command:: 0xc10d4154          
Entering bootloader...
(Press and release the button only to abort)
Waiting for Proxmark to appear on /dev/ttyACM0 .......
 Found.

Flashing...
Writing segments for file: bootrom/obj/bootrom.elf
 0x00100000..0x001001ff [0x200 / 1 blocks]. OK
 0x00100200..0x00100e83 [0xc84 / 7 blocks]....... OK

Resetting hardware...
All done.

Have a nice day!
````
2. Flash the full image;
````
$ sudo client/flasher /dev/ttyACM0 armsrc/obj/fullimage.elf 
Loading ELF file 'armsrc/obj/fullimage.elf'...
Loading usable ELF segments:
0: V 0x00102000 P 0x00102000 (0x0002f070->0x0002f070) [R X] @0x94
1: V 0x00200000 P 0x00131070 (0x000018e0->0x000018e0) [RW ] @0x2f104
Note: Extending previous segment from 0x2f070 to 0x30950 bytes

Waiting for Proxmark to appear on /dev/ttyACM0 .
 Found.
Entering bootloader...
(Press and release the button only to abort)
Waiting for Proxmark to appear on /dev/ttyACM0 .......
 Found.

Flashing...
Writing segments for file: armsrc/obj/fullimage.elf
 0x00102000..0x0013294f [0x30950 / 389 blocks]............................Waiting for a response from the proxmark...          
You can cancel this operation by pressing the pm3 button          
^C
````

I used ^C to stop because after waiting for ~10 minutes the device is already audibly power-cycling (there seems to be a relay in it to produce the clicking sound once 5 seconds).

3. Attempt to redo the flashing by pressing the button down and holding it while reconnecting it to the USB port:
````
$ sudo client/flasher /dev/ttyACM0 armsrc/obj/fullimage.elf 
Loading ELF file 'armsrc/obj/fullimage.elf'...
Loading usable ELF segments:
0: V 0x00102000 P 0x00102000 (0x0002bb80->0x0002bb80) [R X] @0x94
1: V 0x00200000 P 0x0012db80 (0x00001960->0x00001960) [RW ] @0x2bc14
Note: Extending previous segment from 0x2bb80 to 0x2d4e0 bytes

Waiting for Proxmark to appear on /dev/ttyACM0 .
 Found.
Waiting for a response from the proxmark...          
You can cancel this operation by pressing the pm3 button          
^C
````
I quit with `^C` because the device has reset, evident by the audible clicking sound from a relay + `dmesg` shows the device hopped to `ttyACM1`

**Expected behavior**
Flash elf successfully.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu 19.04.

I could do `hw version` and `hw status` because it no longger works. But before I flash it, it works fine reading and cloning my devices (using viking)

**Additional context**
The repair procedure I took is copied from [another bug](https://github.com/RfidResearchGroup/proxmark3/issues/35#issuecomment-431681848) in which it worked. In my case, it didn't.

> **In short** (if your device gets enumerated on ACM0 adjust accordingly):
> 
>     1. unplug device
> 
>     2. press button and keep it pressed (IMPORTANT)
> 
>     3. plug in device
> 
>     4. run flash command  sudo client/flasher /dev/ttyACM0 armsrc/obj/fullimage.elf
> 
>     5. wait until flash is finished.
> 
>     6. release button.
> 
>     7. un/plug device.

colourful-land, 2019-09-10:
I attempted to use https://github.com/RfidResearchGroup/proxmark3/ fork and change repo target from RDV4 to `PLATFORM=PM3OTHER` in `Makefile.platform`, and got a different, slightly larger `bootrom/obj/bootrom.elf` and flashed it.

I got the same happy ending:

````
Resetting hardware...
All done.

Have a nice day!
````

And the resulting device keeps resetting itself just like before.

pwpiwi, 2019-09-10:
Try holding the button and using /dev/ttyACM1

colourful-land, 2019-09-10:
> Try holding the button and using /dev/ttyACM1

Wouldn't work, because `ls /dev/ttyACM1` returns "File Not Found". Also yes I did hold the button before connecting the USB port and held it for a while after.

pwpiwi, 2019-09-10:
You said

> the device hopped to ttyACM1

pwpiwi, 2019-09-21:
Any news? Did you succeed?

assasinfil, 2019-10-07:
Duplicate of https://github.com/Proxmark/proxmark3/issues/785#issuecomment-464436952#
this is the problem of the modem manager, disable it as in that comment

closed 2019-10-14 by pwpiwi