Proxmark/proxmark3 #868: Build error on Arch Linux.

mrsmith0x00, 2019-09-20:
Hello dear team of Proxmark!

I'm trying to build proxmark3 (git 2337 commit) on Arch Linux (gcc version 9.1.0), but I get this error:
```
appmain.c: In function 'UsbPacketReceived':
appmain.c:1001:40: error: taking address of packed member of 'struct <anonymous>' may result in an unaligned pointer value [-Werror=address-of-packed-member]
 1001 |    setSamplingConfig((sample_config *) c->d.asBytes);
      |                                        ^
cc1: all warnings being treated as errors
make[1]: *** [../common/Makefile.common:81: obj/appmain.o] Error 1
make[1]: Leaving directory '/home/user/src/proxmark/armsrc'
make: *** [Makefile:31: armsrc/all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
```

Thanks in advance!

pwpiwi, 2019-09-21:
Which ARM toolchain is included with your Arch Linux? Please type
```
arm-none-eabi-gcc --version
```
and post the result.

mrsmith0x00, 2019-09-21:
```
arm-none-eabi-gcc (Arch Repository) 9.2.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

pwpiwi, 2019-09-21:
I am not using gcc 9.1.0 yet (not sure if it is available at all for Windows). Tried to fix with PR #869 nevertheless. Please give it a try. There might be more of this kind of warnings...

mrsmith0x00, 2019-09-21:
@pwpiwi now it's working. Thanks a lot!

closed 2019-09-21 by pwpiwi