Proxmark/proxmark3 #871: Proxmark III "getting started guide" devkitARM

Leseratte10, 2019-09-30:
Hello, 

this github issue is not strictly related to this repository but I hope someone will be able to help me anyways

I found on the internet the "Proxmark III getting started guide"
https://initrd.net/stuff/proxmark3/doc/Proxmark%20III%20-%20Ubuntu%20GSG%20v1.pdf

That guide mentions devkitARM and that it needs to be installed in version "release 32". Unfortunately, the link to this version is down, and I need exactly this version for another project of mine (unrelated to proxmark). I searched your forum and wiki, and found this old revision of a wiki page where it says you need devkitARM r41 (but the download link is dead as well): 
https://github.com/Proxmark/proxmark3/wiki/Ubuntu-Linux/1c06bcc8487e76eb4285096dcec17e42922af61c

I looked through the forum and found various mentions that devkitARM is required (or isn't required in some cases, I guess that's why it was removed from the github wiki page), but I was unable to find a working download link and I really need that version of devkitARM (r32). 

Does anyone of the proxmark3 team still happen to have a working download link for devkitARM_r32-i686-linux.tar.bz2? If not, does anyone have a working link for any other version, like r41?

I really need that old r32, and the PDF file I found is the only hit on Google that contains any information about that, so I'm hoping anyone reading this issue report will be able to help me. 
I'm sorry for using this bugtracker for a request that has nothing to do with proxmark3 itself, but the mail address on the webpage (http://www.proxmark.org/) wasn't working (I received a "Undelivered Mail returned to sender" error mail) and I wasn't sure how to best contact you. 

Best regards, 
Leseratte10

RfidResearchGroup, 2019-09-30:
Wayback the proxmark repo was hosted on Google code.  You find it still here
https://code.google.com/archive/p/proxmark3/downloads

If you are extremely lucky you might find the devKitARM that you are looking for in one of the old proxspace zipfiles.
like this one.
https://storage.googleapis.com/google-code-archive-downloads/v2/code.google.com/proxmark3/ProxSpace-20100226-r390.7z

Leseratte10, 2019-09-30:
Thanks a lot for your reply, I really appreciate the help. I downloaded all the archives from google code and looked through them, but sadly I was only able to find some windows binaries of devkitARM in there, no linux ones. I'm hoping that one of the developers might still have the original archive somewhere on their computer from when it was required ...

RfidResearchGroup, 2019-09-30:
Proxspace is mingw setup.  And I have no idea whom actually put it together the first time.
@Gator96100  does maintain the most updated proxspace versions.  He might have an old archive laying around but I wouldn't count on it.

Happy huntin'

Gator96100, 2019-09-30:
I only have Windows binaries of devkitARM. For the proxmark firmware devkitARM was only used on Windows, so I doubt anyone here has the linux binaries.

Leseratte10, 2019-10-01:
Well in the "proxmark 3 Ubuntu getting starting guide" it mentions the need to download the linux version so it looks like at some point it was used on Linux as well.

closed 2020-02-13 by pwpiwi