Proxmark/proxmark3 #874: Proxmark3 v2 cant seem to write block 0 of a t55xx card

kevindacentom, 2019-10-09:
Im trying to clone a keri card (I believe to be t55), info down below. I'm trying to go through block by block and write to the blank, and each line does well except the block 0. When I try to write the block 0 from the actual card onto the blank, it leaves the string of 1's below. I'm brand new to the scene, so I'm truly sorry for my naivety but I'm trying to catch up as fast as I can and thought I was getting close to cloning this card. Also important to note that the blank card mod is ASK/Manchester while the actual card is PSK1, hence the config setup (that Im not entirely sure is doing anything since it still detects the blank as ASK right after I set it up as PSK1).

Do I have to maybe go through and try all the stock passwords from https://github.com/Proxmark/proxmark3/blob/master/client/default_pwd.dic? Any advice helps. 

**Blank card info and writing attempt:**

proxmark3> lf t55 detect
Chip Type  : T55x7
Modulation : ASK
Bit Rate   : 2 - RF/32
Inverted   : No
Offset     : 31
Seq. Term. : No
Block0     : 0x00088040

Downlink Mode used : default/fixed bit length

proxmark3> lf search u p
NOTE: some demods output possible binary
  if it finds something that looks like a tag
False Positives ARE possible

Checking for known tags:

No Known Tags Found!

Checking for Unknown tags:

Possible Auto Correlation of 32 repeating samples

Using Clock:32, Invert:0, Bits Found:513
ASK/Manchester - Clock: 32 - Decoded bitstream:
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000

Unknown ASK Modulated and Manchester encoded Tag Found!

if it does not look right it could instead be ASK/Biphase - try 'data rawdemod ab'

Valid T55xx Chip Found
Try lf t55xx ... commands

proxmark3> lf t55 config b 32 d PSK1 o 28
Chip Type  : T55x7
Modulation : PSK1
Bit Rate   : 2 - RF/32
Inverted   : No
Offset     : 28
Seq. Term. : No
Block0     : 0x00000000

proxmark3> lf t55 write b 0 d 6000F014
Writing page 0  block: 00  data: 0x6000F014
proxmark3> lf search u
NOTE: some demods output possible binary
  if it finds something that looks like a tag
False Positives ARE possible

Checking for known tags:

No Known Tags Found!

Checking for Unknown tags:

Possible Auto Correlation of 256 repeating samples

Using Clock:128, Invert:0, Bits Found:235
ASK/Manchester - Clock: 128 - Decoded bitstream:
1111111111111111
1111111111111111
1111111111111111
1111111111111111
1111111111111111
1111111111111111
1111111111111111
1111111111111111
1111111111111111
1111111111111111
1111111111111111
1111111111111111
1111111111111111
1111111111111111
11111111111

Unknown ASK Modulated and Manchester encoded Tag Found!

if it does not look right it could instead be ASK/Biphase - try 'data rawdemod ab'

proxmark3> lf t55 read b 0
Reading Page 0:
blk | hex data | binary
----+----------+---------------------------------
  0 | 68976897 | 01101000100101110110100010010111

**Info on card to be cloned:**

proxmark3> lf t55 detect
Chip Type  : T5555(Q5)
Modulation : PSK1
Bit Rate   : 2 - RF/32
Inverted   : No
Offset     : 57
Seq. Term. : No
Block0     : 0x6000F014

Downlink Mode used : default/fixed bit length

proxmark3> lf search u
NOTE: some demods output possible binary
  if it finds something that looks like a tag
False Positives ARE possible

Checking for known tags:

No Known Tags Found!

Checking for Unknown tags:

Possible Auto Correlation of 2048 repeating samples

Using Clock:32, invert:0, Bits Found:938
PSK1 demoded bitstream:
0111110000000000
0000000000000000
0001011000010110
1010011001010111
0111110000000000
0000000000000000
0001011000010110
1010011001010111
0111110000000000
0000000000000000
0001011000010110
1010011001010111
0111110000000000
0000000000000000
0001011000010110
1010011001010111
0111110000000000
0000000000000000
0001011000010110
1010011001010111
0111110000000000
0000000000000000
0001011000010110
1010011001010111
0111110000000000
0000000000000000
0001011000010110
1010011001010111
0111110000000000
0000000000000000
0001011000010110
1010011001010111

Possible unknown PSK1 Modulated Tag Found above!

Could also be PSK2 - try 'data rawdemod p2'

Could also be PSK3 - [currently not supported]

Could also be NRZ - try 'data rawdemod nr'

marshmellow42, 2019-10-09:
This is a topic best left for the forum, as more eyes will see it and be more able to help.

On first glance you look to be trying to take a q5 config and putting it on a t5577 which is won't do what you expect.

kevindacentom, 2019-10-09:
Sounds good, ill give it a try thanks.

pwpiwi, 2019-10-10:
... and success (http://www.proxmark.org/forum/viewtopic.php?id=6910).

closed 2019-10-10 by pwpiwi

graysonpike, 2023-05-26:
@pwpiwi Do you have the details of your solution? The forum post that you linked is no longer public (and the signup page for the forum is returning a 404 as well.)